jgyates/genmon #1000: Mopeka Low Fuel Warnings

bcrazycramer, 2024-01-03:
Mopeka Low Fuel Warnings

### Expected Behavior

Report low fuel when fuel is low. 

### Actual Behavior

Fuel Level is Ok, Fuel Level is Low. (10 x in December) The fuel has always been near full. 

### Steps to Reproduce (including precondition)

Random times, I will get Fuel Level is Low, followed by Fuel Level is Ok. Usually, there is about a 1 hour gap in between these warnings. 

### Your Environment

- Generator Model: G0070631
- Genmon Logs: Submitted
- Genmon Version: V1.18.18

Are there any logs I can check to see why I'm getting false warnings?

jgyates, 2024-01-03:
There is a log file for the Mopeka add on: /var/log/genmopeka.log

the first thing I would suggest is to update to the latest version of genmon as the mopeka add on has had some updates since V1.18.18

You can also enabled the debug flag in /etc/genmon/genmopeka.conf and additional info will be printed to the log. To do this, add this line to the  /etc/genmon/genmopeka.conf file:

     debug=True

Once you are finished troubleshooting you should remove this line or set debug=False as the increased log volume will degrade the SD card over time.

Once you have updated to the latest you can send your logs again if the error still occurs.

bcrazycramer, 2024-01-03:
I have enabled the debugging. I have also upgraded to the latest version. 

Here is what I see in genmopeka.log:

2023-12-25 10:18:31,115 : Error in ConvertTankReading: math domain error : genmopeka.py:522

Should we just wait and see if we get that warning again? 

Sending the logs over now.

jgyates, 2024-01-03:
I dont see the debug info in your log. I forgot to mention that you have to restart the software if you make changes to the config file. If you did not make the change to /etc/genmon/genmopeka.conf until after you updated the software, then you will need to restart genmon. Just doing an update again will restart the software.

The issue with the following error

       2023-12-25 10:18:31,115 : Error in ConvertTankReading: math domain error : genmopeka.py:522

is that this error references the old version of the genmopeka.py add so it is difficult to know the context of the error as code was added multiple times to the source file. If it occurs again send the logs and I can take a look.

bcrazycramer, 2024-01-03:
I restarted the PI.

Here is what I'm seeing now:

pi@xxxxgenmon:/var/log$ tail genmopeka.log
2024-01-03 13:20:12,353 : Reading inches: 19.84 inches
2024-01-03 13:20:14,373 : Stop Scan Filtered
2024-01-03 13:20:14,374 : stopped scan
2024-01-03 13:20:14,374 : Tank Level in mm: 504
2024-01-03 13:20:14,374 : Tank Level in inches: 19.84
2024-01-03 13:20:14,374 : Gallons Left: 178.04
2024-01-03 13:20:14,374 : Tank Percentage: 70.95
2024-01-03 13:20:14,375 : Tank1 = 70.95
2024-01-03 13:20:14,375 : Tank Data = {"Percentage": 70.95, "Tank Name": "Mopeka Sensor Tank"}
2024-01-03 13:20:14,399 : OK
pi@xxxxgenmon:/var/log$

jgyates, 2024-01-03:
So far all of this looks normal. Hopefully the updates fixed your issue but if you see the error again let me know. I am going to close this issue for now but post any follow up status to this thread. You may want to put a reminder on your calendar to take out the debug flag in a month or so if you don't see the issue again.

closed 2024-01-03 by jgyates

bcrazycramer, 2024-01-08:
Here is some of information in the genmopeka.log after getting a message that the tank was low on fuel. 

This morning I upgraded to the latest version. 

2024-01-08 06:48:14,894 : starting scan
2024-01-08 06:48:14,894 : Start Scan Filtered
2024-01-08 06:48:15,312 : Unsupported GAP report type 0x2 on sensor BDAddress('FF:6E:D8:DE:7D:1B')
2024-01-08 06:48:15,312 : data: ['0x1', '0x0', '0x1', '0x1b', '0x7d', '0xde', '0xd8', '0x6e', '0xff', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x4f', '0x2a', '0x9a', '0x4', '0xde', '0x7d', '0x1b', '0xd', '0x39', '0x3', '0x2', '0xe5', '0xfe', '0xb2']
2024-01-08 06:48:15,312 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-01-08 06:48:15,313 : Battery Volts: 2.46875 V
2024-01-08 06:48:15,313 : Battery Percent: 41.3%
2024-01-08 06:48:15,313 : Temp: 35.6 F
2024-01-08 06:48:15,313 : Reading MM: 524 mm
2024-01-08 06:48:15,313 : Reading inches: 20.63 inches
2024-01-08 06:48:18,820 : Unsupported GAP report type 0x2 on sensor BDAddress('FF:6E:D8:DE:7D:1B')
2024-01-08 06:48:18,821 : data: ['0x1', '0x0', '0x1', '0x1b', '0x7d', '0xde', '0xd8', '0x6e', '0xff', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x4f', '0x2a', '0x9a', '0x4', '0xde', '0x7d', '0x1b', '0x10', '0x39', '0x3', '0x2', '0xe5', '0xfe', '0xb1']
2024-01-08 06:48:18,821 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-01-08 06:48:18,821 : Battery Volts: 2.46875 V
2024-01-08 06:48:18,821 : Battery Percent: 41.3%
2024-01-08 06:48:18,821 : Temp: 35.6 F
2024-01-08 06:48:18,822 : Reading MM: 524 mm
2024-01-08 06:48:18,822 : Reading inches: 20.63 inches
2024-01-08 06:48:22,327 : Unsupported GAP report type 0x2 on sensor BDAddress('FF:6E:D8:DE:7D:1B')
2024-01-08 06:48:22,327 : data: ['0x1', '0x0', '0x1', '0x1b', '0x7d', '0xde', '0xd8', '0x6e', '0xff', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x4f', '0x2a', '0x9a', '0x4', '0xde', '0x7d', '0x1b', '0xd', '0x3d', '0x3', '0x2', '0xe5', '0xfe', '0xb2']
2024-01-08 06:48:22,327 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-01-08 06:48:22,328 : Battery Volts: 2.46875 V
2024-01-08 06:48:22,328 : Battery Percent: 41.3%
2024-01-08 06:48:22,328 : Temp: 35.6 F
2024-01-08 06:48:22,328 : Reading MM: 524 mm
2024-01-08 06:48:22,328 : Reading inches: 20.63 inches
2024-01-08 06:48:25,837 : Unsupported GAP report type 0x2 on sensor BDAddress('FF:6E:D8:DE:7D:1B')
2024-01-08 06:48:25,838 : data: ['0x1', '0x0', '0x1', '0x1b', '0x7d', '0xde', '0xd8', '0x6e', '0xff', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x4f', '0x2a', '0x9a', '0x4', '0xde', '0x7d', '0x1b', '0x12', '0x3b', '0x3', '0x2', '0xe5', '0xfe', '0xb2']
2024-01-08 06:48:25,838 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-01-08 06:48:25,838 : Battery Volts: 2.46875 V
2024-01-08 06:48:25,838 : Battery Percent: 41.3%
2024-01-08 06:48:25,838 : Temp: 35.6 F
2024-01-08 06:48:25,839 : Reading MM: 524 mm
2024-01-08 06:48:25,839 : Reading inches: 20.63 inches
2024-01-08 06:48:29,346 : Unsupported GAP report type 0x2 on sensor BDAddress('FF:6E:D8:DE:7D:1B')
2024-01-08 06:48:29,346 : data: ['0x1', '0x0', '0x1', '0x1b', '0x7d', '0xde', '0xd8', '0x6e', '0xff', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x4f', '0x2a', '0x9a', '0x4', '0xde', '0x7d', '0x1b', '0x12', '0x36', '0x3', '0x2', '0xe5', '0xfe', '0xb1']
2024-01-08 06:48:29,346 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-01-08 06:48:29,346 : Battery Volts: 2.46875 V
2024-01-08 06:48:29,347 : Battery Percent: 41.3%
2024-01-08 06:48:29,347 : Temp: 35.6 F
2024-01-08 06:48:29,347 : Reading MM: 524 mm
2024-01-08 06:48:29,347 : Reading inches: 20.63 inches
2024-01-08 06:48:29,895 : Stop Scan Filtered
2024-01-08 06:48:29,895 : stopped scan

jgyates, 2024-01-08:
I have received your logs. I don't see anything unusual in the data in the logs. I also do not see a low tank reading so I am guessing that the message about low tank volume occurred at a different time than when you sent the logs and the issue was temporary in nature.

The message about Unsupported GAP data is normal if the data receive by the bluetooth controller does not match what we are expecting (there are multiple types of GAP reports that can be sent). After this message we do not record tank readings from the packet.

Do you have more details on the tank reading that was reported low (how low, what is the reading in inches or mm)?

If you have ssh access you could zip the archived log files:

     zip  test.zip /var/log/genmopeka*

Then send me the file test.zip (attach it to this thread). Let me know if you have any questions.

bcrazycramer, 2024-01-08:
Thanks for checking. The last fuel warning we got was Friday at 6:41PM. 

The zipped file is attached. 

Thanks,
Brandon
[test.zip](https://github.com/jgyates/genmon/files/13865585/test.zip)

jgyates, 2024-01-08:
That helps. I see where there we low readings on the 4th and 5th. Let me analyze this and try to find the source.

jgyates, 2024-01-09:
I checked in an update that has more debug logging and a new option for the config file. Log into your pi and edit /etc/genmon/genmopeka.conf by adding this line:

     min_reading_quality=1

this value can be an integer from 0 to 3. Zero does not check the quality of the packet, 3 will reject all packets. Lets try this with a quality of 1 (i.e. reject packets with a zero quality rating).

Let it run for a while, you can send me the logs and I can check them to see if any packets were rejected. If you get no packets (your tank is reading zero) then remote this line or set it to zero (min_reading_quality=0). Let me know how it goes.

bcrazycramer, 2024-03-04:
I'm having trouble with fuel levels on a remote site at random times of the day. 

2024-03-04 11:22:14,273 : starting scan
2024-03-04 11:22:14,274 : Start Scan Filtered
2024-03-04 11:22:20,855 : Unsupported GAP report type 0x2 on sensor BDAddress('C4:45:32:B9:C6:C1')
2024-03-04 11:22:20,855 : data: ['0x1', '0x0', '0x1', '0xc1', '0xc6', '0xb9', '0x32', '0x45', '0xc4', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x58', '0x2c', '0x84', '0x3', '0xb9', '0xc6', '0xc1',
 '0x7f', '0xe6', '0x3', '0x2', '0xe5', '0xfe', '0xb1']
2024-03-04 11:22:20,855 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-03-04 11:22:20,856 : Battery Volts: 2.75 V
2024-03-04 11:22:20,856 : Battery Percent: 84.6%
2024-03-04 11:22:20,856 : Temp: 39.2 F
2024-03-04 11:22:20,856 : Reading MM: 394 mm
2024-03-04 11:22:20,856 : Reading inches: 15.51 inches
2024-03-04 11:22:20,857 : Reading Quality: 0
2024-03-04 11:22:27,874 : Unsupported GAP report type 0x2 on sensor BDAddress('C4:45:32:B9:C6:C1')
2024-03-04 11:22:27,874 : data: ['0x1', '0x0', '0x1', '0xc1', '0xc6', '0xb9', '0x32', '0x45', '0xc4', '0x12', '0xd', '0xff', '0x59', '0x0', '0x3', '0x58', '0x2c', '0xb9', '0x64', '0xb9', '0xc6', '0xc1'
, '0x7f', '0xed', '0x3', '0x2', '0xe5', '0xfe', '0xb2']
2024-03-04 11:22:27,874 : Sensor Type: Pro Check, Bottom-up propane sensor
2024-03-04 11:22:27,874 : Battery Volts: 2.75 V
2024-03-04 11:22:27,875 : Battery Percent: 84.6%
2024-03-04 11:22:27,875 : Temp: 39.2 F
2024-03-04 11:22:27,875 : Reading MM: 4122 mm
2024-03-04 11:22:27,875 : Reading inches: 162.28 inches
2024-03-04 11:22:27,875 : Reading Quality: 1
2024-03-04 11:22:29,274 : Stop Scan Filtered
2024-03-04 11:22:29,274 : stopped scan
2024-03-04 11:22:29,275 : Processed Ads:5
2024-03-04 11:22:29,275 : Rejected Ads:11
2024-03-04 11:22:29,275 : Skipped Ads:0
2024-03-04 11:22:29,275 : Zero Length Ads:12 
2024-03-04 11:22:29,275 : Ignored Ads:0
2024-03-04 11:22:29,275 : Tank Level in mm: 4122
2024-03-04 11:22:29,276 : Tank Level in inches: 162.28
2024-03-04 11:22:29,276 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-04 11:22:29,276 : Tank Data = {"Percentage": 0, "Tank Name": "Mopeka Sensor Tank"}
2024-03-04 11:22:29,277 : SendCommand Result: OK
pi@birdgenmon:/var/log$       

I have the the line set for min_reading_quality=1 in my genmopeka.conf file. Should I change it to 2?

Thanks,
Brandon

jgyates, 2024-03-04:
you should probably first look at the signal strength of the sensor relative to the pi. If you are using a sensor mounted on the bottom of the tank, are you using stand offs on the bottom of the tank to allow for an air gap between the bottom of the tank and the ground surface?

You have two packets received in your log, both are low signal quality. If you increase them min_reading_quality that may help but it also may prevent any readings, depending on the signal quality. It is worth a try to increase the min_signal_quality just keep in mind that if the issues is poor signal strength, the only solution **might** be to try to get a better signal, not filter out low quality packets.

also, do you have a referene of what the tank reading actully is?
Are your settings on the add on page matching your tank?
Does the mopeka app work?

The mopeka app on your phone vs the pi is not an apples to apples comparison since the pi likley has a smaller bluetooth antenna.

bcrazycramer, 2024-03-04:
Here is what the tank normally shows:

pi@birdgenmon:/var/log$ tail genmopeka.log
2024-03-04 08:21:44,236 : Skipped Ads:0
2024-03-04 08:21:44,236 : Zero Length Ads:3
2024-03-04 08:21:44,236 : Ignored Ads:0
2024-03-04 08:21:44,237 : Tank Level in mm: 385
2024-03-04 08:21:44,237 : Tank Level in inches: 15.16
2024-03-04 08:21:44,237 : Gallons Left: 127.24
2024-03-04 08:21:44,237 : Tank Percentage: 50.71
2024-03-04 08:21:44,238 : Tank1 = 50.71
2024-03-04 08:21:44,238 : Tank Data = {"Percentage": 50.71, "Tank Name": "Mopeka Sensor Tank"}
2024-03-04 08:21:44,262 : SendCommand Result: OK

The Genmon is mounted on the backside of the generator which is the farthest possible point to the tank. That probably doesn't help with the signal. I will try setting the min_signal_quality to 2. 

The Mopeka app picks up the signal fine. The Mopeka app was showing the tank 50% full last time I was there.  

Thanks for the help.

bcrazycramer, 2024-04-03:
Hello,

I installed the new Mopeka Pro+. I'm still getting alarms that the fuel is low. I currently have the genmopeka.conf set for min_reading_quality=2.

I don't have the logs for when it happened on Monday but here is what I see in the genmopeka logs currently:

2024-04-03 10:42:55,794 : starting scan
2024-04-03 10:42:55,795 : Start Scan Filtered
2024-04-03 10:43:04,929 : Unsupported GAP report type 0x2 on sensor BDAddress('D3:8D:C1:EA:B1:BE')
2024-04-03 10:43:04,930 : data: ['0x1', '0x0', '0x1', '0xbe', '0xb1', '0xea', '0xc1', '0x8d', '0xd3', '0x12', '0xd', '0xff', '0x59', '0x0', '>
2024-04-03 10:43:04,930 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities
2024-04-03 10:43:04,930 : Battery Volts: 3.5 V
2024-04-03 10:43:04,930 : Battery Percent: 100.0%
2024-04-03 10:43:04,930 : Temp: 50.0 F
2024-04-03 10:43:04,931 : Reading MM: 359 mm
2024-04-03 10:43:04,931 : Reading inches: 14.13 inches
2024-04-03 10:43:04,931 : Reading Quality: 2
2024-04-03 10:43:10,795 : Stop Scan Filtered
2024-04-03 10:43:10,796 : stopped scan
2024-04-03 10:43:10,796 : Processed Ads:374
2024-04-03 10:43:10,796 : Rejected Ads:48
2024-04-03 10:43:10,796 : Skipped Ads:0
2024-04-03 10:43:10,797 : Zero Length Ads:364
2024-04-03 10:43:10,797 : Ignored Ads:1187
2024-04-03 10:43:10,797 : Tank Level in mm: 359
2024-04-03 10:43:10,797 : Tank Level in inches: 14.13
2024-04-03 10:43:10,797 : Gallons Left: 115.8
2024-04-03 10:43:10,798 : Tank Percentage: 46.15
2024-04-03 10:43:10,798 : Tank1 = 46.15
2024-04-03 10:43:10,798 : Tank Data = {"Percentage": 46.15, "Tank Name": "Mopeka Sensor Tank"}
2024-04-03 10:43:10,799 : SendCommand Result: OK
2024-04-03 11:43:10,799 : starting scan
2024-04-03 11:43:10,800 : Start Scan Filtered
2024-04-03 11:43:11,305 : Unsupported GAP report type 0x2 on sensor BDAddress('D3:8D:C1:EA:B1:BE')
2024-04-03 11:43:11,305 : data: ['0x1', '0x0', '0x1', '0xbe', '0xb1', '0xea', '0xc1', '0x8d', '0xd3', '0x12', '0xd', '0xff', '0x59', '0x0', '>
2024-04-03 11:43:11,305 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities
2024-04-03 11:43:11,306 : Battery Volts: 3.5 V
2024-04-03 11:43:11,306 : Battery Percent: 100.0%

Thanks,
Brandon

bcrazycramer, 2024-04-03:
Oh right. I forgot about that. 

This might be something:

(Fuel Level is low)

2024-03-30 07:18:10,285 : starting scan
2024-03-30 07:18:10,286 : Start Scan Filtered
2024-03-30 07:18:12,254 : Unsupported GAP report type 0x2 on sensor BDAddress('D3:8D:C1:EA:B1:BE')
2024-03-30 07:18:12,255 : data: ['0x1', '0x0', '0x1', '0xbe', '0xb1', '0xea', '0xc1', '0x8d', '0xd3', '0x12', '0xd', '0xff', '0x59', '0x0', '>
2024-03-30 07:18:12,255 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities
2024-03-30 07:18:12,255 : Battery Volts: 3.46875 V
2024-03-30 07:18:12,255 : Battery Percent: 100.0%
2024-03-30 07:18:12,255 : Temp: 42.8 F
2024-03-30 07:18:12,255 : Reading MM: 1391 mm
2024-03-30 07:18:12,256 : Reading inches: 54.76 inches
2024-03-30 07:18:12,256 : Reading Quality: 2
2024-03-30 07:18:22,499 : Unsupported GAP report type 0x2 on sensor BDAddress('D3:8D:C1:EA:B1:BE')
2024-03-30 07:18:22,499 : data: ['0x1', '0x0', '0x1', '0xbe', '0xb1', '0xea', '0xc1', '0x8d', '0xd3', '0x12', '0xd', '0xff', '0x59', '0x0', '>
2024-03-30 07:18:22,499 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities
2024-03-30 07:18:22,500 : Battery Volts: 3.46875 V
2024-03-30 07:18:22,500 : Battery Percent: 100.0%
2024-03-30 07:18:22,500 : Temp: 42.8 F
2024-03-30 07:18:22,500 : Reading MM: 1391 mm
2024-03-30 07:18:22,501 : Reading inches: 54.76 inches
2024-03-30 07:18:22,501 : Reading Quality: 2
2024-03-30 07:18:25,286 : Stop Scan Filtered
2024-03-30 07:18:25,287 : stopped scan
2024-03-30 07:18:25,287 : Processed Ads:272
2024-03-30 07:18:25,287 : Rejected Ads:31
2024-03-30 07:18:25,287 : Skipped Ads:0
2024-03-30 07:18:25,288 : Zero Length Ads:267
2024-03-30 07:18:25,288 : Ignored Ads:1153
2024-03-30 07:18:25,288 : Tank Level in mm: 1391
2024-03-30 07:18:25,288 : Tank Level in inches: 54.76
2024-03-30 07:18:25,288 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-30 07:18:25,289 : Tank Data = {"Percentage": 0, "Tank Name": "Mopeka Sensor Tank"}
2024-03-30 07:18:25,290 : SendCommand Result: OK

(Fuel Level is OK)  The time I got a text about Fuel level is OK is about 10 minutes later at 12:28

2024-03-30 12:19:25,309 : starting scan
2024-03-30 12:19:25,309 : Start Scan Filtered
2024-03-30 12:19:35,506 : Unsupported GAP report type 0x2 on sensor BDAddress('D3:8D:C1:EA:B1:BE')
2024-03-30 12:19:35,506 : data: ['0x1', '0x0', '0x1', '0xbe', '0xb1', '0xea', '0xc1', '0x8d', '0xd3', '0x12', '0xd', '0xff', '0x59', '0x0', '>
2024-03-30 12:19:35,507 : Sensor Type: PRO+ Bottom-up Boosted BLE sensor, all commodities
2024-03-30 12:19:35,507 : Battery Volts: 3.46875 V
2024-03-30 12:19:35,507 : Battery Percent: 100.0%
2024-03-30 12:19:35,507 : Temp: 50.0 F
2024-03-30 12:19:35,507 : Reading MM: 357 mm
2024-03-30 12:19:35,508 : Reading inches: 14.06 inches
2024-03-30 12:19:35,508 : Reading Quality: 2
2024-03-30 12:19:40,310 : Stop Scan Filtered
2024-03-30 12:19:40,310 : stopped scan
2024-03-30 12:19:40,310 : Processed Ads:277
2024-03-30 12:19:40,311 : Rejected Ads:32
2024-03-30 12:19:40,311 : Skipped Ads:0
2024-03-30 12:19:40,311 : Zero Length Ads:272
2024-03-30 12:19:40,311 : Ignored Ads:1153
2024-03-30 12:19:40,311 : Tank Level in mm: 357
2024-03-30 12:19:40,311 : Tank Level in inches: 14.06
2024-03-30 12:19:40,312 : Gallons Left: 115.02
2024-03-30 12:19:40,312 : Tank Percentage: 45.84
2024-03-30 12:19:40,312 : Tank1 = 45.84
2024-03-30 12:19:40,312 : Tank Data = {"Percentage": 45.84, "Tank Name": "Mopeka Sensor Tank"}
2024-03-30 12:19:40,313 : SendCommand Result: OK

Thanks,
Brandon

bcrazycramer, 2024-04-03:
This is from another remote but I must not have logging enabled. I really want to get the R3D Propane Guage Modules but I can't seem to get them from anyone. 

2024-03-17 21:57:59,978 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-17 22:58:14,980 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-17 23:58:29,983 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 00:58:44,984 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 01:58:59,986 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 02:59:14,988 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 03:59:29,990 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 04:59:44,992 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 05:59:59,994 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 07:00:14,996 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 08:00:29,998 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 09:00:45,000 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 10:01:00,002 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 11:01:15,004 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 12:01:30,006 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 13:01:45,008 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 14:02:00,009 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 15:02:15,011 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 16:02:30,013 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 17:02:45,015 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 18:03:00,017 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 19:03:15,019 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 20:03:30,021 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 21:03:45,022 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 22:04:00,024 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-18 23:04:15,026 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 00:04:30,028 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 01:04:45,030 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 02:05:00,032 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 03:05:15,034 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 04:05:30,036 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 05:05:45,037 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 06:06:00,040 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 07:06:15,041 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 08:06:30,044 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 09:06:45,045 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 10:07:00,047 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 11:07:15,049 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 12:07:30,051 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 13:07:45,053 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 14:08:00,061 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 15:08:15,062 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 16:08:30,065 : Error in ConvertTankReading: math domain error : genmopeka.py:546
2024-03-19 17:08:45,067 : Error in ConvertTankReading: math domain error : genmopeka.py:546

Thanks,
Brandon

jgyates, 2024-04-03:
Note these two lines:

2024-03-30 07:18:25,288 : Tank Level in inches: 54.76
2024-03-30 07:18:25,288 : Error in ConvertTankReading: math domain error : genmopeka.py:546

The tank level normally is around 14 inches based on the recent logs. This looks like the sensor is returning a value that is higher than the size of the tank. This is causing the fluids library (the python library that handles tank geometry) to issue the domain error. I can add a check in the code that will disregard this reading that will fix the issue, not sure why the sensor is reporting this value...

Can you send the log output of when the add on restarts. There is some data that that will help. I am looking for the data for the tank selected.

jgyates, 2024-04-03:
I checked in an update that will ignore values that trigger this error. It will fall back to the last valid reading. If not valid reading has occurred for the sensor it will return a tank level of zero to genmon. Update from the About page and let me know if you have any other issues.

bcrazycramer, 2024-04-03:
Thank you!!

Just send you the logs from the about page after updating?