jgyates/genmon #1002: alarm status

cdonaven, 2024-01-06:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior
{Please write here}

### Actual Behavior

finally got Genmon installed & working.....the only issue is that under "status" it says "system in alarm" and "maintenance required".....I have a Briggs & Stratton GC1032 controller and going directly out to the generator, there are no alarms or maintenance required.......I am using a Raspberry Pi 4B and a USR-TCP232-410S converter
{Please write here}

### Steps to Reproduce (including precondition)

[image]
any idea's? tips?......thanks in advance

{Please write here}

### Your Environment

- Generator Model: Briggs & Stratton 17kW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.19.00

jgyates, 2024-01-06:
I have not received your log files. Do you have outbound email setup and working (you should get an email when genmon starts if you have it setup)? Outbound email is used to send the log files to me.

You can look on the live register view page (single click the gear icon in the upper right of the web interface). On this page look at register 0044 (Alarm Register 5). If the low four bits are zero then the "Maintenance Required" condition is active on the controller.

Please send your log files if you can as this helps me tests as I can use the data sent in my testing.

You logs may have some more insight, but this appears that your maintenance timer has expired and you need a routine maintenance of the generator.

jgyates, 2024-01-06:
I received you logs. Thanks.

I think there may be a bit definition incorrect for that value in the config file for the Briggs controller . Let me look into this and get back with you. Thanks for pointing this out.

jgyates, 2024-01-06:
Good news and bad news. The good news is that it does look like there was an issue with maintenance alarm condition. I corrected that issue and checked in an update. The bad news is that it does not appear that this correct the issue for you based on the data in the register / log submission you sent. I will have to look into this further. Unfortunately  I don't have great data to work with (e.g. not data directly from Briggs). If you have any info from your dealer on the modbus mapping from your dealer that would be helpful. If you have any you send it to generatormonitor.software@gmail.com

jgyates, 2024-01-06:
when is your maintenance due. Some controllers treat this condition as a warning and others will set this condition days before the actual due date.

cdonaven, 2024-01-06:
to be honest I don't know.......like I said the generator was installed, up & running on December 9, 2023........I haven't even had a power outage yet

jgyates, 2024-01-07:
@cdonaven 

I checked in an update that should remove the error condition. I made a guess as to the actual error condition for maintenance  due. I based this off 3 data points of registers for Briggs and Stratton that have been submitted. Once your maintenance is due let me know how it goes. I am going to close this issue but post any follow up to this thread.

Thanks
Jason

closed 2024-01-07 by jgyates