jgyates/genmon #1003: Kohler and Callmebot feature request

Mehrdadg, 2024-01-07:
Simply this project is amazing and it is finally giving me insight to what is going on with my 48KW liquid cooled generac as it has never been reliable since I owned this unit. 
I am about to replace the generac with a Kohler and wanted to know if this application also supports Kohler units?
Additionally, was wondering if there is any plan to add Signal support for Callmebot addon.

Again, many thanks to jgyates and everyone else involved with creating such a great program.
regards
PS I pay annual fee for generac LTE monitoring which in my view is absolutely useless and waste of money.

jgyates, 2024-01-07:
Regarding Kohler support, I have some docs for the APM603 controller. I can add support for Kohler however I would need someone with access to the Kohler controller to get a dump of the modbus registers. [This thread](https://github.com/jgyates/genmon/issues/965) describes the process. The [modbusdump.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsmodbusdumppy) program would be used to capture the data from the controller. That data along with the controller modbus registers definitions will allow me to add support for a given controller to the project.

 The support for Signal and Facebook is already in the add on. At the time it was written I ran into some initial issues with testing and I opted to wait to test further until someone asked for this feature. At the time I also opted to not include the Signal and Facebook stuff in the web interface setup so the only way to enable it is by editing the file /etc/genmon/gencallmebot.conf directly and then restarting genmon. The details for the conf file are [here](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gencallmebotpy-optional).

If you want to test Signal, let me know how it goes hopefully I can help troubleshoot any issues if they come up.

Let me know if you have any question.

Mehrdadg, 2024-01-08:
Hi-
As soon as I get my Kohler installed I will follow the instructions and send a dump of the modbus register - it will probably be around spring time to have warmer weather to change to kohler.

Also I will follow your instructions on signal and give it a go and will share results.
Many thanks

jgyates, 2024-01-08:
Did you send a message from your Signal account to Callmebot to enable messages as described here: https://www.callmebot.com/blog/free-api-signal-send-messages/

Yes, the lack of signal in the drop down was because of the lack of testing for Signal. Once it is validated to working I will enable that option in the web interface.

jgyates, 2024-01-09:
I updated the code to show signal in the UI, I also updated the help text in the phone number field, updated the wiki, and the text in the conf file to reflect this. Thanks!

I am going to close this thread, but when you want to start with Kohler let me know.

closed 2024-01-09 by jgyates

Mehrdadg, 2024-01-14:
Hi 
I don't think the signal protocol is working... I just don't get any messages even with the correct API key and Phone #.

The Signal documentation requires the following: 
https://signal.callmebot.com/signal/send.php?phone=[phone_number]&apikey=[your_apikey]&text=[message]

I can confirm I am using the proper API key and phone_number (xxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx format in the API Key and Recipient Number for the fields of CallmeBot add-on.

 Additionally, when I can confirm when I use chrome to send the http request.

Any help is greatly appreciated.

jgyates, 2024-01-14:
are you seeing any errors in /var/log/gencallmebot.log?

jgyates, 2024-01-14:
I checked in an update. I have misspelled the name of a function in the new debug prints. I got it to send however, I am getting this message on the 2nd message:

     Request to callmebot returned an error 503, the response is: <h1>Oops! Too many requests...</h1>
     <p>You have called to the API to often. Please review your script/code/app.</p>

Not sure how often I we can send messages with Signal

jgyates, 2024-01-14:
FYI, the messages eventrually go thru, they just appear to get delayed for a little bit. Empirically, less than a minute.

jgyates, 2024-01-14:
let me know how it goes with you.

Mehrdadg, 2024-01-15:
Hi
here is the /var/log output:
2024-01-14 12:38:45,749 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 12:41:10,767 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 12:42:57,412 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 12:44:57,571 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 12:45:54,658 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 12:48:43,665 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 13:39:25,545 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 14:06:02,056 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 14:10:17,021 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 14:19:13,549 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 14:21:22,523 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 14:26:47,810 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 15:01:57,450 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 15:03:24,520 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 15:12:17,346 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 19:38:32,838 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-14 19:39:19,191 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-15 14:41:56,259 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-15 14:59:28,506 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120
2024-01-15 15:03:02,664 : Error: 'GenCallMeBot' object has no attribute 'LogDegug' : gencallmebot.py:120

Thanks
mehrdad

Mehrdadg, 2024-01-15:
> I checked in an update. I have misspelled the name of a function in the new debug prints. I got it to send however, I am getting this message on the 2nd message:
>
> ```
>  Request to callmebot returned [...]

I never had any limitations when I used the HTTP request directly (e.g. 1x request in every 10-15 seconds). however if I try to send HTTP requests one after the other I get the same message:

Oops! Too many requests...
You have called to the API to often. Please review your script/code/app.

jgyates, 2024-01-15:
Updating to the latest code from the About page will fix the "Error: 'GenCallMeBot' object has no attribute 'LogDegug' issue. Once I fixed that it works for me with signal.

Mehrdadg, 2024-01-16:
it works now! It looks like the app tries to send too many Signal messages and now I am getting the following in the callmebot.log:
2024-01-15 20:44:09,029 : Error in SendNotice: Request to callmebot returned an error 503, the response is: <h1>Oops! Too many requests...</h1>
<p>You have called to the API to often. Please review your script/code/app.</p>
 : gencallmebot.py:308
2024-01-15 20:44:09,031 : Error sending message in QueueWorker, callback failed, retrying
2024-01-15 20:46:10,809 : Error in SendNotice: Request to callmebot returned an error 503, the response is: <h1>Oops! Too many requests...</h1>
<p>You have called to the API to often. Please review your script/code/app.</p>
 : gencallmebot.py:308
2024-01-15 20:46:10,812 : Error sending message in QueueWorker, callback failed, retrying
2024-01-15 20:48:14,590 : Error in SendNotice: Request to callmebot returned an error 503, the response is: <h1>Oops! Too many requests...</h1>
<p>You have called to the API to often. Please review your script/code/app.</p>
 : gencallmebot.py:308
2024-01-15 20:48:14,591 : Error sending message in QueueWorker, callback failed, retrying

I believe it you queue the messages and send them 10-15 seconds apart, the API call limitations will go away.
thanks

jgyates, 2024-01-16:
It does queue and resends, it just does not wait for x number of seconds between messages. They will eventually go thru because of the queue and retry. I will add a new parameter that will allow for a delay between messages. this will allow operation without the error. 

When genmon (and the add on) is restarted you will get about 5 or so messages with the initial state of things. After then the messages only come when there is a change in state of something being monitored.

Mehrdadg, 2024-01-16:
Thank you so much for all of your effort!

jgyates, 2024-01-18:
I checked in an update that adds a default 5 second delay between messages for callmebot signal. All other messaging (signal, telegram) default to 0 seconds. You can change the default by editing the /etc/genmon/gencallmebot.conf file with this setting:

     minimum_wait_between_messages = 5

I got it to work with the delay as low as 1, but I was restarting the software and 5 messages are sent on startup by default. You may find you can set it lower, but after a few restarts with it at 1 second I got the "too may requests" error again.

If there is not an entry for this setting the default is used. 

Also, the same setting can be used for other short message add ons like Pushover, Slack, twillio.

Let me know if you have any questions or issues.