jgyates/genmon #1004: issue with model and SN reporting

Mehrdadg, 2024-01-07:
I have a 48KW generac liquid cooled generator and under maintenance status window the status does not show the model number and the serial number. I can confirm when using the generac mobilelink app, both of these info are present.

this is what the SW shows:

Maintenance :
    Model : Unknown
    Generator Serial Number : None - Controller has been replaced

### Your Environment
- Running with RPRI3b with pHAT
- Generator Model: {RG04854ANAC
- Genmon Version: V1.19.00

thanks

jgyates, 2024-01-07:
If you have outbound email setup and working you can submit your logs and I can validate your registers, where the serial number is read from.

If your controller has been replaced, the serial number is not programmed into the controller like it is when a controller leaves the factory installed in a generator. If your controller has been replaced in the field you will see this behavior. I am guessing that if you were using mobile link the serial number may have been read and stored by Generac and the serial number associated with your account, then at a later time the controller was replaced with a controller without the serial number programmed into it. 

The controller with a serial number have to be activated, the ones without one do not. This is why you you replace a defective controller you dont have to re-activate it. 

You can get around this by entering your serial number on the Advanced Settings page. To access the Advanced Settings page, double click the gear icon in the upper right corner of the web interface. once you pull up the Advanced Setting page, enter your serial number in the "Serial Number if Missing" filed and click save at the bottom of the page. Wait for genmon to restart, then refresh your browser. This will allow genmon to use your serial number to look up your model. You may have to update the software version on the about page to get the model lookup to occur as this only happens when the version number changes.

jgyates, 2024-01-07:
Let me know if you have any questions. I am going to close this thread but feel free to post any follow up questions to this thread.

closed 2024-01-07 by jgyates