jgyates/genmon #1007: Recommendations for Moving Existing Genmon Installation to New Operating System

jwtaylor310, 2024-01-08:
Environment:  Genmon V1.19.01 running on Raspberry Pi 3B+ connected to Generac Evolution 2.0 via USB-to-RS232 adapter.

I have been running Genmon for several years without any problems.  Great product!  The current installation is running on a Raspberry Pi 3B+ using the "Stretch" version of the Raspberry operating system.  I would like to upgrade the O/S to the "Bullseye" version.  Do you have any recommendations for how to migrate Genmon to the new environment?  I plan to do a new install of Bullseye on a new SD card.  Do you recommend that I then do a new install of Genmon or is there a way to copy the Genmon files to the new SD card?  I have a spare Pi 4B if there is any reason to upgrade the hardware.

Thank you!

jgyates, 2024-01-08:
the tips and tricks wiki page has info on backing up and restoring your genmon settings.

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#restoring-settings-from-a-backup

If you can ssh into your pi you can also use scp (ssh except copy files from one system to another) to copy a previously downloaded genmon back to your new system.

1) make a backup of your genmon settings from the About page
2) save the file downloaded in step 1, and copy it to your new system (I use scp to copy the file).

closed 2024-01-08 by jgyates

jwtaylor310, 2024-01-08:
That's exactly what I need.  Thanks for the super fast response!