jgyates/genmon #1008: Power output & long activation of the transfer switch

Mehrdadg, 2024-01-16:
Hi all
Sorry if this issue was already discussed but I wasn't able to find anything searching...

To test the genmon app, I forced a power failure and noticed two things:

1) it took 2 minutes before the generator to activate the transfer switch
2) also the output power reporting is only showing 0.5W while the entire house was running on generator 

I am using the piHat and only 4 wires I see connected to the piHat (GND, +12, RX and TX).

What is the mechanism by which power is reported? is the Evolution controller or something else?

thanks in advance for the help.
Mehrdad

PS running the latest genmon SW, Liquid cooled 48KW, RPI3b+piHat

jgyates, 2024-01-16:
Hi @Mehrdadg 

Two minutes is quite a long time before the generator starts. There is a parameter you can set on the controller via the LCD based menu that controls the delay between a power outage and generator start. I think it is 5 seconds by default. Is this a new generator? Have you tested it before?  [This page](https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon) describes how to access the dealer menu. This setting may be in there, I can't remember if it is in the dealer menu or the non dealer setting.

The Evolution 1.0 Liquid cooled model does not have 12V. It does have 5V. If you are using the piHat from pintsize.me then the cable supplied will make the proper connections.

The power reporting on the Liquid Cooled Evolution is described [here on issue number 5](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues). I have noticed the Liquid Cooled model behaves differently depending on the firmware version or hardware model. Later incarnations returned a more accurate number for current. You may try going to the Advanced settings page (double click the gear icon in the upper right of the web interface), then change parameter "Current Divider" to 0.425 and "Current Offset" to -323.31. The click save at the bottom of the page, then refresh your browser and run your test again to see if you have more accurate results.

Let me know if you have any questions.

Mehrdadg, 2024-01-16:
This is not a new unit (~3 years old) and I had nothing but trouble with my unit:( the techs have been making tons of modifications and the most recent must be the increase in time for the transfer switch. I will follow your instructions when I get the chance this weekend.

_"The Evolution 1.0 Liquid cooled model does not have 12V. It does have 5V. If you are using the piHat from pintsize.me then the cable supplied will make the proper connections."_
  -> I believe the pre-made cable has connections to the battery terminals (pin 3 & 4? going by memory:( ) and the pihat a buck convertor to step down from 12V.

I will update the current divider and offset parameters and give it a go and let you know as soon as I have the chance to carry out this.

thanks

jgyates, 2024-01-16:
Non Liquid Cooled Evolution controllers have current transformers and they often have to be calibrated. I am assuming the tolerances in the manufacturing of the Hall Effect sensor on the liquid cooled model results in the differences we are seeing. This is the reason that I put in the current divider and current offset values. Sounds like you may have do some measuring of the power to properlly calibrate the values. Alternatively, if you still want to measure the power you could use the Hat from pintsize.me that allows to to attached two current transformers to your generator. Genmon can import this data.

Also, if you have outbound email setup you can send me your log files from the About page?  I can take a look and see if anything pops out as strange. It would allow me to compare your 48kw to mine (I have a 48kw generac liquid cooled also).

Mehrdadg, 2024-01-16:
> Non Liquid Cooled Evolution controllers have current transformers and they often have to be calibrated. I am assuming the tolerances in the manufacturing of the Hall Effect sensor on the liquid [...]

happy to send the log files... can you please send me instructions and what email address I can use?

jgyates, 2024-01-16:
If you have genmon setup to use outbound email (setup on the settings page and notification page) it will send out an email to the specificed address when genmon starts and if alarms or errors occur. If this is setup and working then you can submit your logs to me by going to the About page in the web interface and clicking the "Submit Logs" button.

Mehrdadg, 2024-01-16:
great... I just submitted both Registers as well as logs.

jgyates, 2024-01-16:
I got them. Nothing is jumping out as out of the ordinary. If your controller is not set for a long delay then there could be a fuel related issues. Does the generator start from a cold start in manual operation without a delay?

Mehrdadg, 2024-01-16:
generator always start from the cold when starting in manual and almost all weekly exercise is also ok (maybe 1-2 times over the last 6 months it failed). When I need the generator to start in the event of an actual failure, it mostly fails with to start due to over-cranking error.
This generator was heavily modified by the original installer and they did things to the intake regulator that was not Generac authorized. I had new parts installed and went with another generac dealer and they came and overhauled the unit (spark plugs, filer, oil and also up sized the LP pipes to get proper flow). fast forward, it failed again few weeks ago during a power outtage.

just yesterday, I have someone spend most of the day looking over and the only thing they recommended is to use a new battery with higher CCA (currently 550CCA to 750CCA). Given the controller had lost its SN and most recently had some controller faults, it made sense to replace the battery. I just wished I went with Marine battery instead of Diehard gold:)

using your beautiful SW, I am now able to clear faults remotely to give it another start (my network and Wifi is on battery backup).

I was about to rip out generac and replace with Kohler but I am going to use your SW and the new battery and give it a few more months before I make that drastic, costly change, and not to mention doing all this in the winter months.

on a final note, I will play with the load calibration and hope I can get it going. going down the printsize.me is not simple as I have made a box outside of the unit for the genmon. 

regards

jgyates, 2024-01-16:
I am going to close this issue for now but you can post any follow up questions or status to this thread on this topic.

closed 2024-01-16 by jgyates

griffths, 2024-01-16:
If you have not already visited the following forum, the site has been a good source of reliable information related to Generac products for me in the past.

 

I would suggest creating a forum account and posting your problem statement there and I suspect you will quickly get quickly a response.

 

https://www.zillerstore.com/

 

Regards,

 

Jay

Mehrdadg, 2024-01-18:
thank you Jay! I will sign up and hope to get some good pointers on my generator.

Mehrdadg, 2024-01-21:
> I am going to close this issue for now but you can post any follow up questions or status to this thread on this topic.

Hi - 
I entered the offset and scale factors for my generator as prescribed and I am still struggling with this as when exercising the generator with no load, it reads ~48KW. When its off, it reads 0KW.
could it be scale and offset calculations incorrect? I am sure I am doing something wrong as this would have been reported by others.
thanks for a tip on this.

jgyates, 2024-01-21:
Yes, the software will report zero of the generator is not running. You can look at register 0058 on the live register page (labeled "Hall Effect Sensor (EvoLC)" and it will show you the raw value of the register. This value is modified based on this criteria for determining the current:

If current divider and and current offset are not present then the software will assume this is the current in milli-amps, mA (i.e. zero current offset an current divider of 1000).

Otherwise the software will divide the value of this register by the current divider and add the current offset to determine the current.

The current is combined with the output voltage to get the power (V*A=W) for single phase power. If you have a clamp on current meter like [this](https://www.amazon.com/Proster-Auto-Ranging-Capacitance-Temperature-Non-Contact/dp/B075Z1GH5L?th=1) you can measure the current while the generator is under load of both legs in split phase voltage configuration, add the current measurements, then look at register 0058 and calculate the value of current divider and current offset. This will calibrate the values of current offset and current divider for your specific system.

Mehrdadg, 2024-01-23:
Hi
I checked the register 058 and the value when in standby and when in exercise mode, it always reads decimal 1407. Based on this value and following your math, I get the following:

 (1407/0.425)-323.31 * 240 V = 716947 ; very stage to get this value for KW

Also given the register 058 always reads 1407, I am guessing something maybe wrong with my sensor:(
thanks

jgyates, 2024-01-23:
That may be the case. Keep in mind that the register value will not change at all if the generator is not running. It stays in the last value sent by the firmware.  My register 0058 is currently at 1297 sitting idle. It is also does not move much during exercise either. When the generator is under load is when it will increase depending on the load (I guess that is what you mean by standby).

If you want to disable Power and Current from the display you can enable the "Disable Power / Current Display" on the settings page. This will remove the power graph and disable any fuel calculations.