jgyates/genmon #1013: This site can't be reached

philruel, 2024-01-21:
On January 7th I rebooted Raspberry Pi because my Genmon page would not respond altough I could VNC or ping my Pi.
Today, my Genmon page is again not responding: This site can't be reached (Chrome).  I can still VNC, ping or SSH my Pi.
I went to the Generac generator and pressed the OFF button and received an email saying the status has changed as it should.

I did not restart my Pi yet in case it's better for diagnosis.
 
I'm using HTTPS to login to Genmon running on Raspberry OS 12.
 
Thank you!

### Your Environment

- Generator Model: 0065520
- Genmon Logs: Not submitted yet
- Genmon Version: 1.19.01

jgyates, 2024-01-22:
@philruel 

I assume you are also getting the two emails when genmon starts also?

The first thing I would consider is this: has any thing changed on your network or with the pi.  A new router, reboot of an existing router, updated the OS software on the pi, new software to the pi?

If you can post the output of this log file, that may shed some light on things:

     /var/log/genserv.log

This is the web serever. The log for genmon is:

    /var/log/genmon.log

philruel, 2024-01-22:
I am using the IP address in my browser.

I get this after waiting 4-5 minutes
admin@raspberrypi:~ $ curl https://192.168.5.52
curl: (28) SSL connection timeout

I get the error bellow if trying to connect to another / wrong IP.
admin@raspberrypi:~ $ curl https://192.168.5.53
curl: (7) Failed to connect to 192.168.5.53 port 443 after 3057 ms: Couldn't connect to server

Thank you!

jgyates, 2024-01-22:
from the command line type this:

     cd genmon
     python3 ClientInterface.py
     
If genmon is loaded this will give you a prompt, if it is not loaded it will timeout and give you an error and state genmon is not loaded.

if it gives you a prompt then type:

    exit

to exit the program.

If it is not loaded then you likley do not have the settings to load genmon on boot in your cron tab. To valiate this you can type this:

     sudo crontab -l

The output should look like this:

     @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

If it does not then run the setup program again and this time select Y for yes when asked if you want to start genmon on boot. To run the setup, from the genmon folder type:

      ./genmonmaint.sh -i

philruel, 2024-01-22:
All looks good.

```
admin@raspberrypi:~ $ cd genmon
admin@raspberrypi:~/genmon $ python3 ClientInterface.py
OK : Auto, Off - Ready
>exit
admin@raspberrypi:~/genmon $ 
```

```
admin@raspberrypi:~ $ sudo crontab -l
@reboot sleep 30 && /bin/bash /home/admin/genmon/startgenmon.sh start
```

jgyates, 2024-01-22:
I am at a loss. I would reboot the pi. If that does not work then we could try updating your libraries and / or system software. It appears that something relateed to networking on your pi has stopped working. From genmon's perspective everything looks like it is working.

jgyates, 2024-01-23:
No problem. I am glad it is working now.

closed 2024-01-23 by jgyates