jgyates/genmon #1014: EVO 1- Can I move my genmon setup to accessory port 1?

bgiuriceo, 2024-01-25:
EVO 1 22KW Air cooled (2018)

I've been using Genmon now for over 2 years...flawlessly (it's fantastic). I started when the cellular modem was obsoleted by Verizon. Now, my wife asks to have 'the app' back on her phone (and my Ecobee thermostat). I had to disconnect. The WiFi / ethernet module is now working with the free MobileLink account.
I use a Waveshare RS-232 to POE ethernet converter in the genset cabinet and the PI is in the garage:
Can I use accessort port 1 for Genmon....just move over the connector?
Can I keep the setup on port 1 and 'splice/mult' RX & TX & GND over to the WiFi ethernet board?

I assume everything is RS-232?

Thanks again for an outstanding product.

skipfire, 2024-01-25:
I've not seen a controller with both accessory ports having connectors, only the 8-pin.  The other is a 10-pin without a connector.  As long as it is the RS485 WiFi MobileLink then you can use both as they use different data wires.  You can splice things together or buy the PintSize.Me Splitter (disclosure: I'm PintSize).  If the WiFi MobileLink you have uses RS232 (which seems to be a rare unit) then you have to pick one or the other as you can only have 1 RS232 accessory.  If you are looking into the MobileLink plug and it uses the top 2 pins on the left side then it is using the RS485 (or share a picture of the end of the MobileLink cable), unfortunately unplugging MobileLink does reset the WiFi config on it.

bgiuriceo, 2024-01-25:
I used the original MobileLink cables and just replaced with a new Generac 7170 radio (and replugged the port 1 side). I'm using the 2 right side pins (7&8?). So I guess it's RS-232?

skipfire, 2024-01-25:
The 7170 appears to be a combo WiFi + Ethernet, which if it is like the WiFi + Cellular it will use both RS485 and RS232 one for each protocol.  I don't know which is which or what happens if you only connect the RS485 pins to the 7170.

bgiuriceo, 2024-01-25:
I was thinking the same but I can't seem to find a pinout for the unit. I could try the 485 side for the controller to radio and keep Genmon on the 232 side. If that's your thinking...

skipfire, 2024-01-25:
That is what I would be considering, but it's possible RS485 goes to the Ethernet or that the unit requires both connected for either to work.  If you have a depinning tool, pulling the 2 RS232 pins and seeing what it does would be the easiest non-destructive option.

bgiuriceo, 2024-01-25:
Of course I can't find the tool. Are the harnesses and connections anywhere? I can't seem to find them in the maintenace manual. They also supplied 2 cable assemblies:
10000038669 - MobileLink to EVO2
10000030045 - MobileLink to Pre EVO2 (which I should be using but I'm not it breaks the power out.

skipfire, 2024-01-25:
There's a short cable on Amazon you could use.  https://www.amazon.com/gp/product/B000M802RG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

bgiuriceo, 2024-01-25:
Any source for pinouts?

jgyates, 2024-01-25:
https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

bgiuriceo, 2024-01-25:
Thanks for all of your help. I'll report back with progress. Thanks again for a brilliant execution and stellar support.

closed 2024-01-25 by bgiuriceo