jgyates/genmon #1015: Cant get pi to accept my password.

Jkh62, 2024-01-26:
New pi user , new to its OS but not new to PC's and CMD use. Set up my SD card with pi imager and formatted and wrote to a 64 gb micro SD card for my pi ZERO. It boots and I see it and its IP address on my Wi-Fi.  When I use SSH to connect it comes up with my password prompt , I enter it and I get the following message after three tries :  genmon.local: Permission denied (publickey,password).

I downloaded the lates imager and using the standard 32 bit pi OS. Set my user name and password using a standard keyboard and I also tried it on the desktop keyboard with the same results. I am stumped if its my WIN 11 Pro permissions or something I am doing consistently wrong.  

I have tried about 5 different flashes / variables all with the same result.
ssh genmon.local takes me to the login asking for my password but its not taking it.

Any suggestions ????

EXCELLENT work and have my Pi ZERO waiting on me to load and put the HAT on.......I have to get in first.

jgyates, 2024-01-26:
are you using ssh via the command line? what command line are you using? When I log into a pi via ssh I specificy the user name and IP address, for example to log in the pi user to an raspberry pi at IP address 192.168.1.10 is use this command line:

      ssh pi@192.168.1.10

Then I am prompted for a password. Is this how you are doing it?

jgyates, 2024-01-26:
fyi, if you don't specific the user name on the pi in the command line, the ssh will use the current username of the system the ssh client program is running.

jgyates, 2024-01-26:
also, when you set the pw with a keyboard, you need to make sure you have the right country layout for the keyboard. The default is UK. If you use special characters in your password, this can be an issue.

Jkh62, 2024-01-26:
I tried CMD prompt :  ssh genmon.local which returned _
C:\>ssh genmon.local
jkh@genmon.local's password:

[image]

My region I set as USA/Detroit

Jkh62, 2024-01-26:
[image]

jgyates, 2024-01-26:
A few points:

1) The the command (ssh jkh62@192.168.1.100)above you must have a user named jkh62 on the pi with the password you are using. 
2) if you set the password on the pi for user jkh62, you must set the keyboard country. This is different than the time zone (e.g. USA/Detroit). The keyboard country only has a country, not timezone / city. If you are setting the password with a physical keyboard and your password has special characters then you must set the country keyboard code before you set the password, otherwise the special characters will be entered different (for example a * may be and &). You can set the keyboard code from the command line with the raspi-config program (must be run as root so type: sudo raspi-config). Once you are in raspi-config the keyboard country configuration are under "Localisation Options", then "Keyboard".

3) a way to avoid these types of issues is to use the raspberry pi imager program. You can set the user and password on the image before you write it to the SD card
[image]

you can also enable ssh, set the hostname and wifi info. I use this and typically never plug in a keyboard or monitor when setting up a pi.

In short you have only a few possible issues:

1) user name is wrong
2) password supplied is wrong (it is not set to what you think it is due to the keyboard country not being set properly, or your are typing it in incorrectly)
3) The IP address is wrong and something else is at that IP address with ssh enabled.

Jkh62, 2024-01-26:
I am using the pi imager, need to double check the country setting, I thought I set it right but something I am doing is wrong. Thx for the assistance, I appreciate it.

I should be able to connect with my web browser (Brave) by putting in the IP address and port number. If connected it should then ask for a password, correct??

jgyates, 2024-01-26:
no. the pi does not have a web interface. the genmon software does but that must be installed first and then the username and password is not the same as the system login.

You have to setup your pi and have it working on your network and be able to log in first. then you can install genmon per the instructions here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Jkh62, 2024-01-26:
I have all the command lines ready to load it, just have to get in the pi first.
Will give it another look after work today.

Jkh62, 2024-01-26:
Got home late , in early tomorrow  AM at work.
Same results using pi image flash utility (multiple times) and selecting US as keyboard. I did read some other people had issues with the flash actually NOT configuring the keyboard to US . Out of time playing with it tonight.

jgyates, 2024-01-27:
as a test you should try to use a password with only letters.

Jkh62, 2024-01-27:
Had to try your suggestion.
Used a three letter password, lower case and the same results.

Going to do some reading tomorrow at lunch time.
I am positive all my input info is correct on the all fields so I am not fat fingering it.

Jkh62, 2024-01-27:
Similar to what I have going on.

https://forums.raspberrypi.com/viewtopic.php?t=333248

jgyates, 2024-01-27:
the only other suggestions would be:

1) to try the 64 bit version
2) send a screenshots of the Pi Imager screens where you set the ssh and password options.

Jkh62, 2024-01-27:
I don't think the pi zero is compatible with 64 bit os.
Going to look for an older version of imager.

Jkh62, 2024-01-27:
This is interesting.
Similar issue but the image flash was dropping the 1st letter of the password you put it.

https://github.com/raspberrypi/rpi-imager/issues/754

jgyates, 2024-01-27:
I am going to close this issue as it is not related to genmon but sounds like an issue with the imager. Feel free to post any follow up questions or how you resolved this to this thread.

closed 2024-01-27 by jgyates

Jkh62, 2024-01-29:
My pi access problems were completely due to my routers security setting. It threw me a curve when I could connect to it via ssh but it would not accept my password. After changing permissions in the router it let me log in to run scripts.

If you have log in issues check your router security setting. 
I also gave the pi a static Ip address so no potential issue of the IP address changing and losing access.

jgyates, 2024-01-29:
I am glad it is working for you now. Thanks for letting us know how this was resolved.

jgyates, 2024-01-31:
I am not sure what version you are using of the raspberry pi imager, but a new version was released a few days ago (V1.8.5)

Jkh62, 2024-01-31:
I used the latest release then after I was having issues I used version 1.7.4 thinking it was an image problem.
Turned out it was me and not letting SSH pass thru my router.