jgyates/genmon #1016: Replacing PI3

jcherney, 2024-01-26:
### Expected Behavior

Wifi Connection all the time

### Actual Behavior   

Loses all Wifi after a few minutes. Multiple cold reboot tries does the same thing.

### Steps to Reproduce (including precondition)

Happens at every fresh startup

### Screenshot or Pictures relating to the problem (if possible)

(no screenshot) Flashing blue wifi  indicator goes on and then disappears after a few minutes

### Your Environment

- Generator Model: {Please write here}  (cant get to the interface for the model number ) Generac 48 kw 
- Genmon Logs: Submit via the About page  Can't get there to submit logs
- Genmon Version: {Please write here}  Current as of last week. (cant see it)

I am looking at replacing the PI3- my question is- can I just swap out the board and intall the old USB card? Or, do I have to do something to make it work? 

One potential monkey wrench: I had to replace my Xfinity modem in late December. That's when this issue started to happen. I know that the PI needs to be on the 2.4ghz channel, and because I DO get email responses for the couple of minutes I'm assuming it is connecting. properly. 

I can't putty in because the wifi won't stay up long enough.

Any helps is greatly appreciated, thanks.

jgyates, 2024-01-26:
I assume you have a script running on the pi that will reconnect as described [here](https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#setting-up-a-script-to-check-the-wifi-connection-and-reconnect-if-needed). 

Also, your signal strength should be checked aslong with the signal quality. Signal quality can be low and signal strength be high if you have a lot of reflection in the wifi signal which makes it harder to keep a signal. Make sure you have as little as possible obstructing the signal form your router to the pi. 

Do you use the wifi built into your router or is that separate as well. If you changed your router/wifi (assuming you don't have a wifi access point separate from the router) then that could be the case as not all wifi antenna is created equal. If that is the case then just changing the pi will likely not help.

It is possible to swap SD cards as long as you are the same CPU variant (e.g. pi3 to pi3), but there are a few caveats with things like wifi and Ethernet MAC addresses and BT addresses will change.

jcherney, 2024-01-27:
Thanks. Strangely enough it reconnected by itself to send me an email this morning. I can putty into it now but still no web access.

One of the issues is that I have both the xfinity wifi and a couple of Ubiquiti antennas around my home. The generator is closest to the Xfinity modem. Signal strength is decent-showing -69 dbm 

I don't know of if I have a script like that up. Would the fact that it reconnected tell
me that I do?

The IP and Mac are reserved in the modem. And it's showing the correct frequency.

If I swap out PI3 to PI3, the only issue is the new IP? If this is a possible fix I don't mind buying a new PI3.

Re card degradation: I have a backup that's many versions behind. What will happen if I load that? Will the update load all at once?

jgyates, 2024-01-27:
yes, I think if got an email that stated it reconnected then you likely have a reconnect wifi script. This is not part of the OS.

I guess it would be helpful to describe your network a little better. Here is what I know so far:

1) You have an Xfinity modem. 
2) you have multiple wifi netowrks? Xfinity and Ubiquity. Are these using different SSID (wifi network names)?

What I don't know is:

what hardware is acting as the router
how is your pi configured to get it's IP address

i am not clear what you are saying:

"The IP and Mac are reserved in the modem. And it's showing the correct frequency." Are you saying you have a reservied IP address based on the MAC address, or are you saying the wifi is connecting and the MAC and IP is showing as connected to the wifi?

If you swap out the PI3 the mac address will be different. If that changes your pi IP address depends on on how you have your pi setup to get it's IP address. It does not mean that it will not work. 

This is just a guess, but If you can ssh into the pi but the web interface does not function, then I would guess that your new Xfinity hardware (which I a guessing is acting as your router) is blocking genmon as a security feature. Some routers block web servers on the local network as a security feature. Look in the router for security related settings.

An update of an old version will update everything at once. FYI, I don't test upgraing from very old version to the latest, but I do try to code things so it works, so just make sure you have a backup of the /etc/genmon folder.

jcherney, 2024-01-27:
""The IP and Mac are reserved in the modem. And it's showing the correct frequency." Are you saying you have a reservied IP address based on the MAC address, or are you saying the wifi is connecting and the MAC and IP is showing as connected to the wifi?"

Both. When I log into the modem it shows as connected to the MAC address on the reserved IP via a reserved address and signal response. 

The Xfinity is the router

The ubiquity and Xfinity are both sending the same 2 network names and frequencies. 2.4 and 5 

I will
Look at the Xfinity and see if there is a web server restriction, thanks

skipfire, 2024-01-27:
That could explain the WiFi issues.  You have 2 independent systems trying to manage it and that changes the routing.  Even though there may be a significant difference in signal strength, it may occasionally reach out to the weaker one and have just enough connection to stick with it, but not enough to be stable.  If you want multiple nodes with the same WiFi config you really need to have a WiFi mesh system that handles when devices hop between access points as they can help bounce a device back to the stronger signal.  I would suggest giving them different SSIDs or even disabling the Xfinity one entirely (I always split my router and WiFi equipment).

jcherney, 2024-01-27:
Unfortunately I can't disable the Xfinity because its the closest to the generator.  Looking at the Xfinity control panel, I don't see a way to "allow or disallow" web pages. 

I am thinking about what you said re conflicting networks. I may add a special network at the router just for Genmon.  Once I log into putty (after figuring out the new IP address), how do I change the IP setting in Genmon?

jgyates, 2024-01-27:
Changing the network name so they are different is a good thing to do. Multiple wifi networks with the same name is problematic.

there may be settings on the xfinity described here: https://www.xfinity.com/support/articles/disable-xfinity-wifi-home-hotspot

genmon does not have IP settings. the raspberry pi operating system does. In general you edit the file /etc/wpa_supplicant/wpa_supplicant.conf

More info is here: https://pimylifeup.com/setting-up-raspberry-pi-wifi/

I am going to close this issue but feel free to post any questions to this thread.

closed 2024-01-27 by jgyates

skipfire, 2024-05-07:
Possibly, it could also be a problem between the controller and the Pi.  Did you make your own cable or did you buy a HAT?  With the build your own cables one of the wires could come loose, or with any solution the TTL chip could fail.

jcherney, 2024-05-07:
I bought a pre-made cable at the time IIRC. 

So, if the PI has gone bad, is it just a matter of swapping out the new one for the old one? Or are there settings that need to be changed?

skipfire, 2024-05-07:
If it's the same number Pi (3 for 3, 4 for 4) then it typically works fine.  If you change the number then the WiFi may have to be reconfigured.

If you have one, that's a good option to try, but you can also take it to your a desk and do loopback testing to see where the problem is.  Test loopback at the end of the Molex cable and on the Pi directly.  Even premade cables can have TTL chip failures, solder failures, screw-down failures, etc.

jcherney, 2024-05-14:
@skipfire  On a thread that jgyates referenced, you made a suggestion to check and see if the genmon program was looking for port 8000.  I ran that command line (sudo netstat -tulpn | grep LISTEN)  and it is NOT listening for 8000. 

How can I enable this?

Thanks.

skipfire, 2024-05-14:
It depends, what problem are you currently having?  Looking up through this thread I see you had an issue with losing WiFi, if things swapped without issues I'm not sure what your goal is and listening on 8000 might not be the right answer depending on what you are trying to do.

jcherney, 2024-05-14:
@skipfire   The wifi issue was fixed with the new PI3 I put in yesterday.  I'm just trying to bring up the web interface and its not working. The thread I referred to mentioned to another user to see if genmon is looking for port 8000. And by using the command I found out its not.

Does that clear it up?

skipfire, 2024-05-14:
Using SSH to log in remotely and restarting genmon with `./startgenmon.sh restart` is easy to try, after that logs are what will really help to see what's happening.  /var/logs is the folder and there is a genmon and genserv log file which are the two I think would be the most informative.

Did you previously have notifications setup, and if so are you getting them now (at least the startup)?  If not then it may mean none of genmon is starting correctly, but if you are getting the notification it may just be the UI that's failing.

jcherney, 2024-05-14:
It restarted and I got the email notifications.   Trying to change to /var/logs gets me "no such file or directory"

jcherney, 2024-05-14:
Hold on: It's working now!!   Now I'm really confused- but happy!  Why wouldn't the new hardware install have done the same thing? 

In any case- thank you!!!

skipfire, 2024-05-14:
I'm glad you have it working.  For the log directory I was going from memory as I was not at a computer last night and for future reference the correct folder for logs is `/var/log/`.

jcherney, 2024-05-14:
No worries. Thanks again for the restart info. 

Just curious- when I changed the Pi out and applied power to it, why did it not restart at that time?

skipfire, 2024-05-14:
No clue, but the logs would be the most likely place to find answers.  Since it is running again, if you have email setup you can send them with the app to jgyates and he may be able to see.  You can also scroll through the log files and see if anything jumps out at you.

jgyates, 2024-05-14:
I will be happy to look at your logs if you can submit them via the About page.

jcherney, 2024-05-14:
Thank you. I sent those to you.

jgyates, 2024-05-14:
I have not received any logs yet.

jcherney, 2024-05-16:
I just resent them. Not sure why you didn’t get them. Sent from my mobile

jgyates, 2024-05-16:
I have not received any logs. Is you email setup to use a widely used mail service like gmail or are you using one from a custom domain? Have you tested the email setup from the Setting page to make sure it can send to an email other than your notification domain?

jgyates, 2024-05-16:
Not sure why your email is not working but you can also get a zip file of your logs with this URL in the broswer:

     http://192.168.11.10:8000/cmd/get_logs

Then you can email zip file you downloaded in the above step to generatormonitor.software@gmail.com

jcherney, 2024-05-16:
Thank you. I'll do that when I get home.  

Not sure why I get emails, but you dont.

All quotes are in US dollars, ex works unless stated otherwise, and subject to prior sale.
Wire Transfer Order- US$50.00 minimum-(does not include freight charges)
Wire Transfer Order less than US$250.00 - additional US$25.00 transfer fee
Credit Card Order US$25.00 minimum-(does not include freight charges) All terms and conditions of sales and purchases can be found at www.universalbearing.com

Universal Bearing Co.
8521 Lyndon
Detroit, MI 48238
Fax 313 933 8888
Tel 313 933 1100
email  ***@***.*** ***@***.***
"ON AIRCRAFT USE NOT PERMITTED"
Neither this information block, the typed name of the sender, or anything else
in this message is intended to constitute an electronic signature for purposes of the
Uniform Electronic Transactions Act or the Electronic Signatures in Global and National
Commerce Act ("E-Sign") unless a specific statement to the contrary is included in this message

http://192.168.11.10:8000/cmd/get_logs

jcherney, 2024-05-17:
Sent to your email, thanks.

jgyates, 2024-05-17:
You logs mostly look unremarkable. There is one issue/question with the genmon.log file. You have this entry a few times (it prints on restart):

     DetectController auto-detect override (controller). EvolutionController now is False

This indicates you have manually selected the controller type by editing the /etc/genmon/genmon.conf file. Assuming you actually have a Nexus controller this may be OK, however this is not the recommended setting. You should only need to directly edit the config file in rare instances as this leads to unknown and unsupported behavior.

Your /etc/genmon/genserv.conf file (for the web interface) look normal. No signs of abnormal behavior.

jcherney, 2024-05-17:
Thanks. It is a Nexus controller. I had someone set this up for me. Is there an easy way to remove that code for the unexperienced??

All quotes are in US dollars, ex works unless stated otherwise, and subject to prior sale.
Wire Transfer Order- US$50.00 minimum-(does not include freight charges)
Wire Transfer Order less than US$250.00 - additional US$25.00 transfer fee
Credit Card Order US$25.00 minimum-(does not include freight charges) All terms and conditions of sales and purchases can be found at www.universalbearing.com

Universal Bearing Co.
8521 Lyndon
Detroit, MI 48238
Fax 313 933 8888
Tel 313 933 1100
email  ***@***.*** ***@***.***
"ON AIRCRAFT USE NOT PERMITTED"
Neither this information block, the typed name of the sender, or anything else
in this message is intended to constitute an electronic signature for purposes of the
Uniform Electronic Transactions Act or the Electronic Signatures in Global and National
Commerce Act ("E-Sign") unless a specific statement to the contrary is included in this message

jgyates, 2024-05-17:
if you actually have a Nexus you are fine, you just will not be able to plug that system into another controller and have it auto-detect a different controller.

jcherney, 2024-05-17:
Ok great. Thanks again.

All quotes are in US dollars, ex works unless stated otherwise, and subject to prior sale.
Wire Transfer Order- US$50.00 minimum-(does not include freight charges)
Wire Transfer Order less than US$250.00 - additional US$25.00 transfer fee
Credit Card Order US$25.00 minimum-(does not include freight charges) All terms and conditions of sales and purchases can be found at www.universalbearing.com

Universal Bearing Co.
8521 Lyndon
Detroit, MI 48238
Fax 313 933 8888
Tel 313 933 1100
email  ***@***.*** ***@***.***
"ON AIRCRAFT USE NOT PERMITTED"
Neither this information block, the typed name of the sender, or anything else
in this message is intended to constitute an electronic signature for purposes of the
Uniform Electronic Transactions Act or the Electronic Signatures in Global and National
Commerce Act ("E-Sign") unless a specific statement to the contrary is included in this message