jgyates/genmon #1017: 48KW Liquid cooled Battery fault code : 2760

Mehrdadg, 2024-01-26:
Hi all 
not sure if this is the right place for my post so I apologize in advance if I am posting something that should not belong here:(

I have 48KW liquid cooled generac generator and recently a new battery was installed by my local authorized service provider.
 
Since then, I have been getting Battery fault code of 2760 which the only way to reset it is to power cycle the ECU.  Again, this was coincident with battery replacement.
I even changed to another battery thinking there was a faulty battery and still got the code so I know its not the battery.
 
Now sure what is going on but my suspicion is that either the ECU or the battery charger is now faulty.
 
Also I know the generator is constantly charging the battery and the terminal voltage during charging is ~14.2-14.3V and the ambient temperature I am dealing with is ~40-50F. Thanks to genmon SW, I have now have all the visibility to all these:)
 
Question: is the batter charger part of the Evolution ECU or a separate unit?
 
Also if I get a new ECU, is this something I can program and active myself or do I need have an authorized person to do this?
 
Finally, does anyone know what is the normal DC current on the + terminal of the 12V battery when in standby mode? 
 
Any help is greatly appreciated

jgyates, 2024-01-26:
I believe the battery charger is a separate unit from the controller on the Evo but not on the nexus. I am not a generac service expert so others can correct me if I am mistaken.

If you replace the controller you don't need to activate the replacement unit as the serial number is not present in replacement units. Regarding replacing it yourself I will let others weigh in on that as I have not done that myself. I do know that you have to set the fuel type, voltage and parameter code but other than that I don't know what else is involved. It is probably covered in the Evo Liquid Cooled maintenance manual.

jgyates, 2024-01-27:
Also, you can read some info in the code comments in this link related to the battery charger behavior: 
https://github.com/jgyates/genmon/blob/19c11fec604ff3183175ce4115a13840cf08a7db/genmonlib/generac_evolution.py#L4258

I am going to close this thread but feel free to post any follow up to this thread.

closed 2024-01-27 by jgyates