jgyates/genmon #1018: new feature request

Mehrdadg, 2024-01-27:
I use tankutility add-on and one feature request is to have the time stamp of the last reading to show in the dial window.

I find the wifi on these tankutility monitor units very flaky and often it stops its daily reading so it would be good knowing the displayed value on the main page of the genmon is current or not.

Thanks

jgyates, 2024-01-27:
sounds like a good request. I will probably implement this as an email notice that will be sent after x period of time without an update. Tank utility has a delay in the reading time vs the time it shows up via the API of about 1.5 days. I have a propane tank that I monitor (not connected to the generator) and I have it send a message if the reading time is greater than 2 days. I can also optionally have it send an email if the battery is low, but my unit shows the battery is low all the time (even with a new battery of the required type).

Mehrdadg, 2024-01-27:
that is great and thanks for considering this and very much look forward to this feature.
My tankutility sensor is very unreliable and I also have the same issue with the battery monitor.
regards

jgyates, 2024-01-27:
I will update this thread when this is complete.

closed 2024-01-27 by jgyates

jgyates, 2024-01-27:
I have added this feature: https://github.com/jgyates/genmon/blob/master/changelog.md

You can enable this after you update the software, then go to the Add On page. From there you can enable the check for battery and missed readings. Let me know if you have any issues or questions.