jgyates/genmon #1035: Pushover feature requests - button to test, log successful sends

skipfire, 2024-02-05:
I'm running v1.19.01 and my exercise today did not have any pushover messages show on my device (emails did, and I have pushovers from my transfer test on Feb 1).  While investigating I did not find any entries in the genpushover.log file (just some from August) so it would be great if it logged successful sends as well as errors so that I can see that GenMon sent the command and did not receive an error (which is what I suspect) vs GenMon just didn't try to send.  It would also be great if there was a button on the add-on (like there is in the email config) so that I could have GenMon send a test-pushover message which could help diagnose if it is an API change, a configuration issue, or something else.

skipfire, 2024-02-05:
My exercise went from :01 to :06 and I looked at my Pushover console at :15 after it had not shown any, at :17 both the Exercising and the Ready messages showed up, now I know they got sent, but I don't know if it was a delay in GenMon successfully sending or in Pushover processing them.  Logs of attempts and successes rather than just errors would help narrow that down.

jgyates, 2024-02-05:
you can edit /etc/genpushover.conf and add this line:

     debug=True

This will print a line on successful messages. Genmon will send a few messages (assuming you don't have any the categories disabled) one start up with the initial state. On startup is is normal to get 4 or 5 messages, depending on your generator features.

Sending a test message from the UI will require a lot of changes for a few reasons
1) the add is not loaded initially, unlike emails, which are sent from genmon.py which is always loaded if the web interface is running.
2) the add on programs do not support any input except from the config file or notifications from genmon.

I will think on adding this feature but at the moment it the debug functionally may be able to active the same goal.

One thing to check if you messages are getting delayed:

The following optional settings in the config file: max_retry_time (default 600), default_wait 9default 120 and minimum_wait_between_messages (default 0) dictate how long wait conditions are for the add on.

max_retry_time - max time in seconds to attempt to retry (default 600 seconds = 10 min)
default_wait - time to wait after an error to resend the message (default 120 seconds)
minimum_wait_between_messages - minimum wait between messages sent (default is zero) most messages do not need this but it was added for the Signal callmebot addon as there was a need for it with this type of communication.

If any of these changed from the default you may want to look at changing it back.

Let me know if you have any quetsions.