jgyates/genmon #1049: Outage Log Fuel used bug

zekyl314, 2024-02-08:
### Steps to Reproduce

When we lost power for a split second, log shows 0:00:00 seconds, we show an odd amount of fuel consumed.

### Expected Result

I expect when we have a blip of an outage, no fuel is consumed.  I don't even think the generate started up during these blips.

### Actual Result

2024-01-28 09:08:47, Duration: 0:00:00, Estimated Fuel: 17.41 gal
2022-06-13 17:20:19, Duration: 0:00:00, Estimated Fuel: 14.51 gal

### Version Info

Current Version: V1.18.17

### Logs

Clicked on button to send you the logs.

jgyates, 2024-02-08:
You need to be on the latest version. This version is from 4/23. The latest version is V1.19.01. Let me know if you still have the issue with the latest and we can go from there.

closed 2024-02-08 by jgyates

zekyl314, 2024-03-12:
Hi @jgyates I had the issue again this evening.  Please look over info and let me know what you think.
03-12-2024 17:19:26, Duration: 0:00:00, Estimated Fuel: 14.73 gal

zekyl314, 2024-03-12:
Logs have been sent over.

jgyates, 2024-03-12:
I got your logs. Also send these files:

 power log at /etc/genmon/kwlog.txt
outage log at /etc/genmon/outage.txt

You can attach the file to this thread.

zekyl314, 2024-03-13:
Attached.

[kwlog.txt](https://github.com/jgyates/genmon/files/14580057/kwlog.txt)
[outage.txt](https://github.com/jgyates/genmon/files/14580058/outage.txt)

jgyates, 2024-03-13:
would you happen to have the body of the email notice for the outage? It should have something like "Utility Power Restored at <time>. Duration <duration>

jgyates, 2024-03-13:
Never mind, I figured it out. I will let you know when I post an update. I have another update in that same file so it may be a in the next couple of days that I post an update.

jgyates, 2024-03-13:
I checked in the update. Let me know if you have any issues or questions.

lwbeam, 2025-03-31:
Yesterday we experienced an ice storm that resulted in several outages over a 14 hour period. Although some of the outages were brief, likely due to break-before-make switching, several latest for hours. As you can see from the screenshot, for the **Estimated Fuel** used during the longer outages, Genmon appears to report the _accumulated_ value for the day. Is this expected behaviour?

[image]

 I've sent you my logs, and will happily provide any other information you may require. Many thanks.

jgyates, 2025-03-31:
Hi @lwbeam 

I think what may be happening is this:

On Evolution/Nexus models, genmon determine if an outage is active by looking at the utility voltage  and if it less than  the threshold voltage (denotes the start of an outage) or greater than the pickup voltage (denotes the end of an outage). 

The controller also has internal timers that will keep the generator running (cool down mode) once power is restored and another timer that denotes how long the utility voltage must be greater or lower than the pickup or threshold voltage (Start up delay) in order to start the generator.

This is my best guess at the moment: since genmon does not factor in the cool down mode or start up delay, genmon will consider multiple outages if the utility voltage bounces within the start up delay time. The power log (which is where the fule usage is derived) looks for zero power to denote a stopped generator. So genmon may be considering multiple outages but the power log is only logging one. I have not validated this guess yet.

To help with testing, can you send the following files (they are not inclued when you send your logs): /etc/genmon/kwlog.txt and /etc/genmon/outage.txt

lwbeam, 2025-03-31:
No certain if this helps, but I did notice that the Utility Voltage was not indicating zero (2 or 4V) during the first two long outages. Don't know about the last two long ones.

[kwlog.txt](https://github.com/user-attachments/files/19542093/kwlog.txt)
[outage.txt](https://github.com/user-attachments/files/19542095/outage.txt)

Greatly appreciate your time and effort on this.

jgyates, 2025-04-01:
Do you have a value set on the Advanced Settings page for "Outage Notice Delay", "Minimum Outage Duration" or "Outage Recurring Notice Interval" other than zero?

lwbeam, 2025-04-01:
Outage Notice Delay - 0
Minimum Outage Duration - 1
Outage Recurring Notice Interval - 0

jgyates, 2025-04-10:
I wanted to give you an update: I have look at this however I don't have a smoking gun at this point. I did review the code and did some testing and ended up doing some optimizations. I am going to keep an eye on this and brainstorm some testing methods that may allow me to duplicate this behavior.