jgyates/genmon #1051: Switch position under "Maintenance" not showing the status

nrm94ee330, 2024-02-12:
### Steps to Reproduce

[image]

[image]

Since recently, the switch position color has gone away.. the clicking seems to be working though, as it shows correctly under "Status page correctly"

### Expected Result

The button used to show the colored button based on the switch status

### Actual Result

The buttons are always the same color and do not change based on the switch status

### Version Info

Genmon Version V1.19.01

jgyates, 2024-02-12:
what is your generator model? It is and evolution liquid cooled or a Nexus liquid cooled? It helps if I can test simulating the same model. Not all models have this setting on the Maintenance page.

nrm94ee330, 2024-02-12:
> what is your generator model? It is and evolution liquid cooled or a Nexus liquid cooled? It helps if I can test simulating the same model. Not all models have this setting on the Maintenance page.

Model : RG03224ANAX
Controller : Evolution, Liquid Cooled

jgyates, 2024-02-12:
Thanks for finding this. I checked in an update that corrects this issue. The version number has not changed so just update from the about page and refresh your browser.

Let me know if you have any issues.

closed 2024-02-12 by jgyates

nrm94ee330, 2024-02-12:
awesome!  thank you

nrm94ee330, 2024-02-13:
[image]
 Is this changed now, does it not show green color button ?

jgyates, 2024-02-13:
No you are not crazy. I downloaded a few versions back and found a build that had the disabled buttons colored green,
[image]

Not sure where or why this got dropped but I will look into it. There is the possibility that this was intentional, I honestly don't remember. I will update this thread when I have more info.

jgyates, 2024-02-14:
I checked in an update that corrects this issue and reverts it back to the the way it was originally. Thanks for pointing this out. Let me know if you have any issues.

nrm94ee330, 2024-02-14:
Yup, just updated.. worked perfectly..Thank you very much. BTW, I started seeing actual power usage in the recent updates not just 0 and max value.. Thank you for an amazing project!