jgyates/genmon #1052: Platform stats incorrectly showing CPU Throttling and undervoltage

nrm94ee330, 2024-02-12:
### Steps to Reproduce

[image]
I used to get yellow lightning symbol when remoted into raspberry pi, and a corresponding Pi undervoltage under platform stats under "Monitor"
I replaced the power supply, to a 5v 3A and the yellow lightning symbol has gone away from the PI desktop,  but the platform stats still shows CPU throttling and Pi Undervoltage

### Expected Result

platform stats to match the raspberry pi 3 B+ status correctly.

### Actual Result

platform stats do not match the raspberry pi 3 B+ status correctly.

### Version Info

Genmon Version V1.19.01

### Logs

[mymodbus.log](https://github.com/jgyates/genmon/files/14249632/mymodbus.log)
[myserial.log](https://github.com/jgyates/genmon/files/14249633/myserial.log)
[gengpioin.log](https://github.com/jgyates/genmon/files/14249634/gengpioin.log)
[genmon.log](https://github.com/jgyates/genmon/files/14249635/genmon.log)

jgyates, 2024-02-12:
you did not submit your logs on the about page so I don't know what version operating system you are running so the information I give may not be complete.

the information on  CPU throttling and undervoltage comes from one of these two commands:

     /usr/bin/vcgencmd get_throttled

or 

     cat /sys/devices/platform/soc/soc:firmware/get_throttled

If you have a non zero value in either of these commands then you are **currently** or **have had** CPU throttling, under-voltage or frequency cap.

The lightening bolt is in OS the GUI denotes **currently active** situations. The monitor page shows both active conditions and conditions that have occurred since the last boot. All of these are derived from the above commands.

Non zero values of the above commands are interpreted by bit positions:

| Bit | Meaning |
|:---:|---------|
| 0 | Under-voltage detected |
| 1 | Arm frequency capped |
| 2 | Currently throttled |
| 3 | Soft temperature limit active |
| 16 | Under-voltage has occurred |
| 17 | Arm frequency capped has occurred |
| 18 | Throttling has occurred |
| 19 | Soft temperature limit has occurred

Note the differences in bit positions between "detected" and "has occurred". If you are reading non zero values from the above commands then this is the source of the confusion.

Let me know if you have any other questions.

closed 2024-02-12 by jgyates

nrm94ee330, 2024-02-12:
I did add the logs..
here is the snapshot of the two commands.
[image]

But what you makes sense.. It is possible that, the throttling occurs at bootup.. and then just remains there!.
is there a way to manually reset the value of /usr/bin/vcgencmd get_throttled to 0X00000 ?

jgyates, 2024-02-12:
When the log that are sent from the about page rather than attaching some of the logs it does not include the OS version. The additional data is sent is listed here (adjust your IP address fro you pi to see the output of the data sent):

http://192.168.1.10:8000/cmd/support_data_json

It also sends all of the logs, for example you did not send the log for the web server (genserv.log) or the program loader (genloader.log). No need for them at this point.

Yes, the flags are active from boot. This thread has info on how to reset the "sticky" bits: https://forums.raspberrypi.com/viewtopic.php?t=241736

nrm94ee330, 2024-02-12:
cool.. thanks.. that was helpful..