jgyates/genmon #1058: Genmqtt fails to start; missing "callback_api_version"

jwelter1971, 2024-02-18:
### Steps to Reproduce

1) Install Genmon
2) Setup the plugin to point to Mosquitto (2.x)
3) Mosquitto never sees messages from Genmon
4) Genmqtt.log contains the error message (see below in actual results)
5) Other clients can connect and publish to Mosquitto
6) Mqtt browser can browse Mosquitto messages

### Expected Result

MQTT to receive the status from Genmon

### Actual Result

Genmqtt log contains:

2024-02-18 14:28:54,192 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
2024-02-18 14:29:44,233 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
2024-02-18 14:30:34,321 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
2024-02-18 14:33:39,309 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
2024-02-18 14:41:32,463 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
2024-02-18 14:44:57,467 : Error in MyMQTT init: Client.__init__() missing 1 required positional argument: 'callback_api_version' : genmqtt.py:496
root@618

### Version Info

Genmon 1.19.01 (docker container)
Mosquitto 2.0.18

jgyates, 2024-02-18:
This issue was fixed a few days ago: https://github.com/jgyates/genmon/issues/1054

1) update genmon from the about page
2) run the following script to update your paho-mqtt python library

         ./genmonmaint.sh -r

3) restart genmon

        ./startgenmon.sh restart

jgyates, 2024-02-18:
let me know if you still have any issues or questions.

closed 2024-02-18 by jgyates

jwelter1971, 2024-02-18:
Unfortunately as I am running genmon in docker and the about menu upgrade to 1.9.02 doesn’t work (it seems).

I can manually install paho 1.6 with pip but it gets pushed out with the genmonmaint script.

John

jgyates, 2024-02-18:
Did you install via the recommended method? This page has the method for installing:  https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Note the command in the instructions: 

      git clone https://github.com/jgyates/genmon.git

If you do not install this way, then the update will not work. The pi also needs internet access to update to the latest.

You can correct his by doing the following:

1) log into your docker image to the console
2) follow the steps in the install process on this page: 

     https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

NOTE: when the install asks about copying the config files, type 'n' for no and you will keep your existing config files. type 'y' to overwrite your existing config.

3) restart genmon

       cd genmon
      ./startgenmon.sh restart

This should fix your inablity to update the software. If may also fix your issue with paho-mqtt. If it does not then you can run the steps in the first reply (./genmonmaint.sh -r) and that will fix your paho-mqtt issue.

jwelter1971, 2024-02-18:
The docker image is pretty minimal and doesn’t have GIT installed.  As it auto-publishes weekly and due in 2 days will just wait for that as it should pickup the 1.19.02 release; and then I hope your patch works.

John

djstroh, 2024-02-18:
Patch works, I just disabled my docker image of genmon, deleted the
installed docker image, and reinstalled genmon in docker. That did a clone
of the latest code.

I tried to find a way to load git into my docker genmon image. I did not
find a method that worked.

jwelter1971, 2024-02-19:
So today I took the time to build my own docker container with Genmon 1.19.02.

I can confirm Paho 1.61 is in use after the patch provided above.

Everything works with Mosquitto 2.19 once applied.

Thanks!

jgyates, 2024-02-19:
Thanks for the update

djstroh, 2024-02-19:
Would you be willing to share your docker file with me?

jwelter1971, 2024-02-19:
> Would you be willing to share your docker file with me?

docker pull ghcr.io/jwelter1971/my-genmon:main

Docker compose/stack snippet:

```
genmon:
    container_name: genmon
    image: ghcr.io/jwelter1971/my-genmon:main    
    volumes:
      - /var/lib/docker/volumes/genmon_config:/etc/genmon:rw
      - /etc/localtime:/etc/localtime:ro
    restart: always
    ports:
      - 8899:8000      
```