jgyates/genmon #1065: Get either an email or a pushover alert if Genpi unit goes offline

grsthegreat, 2024-02-28:
I have occasionally had my genmon unit go offline and did not realize it until i try and log into unit. Is there any way to have the unit tell me it cant communicate…..then again, rereading this request i dont see how it can communicate if its offline. Maybe i answered my own question.

skipfire, 2024-02-28:
That would require an external system.  If cloud based it would not know the difference between Pi offline and Internet down though. I have a second Pi that runs my home automation and I use it to monitor and alert me if GenMon is unavailable.

grsthegreat, 2024-02-28:
Ok, thank you. I was only wondering because i have a device called Yolink that monitors refrigerator and freezers, water leaks and greenhouse. It alerts me if it loses communication anywhere in system. 

But thanks fir response

jgyates, 2024-02-28:
I use Nagios Core. It will monitor everything from computers to network equipment. On each system it can monitor multiple things. Below is what Nagios is monitoring just on the pi that runs genmon:

[image]

This lets me know if the pi goes down, the web interface is not responding, genmon is not loaded or my fuel is low. Some of these are done with built in Nagios checks (web interface is OK), others use SNMP via the genmon SNMP add on (low fuel). The [check_monitor_system.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappscheck_monitor_systempy) program is a Nagios plug in that will check to make sure genmon is running and the health of genmon is good.

Note that Nagios Core is the [free open source tool](https://github.com/NagiosEnterprises/nagioscore) and it does not have a graphical setup so you have to define your network resources via configuration files so you probably need some linux knowledge or you can use one of the open source front ends for Nagios.

closed 2024-02-28 by jgyates