jgyates/genmon #1066: Genmon exercise runs weekly even when I have it set for bi-weekly.

sfaught22, 2024-03-01:
### Steps to Reproduce

I can reproduce the symptoms.

### Expected Result

I have my Genmon set from the add-ons section to use the enhanced exercise and I have it set to bi-weekly.  For three weeks now it has run every week.  As I read through the issue section, I wonder if the generator is on a different exercise setting and I have the Genmon on the opposite week so it is running every week?  I changed enhanced exercise to monthly to see if this changes but I am waiting for 30 days to go by.  

### Actual Result

Exercise weekly even when set to bi-weekly in enhanced exercise.  

### Version Info

Generator Monitor Stats :
        Monitor Health : OK
        Controller : Evolution 2.0, Air Cooled
        Run time : Generator Monitor running for 2:05:22.
        Power log file size : 0.00 MB of 15.00 MB
        Generator Monitor Version : V1.19.02
        Update Available : No

jgyates, 2024-03-01:
what is the output of the file /etc/genmon/genexercise.conf

This file contains your settings as well as the last exercise time (this is who genmon knows it has been two weeks). If this file is overwritten (for example if you run the install script and copy the config files or restore a backup of genmon config files) then the exercise add on will not be able to determine the last exercise time.

There should be an entry in the /etc/genmon/genexercise.conf file named "last_exercise" that contains the last exericse time and date. This is used to calculate the next exercise time in biweekly mode.

sfaught22, 2024-03-01:
I am sorry, I am not that advance on this system.  Do I need to go somewhere to look for a specific file?  If you could give me a little more information I would so appreciate it.

jgyates, 2024-03-01:
I got the log files you submitted and I loaded your registers contained in the logs in my simulator. On the logs page of the web interface you can see the run log:
[image]

this shows that the last three run cycles were

1) a Remote Start on 2/15
2) an exercise cycle on 2/22 
3) another Remote start on 2/29

The exercise add on does not replace the built in exercise cycle, it adds a 2nd exercise cycle. the add on starts the generator with the remote start commands so numbers 1 and 3 above are from the add on and number 2 is from your built in exercise cycle. Also, it shows that the two remote starts took place two weeks apart, which is the correct operation. 

The wiki page for the the enhanced exercise add on explains that the add on does not replace the built in exercise, but it adds to it. You can not disable the built in add on so if you just want a bi weekly exercise then I would not use the exercise add on, but just set the controller exercise cycle to bi weekly. 

To do that:
1)  first disable the exercie add on, save your settings
2) then set "Enhanced Exercise Time" on the settings page, save your settings at the bottom
3) then go the the Maintenance page and set the exercise cycle to two weeks.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genexercisepy-optional

If you want some other exercise cycle other than weekly, biweekly or monthly then you can use the exercise addon in addition to the built in controller initiated exercise.

they way it is setup now you have a controller based exercise set for biweekly and an additional genmon controlled exercise cycle also at two week but they overlap so it results in an exercise cycle every week. Your run log back this up.

I am going to close this issue as it appears the software is working as expected but feel free to post any follow up questions to this thread.

closed 2024-03-01 by jgyates

sfaught22, 2024-03-01:
I now understand how the system works and I can see what was happening in the log sections.  I am trying to follow your directions 1, 2 and 3.  I have done #1, disable the exercise add on and saved the settings.  I have also done #2 Set "Enhanced Exercise Time" and saved it at the bottom.  I got lost on #3 go to the Maintenance page and set the exercise cycle to two weeks.  When I go to Maintenance page I don't see any place to set the cycle.  I ultimately want "Monthly" if I hit the advance button in the upper right it shows me the logs, I then hit the Advance settings but I still don't see how to set the cycle to monthly?  If you can give me just a bit more info. here I would so appreciate it.  By the way, this system is awesome!!!

jgyates, 2024-03-01:
if you completed number 2 and saved your settings at the bottom of the page, once the software restarts you should go to the Maintanece page and you will be able to set the exercise time:

[image]

if you did not complete number 2 then it will not show an option to set daily, weekly, monthly options. you will only see an option to set the day and time like this:
[image]

No settings are needed on the advanced settings page for exercise cycles

jgyates, 2024-03-01:
under the Maintenance page, change the setting under "Generator Exercise Time:" from "biweekly" to "Monthly"

sfaught22, 2024-03-01:
I am so sorry I feel so stupid!!  I must be missing something, please forgive me.  I have done 1 through 3.  I have turn off everything at the addon page, just to be safe.  I then went to the Setting page and the enhanced was already on.  

[image]

I then went to the Maintenance page and I do not have any of the "Generator Exercise Time".

[image]

I thought, well maybe I need to turn off the Enhanced Exercise and save it.  Then after it restarts, turn on the Enhanced Exercise and save it.  Then it will restart and I should see it.  

[image]

I still do not see the Generator Exercise Times on my screen.  Please just a little more help and I will have this figured out.  Thanks again!!!!

jgyates, 2024-03-01:
You have "Smart Transfer Switch" enabled on the Settings page. Does your transfer switch have an electronic read out? If it does not then you should not enable this setting as it indicates that the transfer switch, not the controller, will handle the exercise cycle and genmon will not display the exercise settings because it assumes that the "Smart Transfer Switch" will do this. The vast majority of transfer switches in use with Generac whole home back up generators use "dumb" transfwer switches. My guess is that you have this type of switch and you should disable the "Smart Transfer Switch" setting on the Settings page and save your settings at the bottom of the page. Once the software restarts you should see the exercise options on the Maintenance page.

If you hold your mouse over the setting on the Setting page it will give you a description of what that setting does. This is the documented description of the "Smart Transfer Switch" setting: 

       "If enabled a smart transfer switch is utilized and the generator controller will not be responsible for starting and stopping or or weekly exercise functions and these items will be disabled in the web interface."

sfaught22, 2024-03-01:
That did it!!!  Thank you!!!!  I read that bit about the transfer switch, over and over, I just had no idea I had the "dumb" switch.  Thanks again for all your help, I think your Genmon system is awesome!!!!

jgyates, 2024-03-01:
Great, I am glad it is working for you now.