jgyates/genmon #1071: Multiple Notifications Pre Email Address

knemo20000, 2024-03-06:
### Steps to Reproduce

1. Add multiple email addresses under Notifications
2. Start the generator, either from GenMon or manually
3. Wait a short while, doesn't matter how long, and stop the generator

### Expected Result

One alert email for the start event and one alert email for the stop event to each of the defined email addresses.

### Actual Result

The number of alert emails received for the stop and start events matches the number of email addresses defined (3 addresses = 3 start alerts and 3 stop alerts to each email address defined)

### Version Info

V1.19.02 (also noticed this behavior on V1.19.01)

jgyates, 2024-03-06:
genmon sends one email but to multiple recipients in the "to" field when it sends an email for a single notification.  If you get multiple responses it may be because you have your email addresses forwarded from another account. 

If what you were saying is actually happening then you would receive 9 email (3 on each account, assuming none of them were forwarded).  If you are only receiving 3 emails in the inbox of one account then you likely have forwarding setup on the the other 2 accounts.  You should be able to test this by sending one email from the account you have setup for outbound email in genmon to the 3 email addresses you have setup on the notification page. 

Let me know if you have any questions or if you do not see the expected results.

closed 2024-03-06 by jgyates

knemo20000, 2024-03-06:
That is what I am saying - If I have 3 email accounts set up, I receive a total of 9 emails, 3 on each account.  If I only have 2 accounts, then 4 emails are received, 2 for each account.  Forwarding is not enabled on any of these email accounts.

jgyates, 2024-03-06:
What are you using for your outbound mail server?

jgyates, 2024-03-06:
I notice that you mentioned the multiple emails are sent from the start and stop of the generator. Do you also see multiple email addresses when you start genmon. Genmon normally sends 2 emails when it starts, one is a statement that genmon is starting, the other is for the initial state of the generaror.

You also did not send your logs or mentioned the type of generator. If you only get multiple emails on start and stop then your generator type may come into play.

I will need to see your log files. Also a screen cap of your email settings and notification page settings. You can block out the user portion of the email address when you post.

Based on the little info that I have my guess is that you are using an outbound (SMTP) mail server this is either a relay or setup incorrectly. The way the mail protocols works, the sending client is not responsible for distribution, the outbound SMTP server is responsible for this task. The client only sends the outbound email to one place, the outbound SMTP server. The sending client only has to populate the "to" or "bcc" field with multiple comma separated addresses. If the multiple recipients addresses reside on different servers (i.e. on is gmail, one is yahoo, etc) then the outbound SMTP server's (sending server) job is to send one email to each IMAP or POP server (receiving server).

Other things to check is that you are not running multipl instances of genmon from the same configuration files. Unless you have modified settings to do this I doubt this is the case as genmon will not allow multiple copies unless you explicitly tell it to do so. You can check that you only have one entry in crontab like this:

[image]

you should only have one line with @reboot at the start of the line

The code in genmon for sending mail is pretty stable as the only changes made in the last few years have been to add more error reporting to the log files.

knemo20000, 2024-03-07:
Thank you for your response.  In troubleshooting this issue, I decided to start over with a fresh install.  Needless to say, that corrected the problem, so I must've had something mis-configured.  I apologize for jumping the gun on this and appreciate your time.  Thank you for putting together this awesome app.

Screenshot.2024-03-06.at.8.16.45.AM.png (view on web)<https://github.com/jgyates/genmon/assets/4513286/3a0ad8ae-65c1-4bc9-bca1-ea8ae2e17e6b>

jgyates, 2024-03-07:
For future readers of this thread here is a link on how to setup gmail with genmon:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#setup-gmail-for-use-with-genmon

A local mail relay on the pi is not required, but can be used if you have some need for it.