jgyates/genmon #1095: Login page does not find jquery

skipfire, 2024-03-30:
### Steps to Reproduce

1. Open developer tools
2. Go to the login page without being logged in
3. Developer tools shows jquery not found and a couple other errors because of that.

### Expected Result

No errors

### Actual Result

jquery is not found and errors are logged.  It does not currently block login, however I don't know if it is impacting any other behaviors such as error messages when there's a bad login (I don't see any error, it just blanks the fields).
[image]

### Version Info

v1.19.02

### Logs

Submitted through the about page.

jgyates, 2024-03-30:
I checked in an update that should fix the issue. Let me know if you have any issues. I was able to duplicate and this fixed it for me.

closed 2024-03-30 by jgyates

skipfire, 2024-03-30:
Thanks, it did.