jgyates/genmon #1106: Enhanced Exercise w/v1.19.03

gm58, 2024-04-21:
### Steps to Reproduce

Just waited for weekly exercise to start.

### Expected Result

After noticing the weekly exercise didn't start on time I looked at the Enhanced Exercise settings on the Add-Ons page and noticed they had been changed.  After seeing the settings were changed I tried to change the settings on the Add-Ons page before the weekly exercise time set in the controller but I didn't have enough time.  Instead I just put the generator in Manual after the exercise started and let it run for 15 minutes.  I'm assuming it was the update that caused the changes because I don't remember making any recent changes to Genmon.  See attached screenshots of both April 14 and April 21 settings.

[image]
[image]

### Actual Result

Setup of Enhanced Exercise on April 14.  I believe my settings were changed during the update to V1.19.03 on April 20.  

### Version Info

V1.19.03

### Logs

Logs and Registers sent

jgyates, 2024-04-21:
I am assuming you ran the upgrade by clicking on the update button on the About page. 

I received you logs. It does look like from the picture above that the settings on the right are the default so your config file being overwritting may be what happened. I don't think that this occurred during the upgrade. There are two ways that config files get copied within the genmon source. The first is from the install script and the second is from genloader. The install scrip copies occur if you run the install scrip (genmonmain.sh -i). The genloader copy occurs if the config file is not detected in /etc/genmon folder. If it is not detected and genmon does the copy, then the copy is logged in /var/log/genloader.log. You log file does not contain any indication that this occurred.

If you ran the install script in the past while genmon was running and did not restart genmon, then the config file would have been copied, but you would not see the change in your settings until genmon was restarted (during the upgrade) since the software add on programs only reads the config file on startup.

If you want to try this to validate you can: Try upgrading again from the about page. The exact same step should be taken by the software.

I am going to close this thread as I don't have enough info to duplicate or confirm what you are describing but feel free to post any follow up data or questions to this thread.

closed 2024-04-21 by jgyates

gm58, 2024-04-21:
Yes, I always upgrade from the About page

I did validate by performing another upgrade and just as you mentioned, everything is fine.

I must have done something when adding Enhanced Exercise back on April 14 that caused the issue.  Thanks for information you provided.

gm58, 2024-04-28:
Same results as last week, Enhanced Exercise would not start prior to weekly exercise.

Settings prior to weekly test;
[image]

Logs/Registers sent

jgyates, 2024-04-28:
First validate that you can actually remotely start your generatror by starting and stoping the generaror form the Maintenance page. If remote commands are not working for you then the Enhanced Exercise add on will not work.

If that works, then send a screen shot of your status page showing the generator time and the pi time. It is at the bottom of the page. This will show the difference in time between the controller and the pi.

Based on the screen shot above, you are using the Enhanced Exercise to start based on the pi's time (not the controller's time), but your controllers built in weekly exercise is based on the controller time. Since your controller exercise is at 11 (based on your registers you submitted) and your add on exercise is at 10:55, if you time is not within 4 minutes (5 minutes minus the 60 seconds possible difference since the controller does not export seconds, only hours and minutes) difference the you may solve the issue by syncing the time (set the time on the maintenance page) or select the sync time option on the settings page. If the time is not in sync then your controller may be attempting to start the built in exercise cycle first.

Alternatively you can enable the "Use Generator Time" option on the add on. Then the add on will start based on the generator time, not the pi time.

If you have validated all of that and it still does not work then you can add this line to the file /etc/genmon/genexercise.conf to get addtional data in the /var/log/genexercise.log log file

     debug=True

You will not want to leave that option set to True for a long time as this will increase the writes to the log file and over time will decrease the life of your SD card, however this could provide more info as to why your exercise is not occurring.

gm58, 2024-04-28:
Thank you for the response.

Although the Enhanced Exercise worked April 14 without issue
[image]

I went ahead and tried it from the Maintenance page as you suggested.

Although I didn't mention it in today's post, I checked the time prior to the test.  Here is a current screenshot

[image]

Thanks

gm58, 2024-04-28:
Also, when should I edit the genexercise file, prior to next weeks weekly exercise?

gm58, 2024-04-28:
Sounds good, hopefully later today.  Just send you the logs afterwards?

jgyates, 2024-04-28:
yes please

gm58, 2024-04-30:
I changed the weekly exercise for today and added the debug=True but I also added the IP address and everything went as planned.

I remembered the very first time I tried Enhanced Exercise I added my Pi's IP address and everything worked as planned.  If this could be the issue, can you add a line to the Add-Ons page for adding the IP address?

Logs/Registers sent

jgyates, 2024-04-30:
that is great this it is working for your. Can you elaborate on what you mean by "I added my Pi's IP address"?

Below is the log from your /var/log/genexercise.log file:

2024-04-30 10:32:32,608 : Execise: Normal, Weekly at 10:55 on Tuesday for 15.0 min. Warmup: 0.0
2024-04-30 10:32:32,610 : Debug Enabled
2024-04-30 10:55:00,186 : Starting normal exercise cycle.
2024-04-30 10:55:00,192 : Last Exercise Cycle: Tuesday April 30, 2024 10:55:00
2024-04-30 11:10:00,248 : Stopping exercise cycle.

Also, don't forget to remove the debug=True line from /etc/genmon/genexercise.conf

gm58, 2024-04-30:
I added it in the genexercise.conf where you have an Optional **monitor_address =**

jgyates, 2024-04-30:
The monitor_address is used for running the add on on a different system than genmon. If you previously edited the genexercise.conf file manually (not recommended), I am guessing that you may have some whitespace character after the equal sign for the monitor_address. If you have *any* value after the equal, it will be used as the IP address. If you have no value then the software will use localhost. Adding monitor_address to a system that is running the add on program on the same system as genmon is not needed and not recommended, however I can add a check to remove any additional white space characters so if what I think I happening will not happen again.

gm58, 2024-04-30:
I see what you're saying but it's just weird that the only 2 times that Enhanced Exercise has worked for me is when I had an IP address entered.  I can remove the address and see what happens next time.  Appreciate your time!

jgyates, 2024-04-30:
The issue I am suggesting is not what happens when you add it, but what happens when you remove it. If you edited the entry and had a space on the line (for not entry) then it would use space as the IP address.  If you added your IP it would work, but if your IP address changed it would stop working. I just checked in a change that will remove whitespace characters from the monitor_address parameter, so if you want to update your software to the latest (version number did not change) and remove the IP address you can try it out.

gm58, 2024-04-30:
Yes, I totally understand what you meant about the Whitespace.  Possible that was the issue but I'm pretty sure I copied a fresh copy of the file when I started having issues.  Time will tell.

jgyates, 2024-04-30:
FYI, I just made one more tweak to the code to make the error checking a bit better so you can update now to get the very latest.

gm58, 2024-04-30:
Thanks for your help and updates!  I'll remove the IP address and report back next test, Sunday.

gm58, 2024-05-05:
Update;

Although I didn't test out your latest update for ignoring whitespace, I did remove the IP address from the monitor_address parameter and made sure I didn't leave any whitespace.  Today's test was perfect!

Thanks again!

gm58, 2024-06-04:
It's still a mystery what caused the issue with Enhanced Exercise last time but I had another issue this last week.

A few days prior to Sunday's weekly exercise I lost communications with Genmon's Pi.  I restarted the Pi and everything came back up.  Come Sunday, the Enhanced Exercise, as showed earlier in this thread, never happened.

1. I looked at genmon/conf/genercise.conf and sure enough it was back to the default settings even though the settings on the Add-Ons page showed my settings.

2. I turned Enhanced Exercise off on the Add-Ons page, clicked save, and after genmon restarted I turned it back on thinking it would change the genexercise.conf file to my settings.. nope, still default settings.

3. I deleted the genexercise.conf thinking it would restore the file after restarting.. nope, file was still missing from the folder.

4. I rebooted the Pi thinking surely that would put the file back.. nope, still missing.

5. I clicked upgrade to latest version on the About page and although that restored the file, but it was still the default settings.

6. I changed Exercise Time Minute from 55 to 50, clicked save, after genmon restarted I checked genexercise.conf.. no change, still default settings.

I just submitted my registers and logs if/when you have a chance to see what I'm doing wrong.

jgyates, 2024-06-04:
In your debugging you have started with a incorrect assumption. The file at ~genmon/conf/genexercise.conf should always have the default settings as this is the copy of the [default config file in the repository](https://github.com/jgyates/genmon/blob/master/conf/genexercise.conf). If any file in the ~genmon/conf folder is modified, it will be overwritten when a software update is performed. The file at /etc/genmon/genexercise.conf should have the settings you make via the web interface.

There are two copies of every config file. The original in ~genmon/conf and the working copy on /etc/genmon. All of your steps above appear to reference the default copy in ~genmon/conf, which should have the default values.

What you describe above is exactly what is expected if you delete the default config file (it will remain deleted until you perform a software update and then it will be replace with the default value).

If your enhanced exercise did not run then I would renable the debug=True value in /etc/genmon/genexercise.conf, restart genmon and leave the setting enabled for a few weeks. This will possibly give more details as to what you are seeing. I would also validate the time on the pi and generator are within a few minutes of each other.

One possible issue is this: I noticed from your screen captures above you do not have "Use Generator Time" enabled. You mentioned that you lost connectivity and rebooted. By default the pi syncs the pi system time to a NTP (Network Time Protocol) sever on the internet when it boots. If internet is not active on boot then the time on the pi can be off (until internet is restored). The pi will resync periodically but if the time is not in sync with the controller (within a couple of minutes) and you do not have "Use Generator Time" enabled then then the exercise add on will use the incorrect pi's time. If you enable "Use Generator Time" then the add on will use the generator controller time.

gm58, 2024-06-04:
Thanks for the explanation of the two sets of files and locations.  Makes a lot of sense why it wasn't changing lol

I will enable debug again and see what happens next test.

As far as the time; when I first installed and connected Genmon, I set the time via the Maintenance page and it's been spot on ever since so I never thought I needed to enable"Use Generator Time" (see attached)
[image]

gm58, 2024-06-09:
Here's hoping you can figure out why Enhanced Exercise didn't run again today during normal weekly exercise.

Logs/Registers just sent

Thanks!

jgyates, 2024-06-10:
Here is the data that I see from your log and register submissions. 

1) The add on appears to issue the commands from the controller. Here are the log entries:

2024-06-08 21:32:34,853 : Exercise: Normal, Weekly at 10:55 on Sunday for 15.0 min. Warmup: 0.0
2024-06-08 21:32:34,857 : Debug Enabled
2024-06-09 10:55:02,960 : Starting normal exercise cycle.
2024-06-09 10:55:02,967 : Last Exercise Cycle: Sunday June 09, 2024 10:55:02
2024-06-09 11:10:03,094 : Stopping exercise cycle.

2) you controller exercise built in cycle is set to overlap the add on. The built in cycle appears to be weekly at Sunday 11AM (5 minutes after the add on exercise cycle.
3) you appear to be using the a recent version of the Evo2 firmware (1.18)

To me this suggest that the behaviro of the controller has changed and the controlelr is preventing the add on exercise cycle. You can try a couple of things:

1) reboot the controller as described here: https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller
then see if that corrects the problem

2) try using the post-controller exercise type instead, this will run a 2nd exercise cycle after the first one is complete.

3) don't try to overlap your two exercise cycles, or make the gap between the two start times larger. The newer firmware may prevent external commands x number of minutes before an exercise cycle.

gm58, 2024-06-10:
Appreciate the response.  I've had the same firmware since I installed Genmon.  Extended Exercise has worked half a dozen times and has worked 4 times in a row until the Genmon Pi lost connection just recently.  Just doesn't make sense to me that all of a sudden the firmware would cause it not to run.

jgyates, 2024-06-11:
Unfortunately the firmware is sort of a black box. The logs indicate that the commands were sent with no error at the expected time. genmon issues the command to the firmware but it is not working as expected. If your firmware version has not changed the best suggestion is to reboot your controller to see if the problem persists. If it does not this will not be the first time that a reboot has resolved Evolution 2.0 issues.

firmware is like a lot of software in that it can have errors that do not manifest a behavior until it has execute for a period of time so this is not outside of the realm of possibilities.

gm58, 2024-06-16:
Update;
Although I didn't reboot the controller I did the following which hopefully will help you determine what might be wrong.

1. I few days ago I turned off Enhanced Exercise via the Add-Ons page.
2. Yesterday I changed the weekly exercise time via the Maintenance page from 11am to 10:45am
3. I then turned Enhanced Exercise back on via the Add-Ons page to start at 10:40am and run for     15 minutes.

and now it works like it should, again.

registers / logs sent

[image]

[image]

jgyates, 2024-06-16:
My guess is that this will not work either. I gave 3 suggestions and so far you have not tried any. 

You are trying to stop the built in exerciser cycle from running by the enhanced exercise at a time that overlaps the built in exercise and I don't think that will work. My guess is that after a period of time of not running the built in exercise cycles the controller stops accepting remote commands. This is why I suggested number 2 above (use the post controller exercise option in the Exercise Frequency menu). Using this option will  make a 2nd exercise cycle to run after the built in exercise cycle stops.

gm58, 2024-06-16:
Yes, that's exactly what I'm trying to do, overlap the built in exercise like atleast one other user is doing with no complaints that I've heard of.

Although I appreciate your suggestions; option 2 is not an option I'd be happy with.  I don't want the extra wear and tear on the starter.  I'll continue extending the weekly test like I've been doing before I do that.