jgyates/genmon #1110: Enhanced Exercise, fails to exicute after upgrade to V1.19.02

msteil4vse, 2024-04-30:
### Steps to Reproduce

Title indicates issue, Enhanced Exercise, fails to execute after upgrade to V1.19.02.  submitted log and registers to development. 

### Expected Result

anticipated Generator to run on enhanced schedule 

### Actual Result

submitted

### Version Info

I believe it worked correctly on the V1.19.01 release, but defiantly does not work on V1.19.02.  Consistent on 3 installations, and attempting to install V1.19.03 to see if issue is resolved.

### Logs

logs submitted

jgyates, 2024-04-30:
I have not received any log files today. Do you have outbound email setup and working?  

This issue appears to be related to a [recent issue](https://github.com/jgyates/genmon/issues/1106), but I don't have enough data to see any evidence of an error or software issue to date, although I am looking for more data from people who have experienced this issue (you are number 2)

Your logs will provide a first point of data. Also you will need to provide your Enhanced Exercise settings (a screen capture will work). Also, you can look at the thread linked above and I outline the steps to provide more detailed logging that may provide more insight.

msteil4vse, 2024-04-30:
[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/15167180/genmon_logs.tar.gz)

hopefully you will get the logs this way.

jgyates, 2024-04-30:
I got your logs. You issue is that your controller is in an alarm state. This is from the file /var/log/genexercise.log:

     Generator not in Ready state, exercise cycle not started: ALARM

If you clear the alarm, it should work as expected.

Let me know if you have any other issues.

closed 2024-04-30 by jgyates

msteil4vse, 2024-04-30:
[genmon_logs(1).tar.gz](https://github.com/jgyates/genmon/files/15167249/genmon_logs.1.tar.gz)

It occurred to me, the first set of logs I re-submitted were from the Genmon I was experimenting with and the 1.19.03 code.  I have set this upgraded unit to do an enhanced exercise at 10:45 local, and will advise if it runs and re-submit logs from that unit.  meanwhile, the attached logs are from a different unit, running the 1.19.02 code.  it should have run an enhanced exercise at 9:10 this AM and did not.

msteil4vse, 2024-04-30:
Alarm Cleared. awaiting test

msteil4vse, 2024-04-30:
Generator running 1.19.03 code ran enhanced test after 10:45 after clearing battery inspection alarm.  3 questions, is there a way to clear the alarm(s) from Genmon?  Without looking at log file is there a way to see in Genmon the alarm?  Finally, is there a SNMP query that can be done that would indicate an alarm such as this one?

jgyates, 2024-04-30:
Oddly enough, the 2nd set of logs has the same log entry:

2024-04-30 09:10:58,051 : Generator not in Ready state, exercise cycle not started: ALARM

You can look on the Maintenance page. On models that are support there is a button to clear the alarm condition. Note that this just sends a command to the controller. The controller may not clear all states in the same way (e.g. engine failure vs check battery).

Yes, you can see the alarm condition on the main page of the web interface. If you have email or other notifications enabled it will send you an email or message when an alarm occurs. Also, if you use the SNMP add on your can monitor all of the text on the status and maintenance pages. To detect alarms the switch state reads "System in Alarm". Details on the OIDs are here: https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensnmppy-optional

jgyates, 2024-04-30:
I would try swapping sensors on the tank and see if the problem tank is sensor specific or location specific. If it is sensor specific, check the battery level, if it is location specific, it may be a signal issue (weak signal) and you could possible look at anything obstructing the signal. Some bottom mounted tanks require spacers at the bottom of the tank footing to allow the bluetooth signal to escape.