jgyates/genmon #1113: GenMon website not responding after update

jjmacneil, 2024-05-05:
### Steps to Reproduce

After upgrading from 19.0.2 to 19.0.3 the GenMon website is not responding.

I rebooted the Pi3 to see if that would help, but the login page still won’t load and eventually times out.

I logged into the Pi and issued commands to install the latest available GenMon update and restarted after the install completed.

After the Pi restarted I was able to access the login page for a bit, but then started having trouble again so login page no longer loads and eventually times out.

GenMon was working fine on previous release 19.0.2

### Expected Result

I was expecting to be able to login to the GenMon interface to see the status of the generator.

### Actual Result

The GenMon login page does not appear and browser times out after a period of time.

### Version Info

19.0.3

jgyates, 2024-05-05:
@jjmacneil 

I have seen this issue before a few times and every time we dig down the issue is related one of two issues:

1) The router on the local network blocking local web servers.  https://github.com/jgyates/genmon/issues?q=label%3Asecurity
2) you are exposing the web interface to the outside internet and experiencing a denial of service attack (DoS) of someone trying to brute force your password. This causes the web interface to become unresponsive. https://github.com/jgyates/genmon/issues?q=+label%3ARouter+

As you can see there are a few of these issues that are exactly like what you are describing, including the symptom showing up after the upgrade and restarting fixing the issue and then it stops some time later.

There have been no changes to the web interface in quite some time so the symptom of this occuring after the restart is likely coincendiental or related to how your router is probing your network, or  (if you are expoing the web interface to the internet) possibly a timing issue related to a bot attempting remotely access your web interface.

My recommendation is:

1) if you are exposing the web interface to the outside internet, stop doing this as it it will be attacked. [Use a VPN](https://github.com/jgyates/genmon/issues/702)
2) If you are not exposing your web interface to the internet, then look at your router settings for any AI security or settings that block local web servers.

If you like you can send your log files from the web interface to see if there is anything wrong that shows up in the log files. If you can get the web interface to show up briefly, the send your log files from the about page.

Let me know if you have any questions.

jjmacneil, 2024-05-06:
@jgyates

Thanks for the quick response!

1. I confirmed my router is not blocking local web servers
2. The web interface is only available inside the local network

The interface is responsive at the moment so I'll monitor this week and re-open the issue if I experience more trouble.

Thanks!

jgyates, 2024-05-06:
ok. You can just post to this thread on any follow up. I received your logs. Nothing noteworthy in the logs.

closed 2024-05-06 by jgyates