jgyates/genmon #1123: Can you support some sort of auth token or basic authentication for API calls

ericsmith66, 2024-05-26:
I am making a plugin for HomeKit to allow the generator to appear as a switch in Apple Home. Unfortunately I have been unable to make API call into the genome server when SSL and Login ID/Password are enabled. I tried using Basic Auth and that did not work. 

It would be great if there was a way to authenticate via an API or use basic AUTH.

Thanks 
Eric Smith

skipfire, 2024-05-26:
I integrated by calling the login with a FormPost and then using the same connection to call the status json, it isn't API or basic auth but it does work.

ericsmith66, 2024-05-26:
Thanks , I will give it a shot

closed 2024-05-29 by jgyates

ericsmith66, 2024-06-01:
It worked fine. you need to copy the session cookie to authenticate between calls. There are a couple of redirects that happen on login. but other than that it worked fine. I have SPAN.io breaker panels which also has an API so I am wiring the two together.  Still early days.

jgyates, 2024-06-01:
Sounds good. Thanks for the update