jgyates/genmon #1139: Output Voltage graph nominal value not matching

rsockanc, 2024-06-22:
### Steps to Reproduce

[image]
[image]

### Expected Result

graph green center would be at 220

### Actual Result

is set to 240

### Version Info

Version V1.19.03

jgyates, 2024-06-22:
You did not include your generator / controller type or logs so I am assuming you have a Evoution/Nexus.

Do you have "Use Nominal Volts Override" enabled on the Advanced Setting page?

rsockanc, 2024-06-22:
How embarrassing, I have the checkbox for that value and hit, save, but forgot to restart services. Issue can be closed.

jgyates, 2024-06-22:
No worries. Have a good one.

closed 2024-06-22 by jgyates