jgyates/genmon #1140: MQTT: System in Alarm not clearing when no alarm present

chrkov, 2024-06-27:
### Steps to Reproduce

1.  System goes into alarm
2. MQTT (Status/Engine/System In Alarm) changes as expected
3. Clear alarm
4. MQTT (Status/Engine/System In Alarm) alarm remains

An alarm was triggered for battery maintenance.   Cleared alarm on generator.  But the MQTT field still says "Check Battery"

### Expected Result

I would expect this message to clear out or change to OK or No Alarms (or something of that nature).

### Actual Result

Last Alarm remains even though it is cleared

### Version Info

1.19.03

### Logs

[genmon_logs.tar.gz](https://github.com/user-attachments/files/16014207/genmon_logs.tar.gz)

jgyates, 2024-06-27:
@chrkov 

The contents of your registers would be helpful. My guess is that this has nothing to do with the MQTT add on since the add on does not keep any state info, it just reads the data form genmon. 

Does the web interface still show an alarm is active? If it does then you are likely not clearing the current alarm.

How did you clear the alarm? Not all alarms are cleared the same way. If memory serves, "Check Battery" is a maintenance timer for checking the battery which is cleared differently than and error type of alarm.

What is the contents of register 0001 on the live register page?

chrkov, 2024-06-27:
I cleared the alarm from the generator (had a yellow alarm light also.   Per the instructions.  Turned generator off.  Press Enter Twice and then set generator back to auto.   Yellow light on the generator clears and it did in genmon also.

Genmon shows no alarms now  (previously did before clearing it)
[image]
[image]

register 0001 shows all zeros
[image]
[image]

MQTT still shows the alarm
[image]
[image]

chrkov, 2024-06-27:
Log also shows the clearing.

[image]

chrkov, 2024-06-27:
What you said makes perfect sense.   With that, can I change this to a feature request to clear the "System In Alarm" field when alarms are not present?   Also, would an Engine State Alarm cause this field to populate, or is it just the switch state?

jgyates, 2024-06-27:
If there is an alarm of any kind on the controller then the MQTT topic /generator/Engine/Switch State/ topic will have a value of "System In Alarm". If /generator/Engine/Switch State/ is any other value besides "System In Alarm" then the controller is not in an alarm condition.

If you want to know the name of the active alarm condition that is active, then refer to the topic /generator/Status/Engine/System In Alarm.

The topic /generator/Status/Engine/System In Alarm does not indicate an alarm is active and it should not be used for that indication.

The logic behind this is related to the working of the controller. The lower 16 bits of Modbus register 0001 contain the switch state register. If an alarm is present you can not read the switch state via modbus, it only reports an alarm code. The MQTT topic  /generator/Engine/Switch State/  reflects the contents of this register, decided from hex to readable values.

I will think about this but the there is a down side to your feature request. This would result in a "System In Alarm: None" being displayed when an alarm is not present on the main status page and I would prefer that not be the case. Also, since the functionality you are looking for (being able to determine if an alarm is present via MQTT) already exists as described above then I don't see a lot of value in making this change.

Let me know if you have any other questions for feedback.

closed 2024-06-27 by jgyates