jgyates/genmon #1155: Gauges do not appear

gerolymatos, 2024-07-27:
### Steps to Reproduce

I just installed genmon software on PI 2 W. Install went well without errors but the gauges do not appear in the web interface. I have not connected the PI to the generator yet. Only the pHAT is attached. Please help. 

### Expected Result

Gauges didnt appear see image

### Actual Result

[image]

### Version Info

Genmon
Version V1.19.03

### Logs

[genmon_logs (1).tar.gz](https://github.com/user-attachments/files/16400327/genmon_logs.1.tar.gz)

skipfire, 2024-07-27:
Without being connected to the generator there is no data to display.  Since the UI is coming up you should be ready to go to the generator.

jgyates, 2024-07-27:
refresh your browser. If you still do not see the gauges then try it connected to the generator.

closed 2024-07-27 by jgyates

gerolymatos, 2024-07-27:
Apparently it takes a long time for the gauges to appear. Ive been refreshing the browser for about 10 minutes and the gauges didn't show. I left the PC on for a half hour and then refreshed the browser again. Bingo, they appeared. Would this be a specific delay issue with the PI 2 W or is this to be expected? Next step is to connect the PI to the generator...

skipfire, 2024-07-27:
That is not abnormal when you're not connected to a generator. Once connected it will probably be working the way you expect.

jgyates, 2024-07-27:
A fure more details from @skipfire 's post:  The web UI will not fully populate if the initialization of genmon is not complete. The initialization will be complete once all modbus registers have been read once (so the model and specs of the generator can be read). If you are not connected to a controller then a modbus time out occurs for each modbus register read (about 250 registers reads). In most cases it takes about 2-4 minutes to , but with the pi's with slower processors like the Pi2W it does not surprise me that it takes longer. The Monitor page will show the genmon status if initialization is still on going. If you are connected to the controller it will take a few seconds to initialize.

gerolymatos, 2024-07-28:
Thank you for the information. Should I have installed a better RP instead of the Pi2W? I cannot get the web UI to load after installed in the generator. (There is no enclosure for the Pi. It is temporarily zip tied inside the generator) The wifi decibal is about -55dBm while in the generator. It will load on occasion but most of the times I get a blank white screen on the webpage. Should the Pi2W be mounted in an enclosure outside of the generator? Or should I upgrade to a better Pi? What do you recommend?

skipfire, 2024-07-28:
Do you have a Pi 2 or a Pi Zero 2W?  You said you have a pHAT which would be appropriate if you have a Zero 2W which is powerful enough to run GenMon smoothly.  The signal strength you mention is fine.

For the blank white screen, is that the entire page, or just the content section with the left menu still showing?

Have you looked at the monitor page for CRC and timeout percentages?

When it does show the page, did it appear to be getting data?

gerolymatos, 2024-07-28:
logging in to the Pi is quick through SSH and Pinging it has zero % loss. Not sure why UI is having issues loading.

jgyates, 2024-07-28:
Are you running the 32 bit or 64 bit version of the operating system? 

From the pi console, try restarting and post the output here. Type this to restart:

     cd genmon
     ./startgenmon.sh restart

gerolymatos, 2024-07-28:
Running 32 bit version of OS. 
I tried your command. Please see attached image...
[image]

jgyates, 2024-07-28:
are there any errors in /var/log/genserv.log

Do you have an ASUS Router? Some routers, like ASUS have an some security features that block local web servers. For ASUS it is called AIProtection if memory serves.

gerolymatos, 2024-07-28:
[image]

I have an ASUS router. Its a GT-AX11000. Is there a setting that needs to be changed?

jgyates, 2024-07-28:
I don't know the specific setting but this thread may have some info. If you find out which one let me know.

https://github.com/jgyates/genmon/issues/271

gerolymatos, 2024-07-28:
The UI finally just appeared. I just left the browser window running. I took a snapshot of the monito page. Do you see anything abnormal? Im afraid to close the window out and try to load the page again. (It might take an hour to load again, :(

[image]

jgyates, 2024-07-28:
Everything looks normal. The modbus exceptions are normal for the Evolution 2.0. From the past reporting from the AIProtection it would work some times and other times it would not so this is looking like the same thing here.

gerolymatos, 2024-07-28:
I turned off AI protection to see if it would help but the UI is still not loading. There may be something else that slowing it down.

skipfire, 2024-07-28:
Can you try another device, and possibly incognito/private mode?  It's quite possible this is an issue at your computer and not at the Pi.

gerolymatos, 2024-07-28:
I tried from my phones browser and on another PC but it still does the same thing. (blank page)

gerolymatos, 2024-07-28:
I tried private mode/incognito on the browser but it didn't help. Not sure what else to do.

gerolymatos, 2024-07-28:
[image]

A snapshot of the router transmission and signal.

gerolymatos, 2024-07-28:
[image]

gerolymatos, 2024-07-28:
I'm wondering if this is contributing to the issue.

jgyates, 2024-07-28:
I don't know the significance of the transmission rate, is this the rate of the pi? What is the difference between the two screen shots?

Also, I would look at other security related settings in the router. I am not sure if it AIprotection settings or some other settings in the router with AI protection so more experimenting may be needed.

gerolymatos, 2024-07-28:
The two images were take a couple minutes apart. Other Tx and Rx rates on other devices were stable. 
I disabled antivirus apps for any changes but nothing. Ill take a look at any further settings on the router. I may even open the generator enclosure tomorrow for any improvement.

jgyates, 2024-07-28:
You wifi signal looks strong, but the quality and noise is not optimal, possibly due to reflections. Mounting in a plastic enclosure outside the metal generator enclosure may help.
 Maybe a test outside the enclosure may provide a data point.

gerolymatos, 2024-07-28:
Thats what I was thinking too. Ill be testing the Pi outside of the generator. I also just wanted to share this screen shown below as i noticed it just now when i returned to my computer. This screen is new and I have never seen it before. The round icon in the middle was spinning but still no UI page loaded. Ill keep you posted on any changes.

[image]

gerolymatos, 2024-07-28:
I took out the Pi from the generator and it only took about a minute for the UI to appear. Then I closed and reopened the browser  in which the UI appeared within seconds. I tried this multiple times and opened quickly. Please see attached images of where the Pi was located at the time of testing and the Wifi signal while it was out of the generator. I slightly moved the pi to take a picture but now it is not loading the UI. I have attached a few images for review. The images are from when the UI loaded quickly.

[image]

[image]

[image]

jgyates, 2024-07-28:
sounds like that may be the data point we are looking for. Signal quality went up as did the signal. Noise is the same but -256 is likely the wifi driver not supporting noise stats.

skipfire, 2024-07-28:
I have mine in a plastic box on the back of my generator held with 3M double-sided foam adhesives (clean the generator enclosure first for a good stick) and just ran the cable under the side panel.

[image]

gerolymatos, 2024-07-28:
Is there a driver that should be implemented to alleviate the issue?  It is taking a long time to load its been over 5 minutes now and still no UI. 

For skipfire, What is the card I see in the picture? Is that an external Wifi antenna? I like the 3M idea but not so much of the exposed wires.

jgyates, 2024-07-28:
I would not deviate from the standard wifi driver. Not having a signal to noise stat is not really hurting anything. I think the driver has to put the device out of wifi mode to get that number so it may end up reducing performance.

skipfire, 2024-07-28:
The green card is an external antenna, I have it there for when I'm testing with an ESP or NanoPi device as those use external antennas (I'm the guy behind PintSize.Me).

The problem with the cabling is that you either end up with the exposed wires like this or you have to drill.  I have been considering using cable runners (the ones you can put on a wall inside the house) but haven't gotten around to it yet.

gerolymatos, 2024-07-28:
Good idea with the runners skipfire! 

The UI is not reliable in its current state outside of the generator. Im still waiting for the UI to appear. Kind of in a rut. What else can I try? Somehow the UI was loading quickly for a while, Now its back to not loading at all.

gerolymatos, 2024-07-28:
Do you believe I may have a bad Pi?

skipfire, 2024-07-28:
That is possible, I also typically have micro-SD issues manifest in performance or networking issues as the first sign.  What micro-SD card did you use?  Even among the good brands, the cheaper ones do not always do well for a Pi.

One other thing you could try is loading GenMon right on the Pi.  You could cart a display, keyboard, and mouse outside (I'm not recommending that, even though I do it on occasion but I put together a kit for that); or temporarily enable VNC (via raspi-config).  If GenMon loads great from http://localhost:8000/ then the problem is likely not the Pi/micro-SD, if it loads like crap on the Pi then the problem is probably the Pi or micro-SD card.

gerolymatos, 2024-07-28:
Its a Kingston SD card. I have a Sandisk. I could try that one and see if its any better? I think I'm going to wrap it up for today though and come back another day. Ill let you know how the Sandisk works out. Is there a card that you suggest to use which works best with the Pi zero 2W? Thanks for your help so far!

skipfire, 2024-07-28:
It isn't so much the brand as the line within the brand.  I've had issues and success with most brands based on the type of card.  SanDisk High Endurance and Max Endurance do great while the other SanDisk models (such as Ultra and Extreme) I've had some problems with.  Micro-SD cards from the big names and marketed by them for dashcams or other frequent video use are the ones I've had the best success with, and I try to stick with U3 speeds.

jgyates, 2024-07-28:
I agree with @skipfire , at one time the minimum speed for running a raspberry pi was a Class 10 SD card. Most people make them but if you are using an old one that was laying around then you may consider getting a new one as they do have a finite lifetime. It is not small but it can fail.

gerolymatos, 2024-07-29:
I placed an order for an enclosure to mount outside of the generator and a 32GB SanDisk Max Endurance SD card. Ill post again with some hopefully better results.