jgyates/genmon #1158: MQTT "Numeric Topics" option not applied to Calculated Frequency, which gets published with numeric + "Hz"

billpribyl, 2024-07-27:
### Steps to Reproduce

1. Add-ons -> MQTT -> Set Numeric Topics on
2. JSON for Numerics -> Off

Then view the message sent for Home/generator/Status/Engine/Unsupported Sensors/Hz (Calculated)

### Expected Result

60.0

Comment: At least, that's my opinion...maybe there's a reason it needs the units?

### Actual Result

60.0 Hz

Full output from MQTTX tool:

Topic: Home/generator/Status/Engine/Unsupported Sensors/Hz (Calculated) QoS: 0
60.0 Hz
2024-07-11 11:14:57:720

### Version Info

1.19.03

jgyates, 2024-07-27:
The Unsupported sensors are only displayed for debugging purposes and not not included in the Numerics option.

jgyates, 2024-07-27:
To remove the "Unsupported Sensors" from the output, disable the "Display Unknown Sensors" option on the settings page. This option is off by default.

closed 2024-07-27 by jgyates