jgyates/genmon #1162: can not connect to web interface

mullinp, 2024-08-05:
### Steps to Reproduce

Pi died during recent outage. Got a new Zero W. No interface on boot. Did a fresh install (x5) with no good results. Thought maybe it was a router issue (DECO). Tried old router with no luck.  I can interface fine using terminal but can not get the web page at port 8000 to load.

### Expected Result

Web interface on port 8000

### Actual Result

Genserv log:

2024-08-04 20:17:39,960 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True, UseMFA:False
2024-08-04 20:17:48,988 : Error: Connect : [Errno 111] Connection refused : myclient.py:58
2024-08-04 21:13:50,590 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True, UseMFA:False
2024-08-04 21:13:59,620 : Error: Connect : [Errno 111] Connection refused : myclient.py:58

Genloader log:

2024-08-04 20:15:09,322 : Warning: required library pyopenssl not installed. Attempting to install....
2024-08-04 20:17:14,259 : Running one time maintenance check
2024-08-04 20:17:29,043 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 20:17:30,166 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 20:17:31,298 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 20:17:32,421 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 20:17:33,544 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 21:01:43,131 : Warning: required library flask not installed. Attempting to install....
2024-08-04 21:01:57,532 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'flask']: b'error: externally-managed-environment\n\n\xc3\x97 This envir>
2024-08-04 21:01:57,541 : Warning: required library pyserial not installed. Attempting to install....
2024-08-04 21:02:05,254 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'pyserial']: b'error: externally-managed-environment\n\n\xc3\x97 This en>
2024-08-04 21:02:05,264 : Warning: required library crcmod not installed. Attempting to install....
2024-08-04 21:02:12,832 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'crcmod']: b'error: externally-managed-environment\n\n\xc3\x97 This envi>
2024-08-04 21:02:12,841 : Warning: required library pyowm not installed. Attempting to install....
2024-08-04 21:02:20,312 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'pyowm==2.10.0']: b'error: externally-managed-environment\n\n\xc3\x97 Th>
2024-08-04 21:02:20,321 : Warning: required library pytz not installed. Attempting to install....
2024-08-04 21:02:27,817 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'pytz']: b'error: externally-managed-environment\n\n\xc3\x97 This enviro>
2024-08-04 21:02:27,826 : Warning: required library pysnmp not installed. Attempting to install....
2024-08-04 21:02:35,309 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'pysnmp']: b'error: externally-managed-environment\n\n\xc3\x97 This envi>
2024-08-04 21:02:35,318 : Warning: required library ldap3 not installed. Attempting to install....
2024-08-04 21:12:44,328 : Warning: required library pyopenssl not installed. Attempting to install....
2024-08-04 21:13:39,009 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 21:13:40,135 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 21:13:41,708 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 21:13:42,957 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571
2024-08-04 21:13:44,189 : Error in IsRunning: PID still exists but it's a zombie (pid=687, name='python'): mysupport.py:571

### Version Info

fresh install so current

### Logs

see result section

jgyates, 2024-08-05:
Did you run the install script in the main repository folder? Your logs posted indicate that you likely did not run the install script or it did not run correctly as there are several missing libraries.

This page shows the installation procedures:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

The log indicates that the web interface is not loading because the Flask library is not installed. I am assuming the file /var/log/genserv.log does not exist (since Flask is not installed).

If you have not run the install instructions (including the git commands and the execution of the install script genmonmaint.sh) then I would recommend doing that. If you have done this then save the output of the install script and post it here.

mullinp, 2024-08-05:
here were the results:
(maybe should have used sudo bash?)

pi@genmon2:~/genmon $  bash ./genmonmaint.sh -i -n
The operating system is: Raspbian GNU/Linux
Managed system packages found, installing python virtual environment
Setting up virtual python environment for genmon
Installing genmon package requirements....
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Selecting previously unselected package javascript-common.
(Reading database ... 64497 files and directories currently installed.)
Preparing to unpack .../00-javascript-common_11+nmu1_all.deb ...
Unpacking javascript-common (11+nmu1) ...
Selecting previously unselected package libexpat1-dev:armhf.
Preparing to unpack .../01-libexpat1-dev_2.5.0-1_armhf.deb ...
Unpacking libexpat1-dev:armhf (2.5.0-1) ...
Selecting previously unselected package libjs-jquery.
Preparing to unpack .../02-libjs-jquery_3.6.1+dfsg+~3.5.14-1_all.deb ...
Unpacking libjs-jquery (3.6.1+dfsg+~3.5.14-1) ...
Selecting previously unselected package libjs-underscore.
Preparing to unpack .../03-libjs-underscore_1.13.4~dfsg+~1.11.4-3_all.deb ...
Unpacking libjs-underscore (1.13.4~dfsg+~1.11.4-3) ...
Selecting previously unselected package libjs-sphinxdoc.
Preparing to unpack .../04-libjs-sphinxdoc_5.3.0-4_all.deb ...
Unpacking libjs-sphinxdoc (5.3.0-4) ...
Selecting previously unselected package zlib1g-dev:armhf.
Preparing to unpack .../05-zlib1g-dev_1%3a1.2.13.dfsg-1_armhf.deb ...
Unpacking zlib1g-dev:armhf (1:1.2.13.dfsg-1) ...
Selecting previously unselected package libpython3.11-dev:armhf.
Preparing to unpack .../06-libpython3.11-dev_3.11.2-6+deb12u2_armhf.deb ...
Unpacking libpython3.11-dev:armhf (3.11.2-6+deb12u2) ...
Selecting previously unselected package libpython3-dev:armhf.
Preparing to unpack .../07-libpython3-dev_3.11.2-1_armhf.deb ...
Unpacking libpython3-dev:armhf (3.11.2-1) ...
Selecting previously unselected package python3.11-dev.
Preparing to unpack .../08-python3.11-dev_3.11.2-6+deb12u2_armhf.deb ...
Unpacking python3.11-dev (3.11.2-6+deb12u2) ...
Selecting previously unselected package python3-dev.
Preparing to unpack .../09-python3-dev_3.11.2-1_armhf.deb ...
Unpacking python3-dev (3.11.2-1) ...
Selecting previously unselected package python3-setuptools.
Preparing to unpack .../10-python3-setuptools_66.1.1-1_all.deb ...
Unpacking python3-setuptools (66.1.1-1) ...
Selecting previously unselected package python3-wheel.
Preparing to unpack .../11-python3-wheel_0.38.4-2_all.deb ...
Unpacking python3-wheel (0.38.4-2) ...
Selecting previously unselected package python3-pip.
Preparing to unpack .../12-python3-pip_23.0.1+dfsg-1+rpt1_all.deb ...
Unpacking python3-pip (23.0.1+dfsg-1+rpt1) ...
Setting up javascript-common (11+nmu1) ...
Setting up python3-setuptools (66.1.1-1) ...
Setting up python3-wheel (0.38.4-2) ...
Setting up libexpat1-dev:armhf (2.5.0-1) ...
Setting up python3-pip (23.0.1+dfsg-1+rpt1) ...
Setting up zlib1g-dev:armhf (1:1.2.13.dfsg-1) ...
Setting up libjs-jquery (3.6.1+dfsg+~3.5.14-1) ...
Setting up libjs-underscore (1.13.4~dfsg+~1.11.4-3) ...
Setting up libpython3.11-dev:armhf (3.11.2-6+deb12u2) ...
Setting up libjs-sphinxdoc (5.3.0-4) ...
Setting up libpython3-dev:armhf (3.11.2-1) ...
Setting up python3.11-dev (3.11.2-6+deb12u2) ...
Setting up python3-dev (3.11.2-1) ...
Processing triggers for man-db (2.11.2-2) ...
Extracting templates from packages: 100%
Selecting previously unselected package libhttp-parser2.9:armhf.
(Reading database ... 65795 files and directories currently installed.)
Preparing to unpack .../00-libhttp-parser2.9_2.9.4-5_armhf.deb ...
Unpacking libhttp-parser2.9:armhf (2.9.4-5) ...
Selecting previously unselected package libmbedcrypto7:armhf.
Preparing to unpack .../01-libmbedcrypto7_2.28.3-1_armhf.deb ...
[... 302 lines cut ...]
Finished installing rpirtscts.
~/genmon
~/genmon ~/genmon

Using files: /boot/firmware/config.txt and /boot/firmware/cmdline.txt

Model: Raspberry Pi Zero W Rev 1.1

Enable UART: Success
Change BT serial port: Success
Enable Turbo clocking: Success
Disable BT Disable Overlay: Success
Disable serial console: Success
Disable serial console service: Success
Unit /lib/systemd/system/hciuart.service is added as a dependency to a non-existent unit dev-serial1.device.
Enable BT service service: Success

Serial port settings are OK. A reboot is needed if changes were made.

Finished setting up the serial port.
~/genmon
Done.
no crontab for root
Updating crontab...
adding < @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start > to crontab

jgyates, 2024-08-05:
Was this before log from before or after the logs in the first message?

This looks like a successful install. The install script indicates you are installing in a managed python environment, but your logs indicate that libraries are not installed, which makes me question how you are starting genmon. If you do not use the scripts (startgenmon.sh) to start genmon then it will not start in a managed environment.

How are your starting genmon (on boot?, manually?)? What are the exact commands you are using to start? 

Also, are you using a 32 bit or 64 bit os and is this the Raspberry Pi OS?

mullinp, 2024-08-05:
This was prior to the logs.

Starting on boot.

32 bit (lite version)

I will try using startgenmon and post results...

mullinp, 2024-08-05:
pi@genmon2:~/genmon $ ./startgenmon.sh start
using binary: /home/pi/genmon/genenv/bin/python
Starting genmon python scripts
Warning: required library pyopenssl not installed. Attempting to install....
Running one time maintenance check
Starting....
Starting /home/pi/genmon/genmon.py 
Stopping genserv.py
Starting /home/pi/genmon/genserv.py

magomez96, 2024-08-05:
I had this same problem just now. This fixed it for me: sudo apt install libssl1.1

mullinp, 2024-08-05:
That did it!!!! Thanks magomez!

> I had this same problem just now. This fixed it for me: sudo apt install libssl1.1

That did it!

jgyates, 2024-08-05:
Thanks for the input @magomez96.

From time to time the install breaks due to dependency issues with the required python packages. Usually this is resolved by the package maintainers in a few days time. Also, empirically, this appears to happen in the less popular distribution (e.g. 32 bit version of the OS / packages) more often.

closed 2024-08-05 by jgyates

magomez96, 2024-08-05:
@jgyates FYI this is the error that led me to the solution:
python3 ./genmon.py 

This program requires the modules located in the genmonlib directory in the github repository.

Please see the project documentation at https://github.com/jgyates/genmon.

Error: libssl.so.1.1: cannot open shared object file: No such file or directory

This was on a fresh install of Raspbian Bookwork 12  32 bit on a Pi Zero 2 W

cbp0616, 2024-08-13:
> I had this same problem just now. This fixed it for me: sudo apt install libssl1.1

I know this issue is closed but I just wanted to say I had the same issue and beating my head for hours I stumbled upon this reply.  Guessing this is a common isssue for folks following guides which start with the raspbian 32 bit build. 

Thank you!