jgyates/genmon #1163: Did upgrade and now I just get “loading generator monitor”

rhombixt, 2024-08-08:
### Steps to Reproduce

Clicked on upgrade and I’m stuck. I have teamviewer remote access if needed . 

### Expected Result

Genmon said update complete but won’t load

### Actual Result

Loading generator monitor …

### Version Info

[image]

### Logs

Can’t access due to loading window

rhombixt, 2024-08-08:
Now it’s crashed .

jgyates, 2024-08-08:
What crashed, the pi? That may indicate a failing SD card if the OS has crashed.

Can you use ssh to log into the pi?

Were you experiencing any issues before you upgraded, for example communication errors (it should show a red error at the bottom of the web interface before the upgrade)?

you can show the output of the following files:

/var/log/genmon.log
/var/log/genserv.log
/var/log/genloader.log

The "Loading Generator Monitor" is often an indication that genmon is loaded but if you are getting communication errors then this message will be displayed. Genmon will show this message until all of the modbus registers are read at least once. If you lost serial comms then this could take between 2 and 10 minutes, depending on the speed of your pi.

rhombixt, 2024-08-08:
The web timed out eventually- so it went from loading to timed out. 
then I rebooted the pi however it did not load the program . I posted my issue then tried looking at other folks who had upgrade issues and tied your fixes and it’s not working . I use genmon and the gengpioin application that both need to start  at the same time .  This happens the last time I tried to upgrade as well in 2017 and you teamed in and fixed a bunch of issues . I have it setup if you want to access . 
[image]
The other logs don’t show anything , the loader log does

rhombixt, 2024-08-08:
I had no issues before but it was a very old version I was running . I did an upgrade that went sideways that you fixed in 2017 and I have not touched it since.

rhombixt, 2024-08-08:
2024-08-08 15:33:49,770 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'paho-mqtt==1.6.1']: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:49,811 : Warning: required library voipms not installed. Attempting to install....
2024-08-08 15:33:50,687 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'voipms==0.2.5']: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:50,724 : Running one time maintenance check
2024-08-08 15:33:51,930 : Error in GetLibararyVersion using pip : pyowm: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:51,937 : Error in FixPyOWMMaintIssues: pyowm version not found

jgyates, 2024-08-08:
That is your issue. If this is running from a version in 2017 there is no telling what would happen. the genmon github repository was created in March of 2017 so you are dealing with a version from the first year of genmon. I suggest you run the install again. There have been many changes to not only genmon but all of the dependent libraries in the past 7 years.

The recommendation is to either:

1) upgrade genmon periodically on regular intervals (no more 1 year at most between updates)
or 
2) install it once and live with the current functionality

this is the case with most open source software. Since most open source project rely on software from multiple source (for example genmon uses about 22 different open source libraries and some of those have other dependancies as well) it is impossible to do testing with all variations of all software so staying relatively current is the key to maintaining the health of open source software. 

You can make a copy of the files in the /etc/genmon folder by logging into genmon and running this command (these files contain your genmon settings):

     cd genmon
     ./genmonmaint.sh -b

The above command will create a file named genmon_back.tar.gz in the genmon folder.

Then stop genmon with this command:

     ./startgenmon.sh stop

Next I would update your system software:

     sudo apt-get update
     sudo apt-get dist-upgrade

then update to the latest software with this command (from the genmon directory):

      ./genmonmaint.sh -u

type 'y' at the prompt to update.

then run the setup script:

     ./genmonmaint.sh -i

This will ask you some questions that you will either respond yes (y) or no (n). When it ask to copy the config files type N for no. When asking to start genmon on boot, type N for no as you already have that from the last install.

Once this is complete you can restart your pi with this command:

     sudo reboot now

then wait for the pi to restart and check with web interface

rhombixt, 2024-08-08:
What about the gengpioin application ?

rhombixt, 2024-08-08:
How do you uninstall it ?

jgyates, 2024-08-08:
The script startgenmon.sh should also stop any add on programs.

Also, type this command:

     sudo crontab -l

that is a lowercase L at the end not a one. Just making sure your old install used the startgenmon.sh command on startup. If the above returns nothing then select Y when asked to start on boot.

rhombixt, 2024-08-08:
[image]

When I run this comes up..

rhombixt, 2024-08-08:
Crontab command :

# Edit this file to introduce tasks to be run by cron.
# 
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
# 
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').# 
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
# 
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
# 
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
# 
# For more information see the manual pages of crontab(5) and cron(8)
# 
# m h  dom mon dow   command
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

rhombixt, 2024-08-08:
# Edit this file to introduce tasks to be run by cron.
# 
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
# 
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').# 
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
# 
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
# 
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
# 
# For more information see the manual pages of crontab(5) and cron(8)
# 
# m h  dom mon dow   command
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

rhombixt, 2024-08-08:
web page did not load

rhombixt, 2024-08-08:
Genloader

2024-08-08 15:33:49,770 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'paho-mqtt==1.6.1']: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:49,811 : Warning: required library voipms not installed. Attempting to install....
2024-08-08 15:33:50,687 : Error in InstallLibrary using pip : ['/usr/bin/python3', '-m', 'pip', 'install', 'voipms==0.2.5']: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:50,724 : Running one time maintenance check
2024-08-08 15:33:51,930 : Error in GetLibararyVersion using pip : pyowm: b'Traceback (most recent call last):\n  File "/usr/lib/python3.4/runpy.py", line 151, in _run_module_as_main\n    mod_name, mod_spec, code = _get_module_details(mod_name)\n  File "/usr/lib/python3.4/runpy.py", line 118, in _get_module_details\n    return _get_module_details(pkg_main_name)\n  File "/usr/lib/python3.4/runpy.py", line 104, in _get_module_details\n    spec = importlib.util.find_spec(mod_name)\n  File "/usr/lib/python3.4/importlib/util.py", line 86, in find_spec\n    parent = __import__(parent_name, fromlist=[\'__path__\'])\n  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>\n    from pip.vcs import git, mercurial, subversion, bazaar  # noqa\n  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'
2024-08-08 15:33:51,937 : Error in FixPyOWMMaintIssues: pyowm version not found
2024-08-08 17:56:04,264 : Error in ExecuteCommandList  : b'E: Command line option --allow-releaseinfo-change is not understood\n'
2024-08-08 17:56:04,272 : Error: Unable to run apt-get update. Retrying...
2024-08-08 17:56:28,323 : Error in ExecuteCommandList  : b'W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/main/binary-armhf/Packages  404  Not Found [IP: 93.93.128.193 80]\n\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/contrib/binary-armhf/Packages  404  Not Found [IP: 93.93.128.193 80]\n\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/non-free/binary-armhf/Packages  404  Not Found [IP: 93.93.128.193 80]\n\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/rpi/binary-armhf/Packages  404  Not Found [IP: 93.93.128.193 80]\n\nE: Some index files failed to download. They have been ignored, or old ones used instead.\n'
2024-08-08 17:56:28,325 : Error: Unable to run apt-get update. 
2024-08-08 17:56:28,326 : Installing cmake...
2024-08-08 17:56:44,578 : Error in ExecuteCommandList  : b'E: There are problems and -y was used without --force-yes\n'
2024-08-08 17:56:44,580 : Error: Unable to install cmake.
[... cut ...] st-packages/pip/vcs/mercurial.py", line 9, in <module>\n    from pip.download import path_to_url\n  File "/usr/lib/python3/dist-packages/pip/download.py", line 22, in <module>\n    import requests, six\n  File "/usr/local/lib/python3.4/dist-packages/requests/__init__.py", line 43, in <module>\n    import urllib3\n  File "/usr/local/lib/python3.4/dist-packages/urllib3/__init__.py", line 37\n    f"the \'ssl\' module is compiled with {ssl.OPENSSL_VERSION!r}. "\n    ^\nSyntaxError: invalid syntax\n'

jgyates, 2024-08-09:
in your logs you can see this URL:

http://mirrordirector.raspbian.org/raspbian/dists/jessie/main/binary-armhf/Packages 

This indicates you are running the Jessie version of the OS which is based on Debian 8. The current version of Raspbian / Debian is 12. The long term support for your OS (Jessie) ended in 2020 so you can not update your software via the apt package manager. This is causing the genmon install script to fail. genloader tries to install packages that are missing as a back up to the install script and it is failing as well. 

You really have 2 choices:
1) down load an older version of genmon and put it in the genmon folder on your pi. You can scroll back through old releases here: https://github.com/jgyates/genmon/releases . With this option I would not be able to help much as I only support the latest version

2) get a new SD card, install the latest version of the Raspberry pi OS and install genmon from scratch using these instructions: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software . This will put you on the latest version with an operating system that still has several years of long term support.

Every version of linux has some cutoff date for support. Here is history of Debian based OSes (like the Raspberry pi OS): https://en.wikipedia.org/wiki/Debian_version_history

The current stable version of Debian is bookworm.

rhombixt, 2024-08-09:
ok i will complete tonight. Update when I am done.

rhombixt, 2024-08-09:
I assume a 64 bit os is ok for pi 3?

rhombixt, 2024-08-09:
Also - can I restore my settings with the backup you made me perform or is it not necessary?

skipfire, 2024-08-09:
If you plan on using the OS GUI on the Pi itself for anything you may want to go with 32-bit OS, but if you are going with Lite then 64-bit should be fine.

jgyates, 2024-08-09:
Personally I use the 64 bit lite version.

jgyates, 2024-08-09:
Yes you can restore from the backup. You may just want to run the install scrip and use the default settings. You can always restore from you backup later, after you get everything up and running. For example you can just restore your email settings if you don't remember the password. You can just restore your outtage log, etc.

rhombixt, 2024-08-09:
Getting an error when running the genmonmaint/sh -i 
Error 13 permission denied /genmon/genenv

seems everything with the new OS is about permissions. I had to use the sudo command to get the git installer to work

how do I change the genmon directory so it's open?

jgyates, 2024-08-09:
If you log in via ssh you should be in the directory for the user you logged in as. In this situation you should not have permission issues with the file system. What directory did you run the install commands?

rhombixt, 2024-08-09:
I have the terminal window open , changed directory to genmon

rhombixt, 2024-08-09:
changed user to root and ran - its going now.

Some observations with your instructions in the wiki

 sudo apt-get update
 sudo apt-get dist-upgrade  (should just be sudo dist-upgrade )

 git clone https://github.com/jgyates/genmon.git did not work - had to use sudo in front of it ( i supposed if I was running as root i would not have to use sudo )

jgyates, 2024-08-09:
If you used the sudo command with the git command, then the genmon folder is now owned by the root user. You should not have had to use sudo git. What directory where you in when you ran the git command?

rhombixt, 2024-08-09:
have I done something wrong - should I re-format and do everything over again?

rhombixt, 2024-08-09:
/home - there is no pi directory just my user folder and rpi-first-boot-wizard

jgyates, 2024-08-09:
If you run this from your user folder you would have the right permissions. In earlier versions of the raspberry pi os the default user name is pi, so this would be user folder

rhombixt, 2024-08-09:
ok well when I ran it - it said permission denied.
Should I use the chown command on the folder?

jgyates, 2024-08-09:
I would get into you user folder, run git and then genmonmaint.sh from you home/user/genmon folder. Then delete the genmon folder in home with sudo rm.

rhombixt, 2024-08-09:
its up and running despite being in the wrong spot - any issues with leaving as is?

jgyates, 2024-08-09:
Try doing a software update from the about page, then look at /var/log/gender.log for any errors. If no errors you are probably ok

jgyates, 2024-08-09:
I am going to close this thread but feel free to post here with any follow up status or questions.i am glad it is work for you

closed 2024-08-09 by jgyates

rhombixt, 2024-08-09:
Have there been any changes to gpio at all? Different io ports for alarms or are they the same?

rhombixt, 2024-08-09:
I don’t have a gender under the logs folder

rhombixt, 2024-08-09:
Genserv shows no errors

jgyates, 2024-08-09:
There have been many changes over the years, I don't recall changing the default gpio values.

rhombixt, 2024-08-09:
Thanks. I created a second SD card and made a second image . I managed to do everything from start to finish a second time with no issues. Thanks for the assistance.

jgyates, 2024-08-09:
no problem. I am glad it is working for you.