jgyates/genmon #1168: Powerlog isn't showing power consumed details from 3 weeks ago

skipfire, 2024-08-21:
### Steps to Reproduce

I have a monthly transfer test setup for the first of every month and it runs with load for 58 minutes (9:30 to 10:30 including warmup and cooldown).  The kw graph in genmon is not showing much for that timeframe.  The call to /cmd/power_log_json?power_log_json:%2010000 is not returning that data (excerpt below), but kwlog.txt does show significant use (second excerpt & full attached).

[image]

```
	[
		"08/05/24 10:01:35",
		"0.0"
	],
	[
		"08/01/24 10:30:01",
		"0.0"
	],
	[
		"08/01/24 09:31:19",
		"0.0"
	],
	[
		"08/01/24 09:30:50",
		"0.0"
	],
	[
		"08/01/24 09:30:39",
		"0.0"
	],
	[
		"07/29/24 10:06:05",
		"0.0"
	],
```

```
08/01/24 09:41:00,10.188
08/01/24 09:41:10,10.127
08/01/24 09:41:20,10.158
08/01/24 09:41:30,12.532
08/01/24 09:41:40,12.94
08/01/24 09:41:50,12.845
08/01/24 09:42:00,13.472
08/01/24 09:42:10,17.28
08/01/24 09:42:20,17.545
08/01/24 09:42:30,17.256
08/01/24 09:42:40,17.508
08/01/24 09:42:50,17.4
08/01/24 09:43:00,17.256
08/01/24 09:43:10,17.363
08/01/24 09:43:20,15.204
08/01/24 09:43:30,15.207
08/01/24 09:43:40,15.183
```

### Expected Result

poiwer_log_json and thus the graph would return/display the kw generated data for that month.  

### Actual Result

I have 4 datapoints returned from the API call at 0kW that only cover the first and last minute, nothing that covers the significant load between the first and last minute.

### Version Info

v1.19.03

### Logs

[kwlog.txt](https://github.com/user-attachments/files/16694659/kwlog.txt)
[power_log.json](https://github.com/user-attachments/files/16694695/power_log.json)

skipfire, 2024-08-21:
I also have 0.00kWh showing on the maintenance page for "kW Hours in last 30 days".

jgyates, 2024-08-21:
@skipfire 

Thanks for the data on this. I checked in an update (version number has changed to 1.19.04) that should fix this issue. There is some code that reduced the power log to a working number of samples to help make the web browser experience better. I never really liked the way I implemented this code but it was working as far as I knew. Turns out this was not a great way to do things with a reasonably large set of data in the power log. I updated the method to reduce the working set of data in the power log. This update should fix your issue. Let me know how it goes in your testing.

closed 2024-08-21 by jgyates

skipfire, 2024-08-21:
What I see now is along what I had expected, thanks for the fast turnaround.
[image]