jgyates/genmon #1169: exercise time issue

wrvolz, 2024-08-21:
### Steps to Reproduce

can't change exercise time. Normal time is wed at 1300 weekly. Today failed code 1505. Manually started OK. Tried to reset exercise time to 1600 today to force an exercise but the time doesn't change. I get a popup asking to confirm the change but it doesn't take. Generator model is G0062500. I found genexercise.conf but the setting in that (Monthly at 1200) don't match the actual (weekly on Wed at 1300).

### Expected Result

Time for exercise should change

### Actual Result

Time for exercise doesn't change

### Version Info

1.19.04

jgyates, 2024-08-21:
Your post is missing a lot of information that I would need to provided any input other than a guess.

What controller do you have (Evolution 1 or 2)?

The file genexercise.conf has nothing to do with the required exercise cycle built into the controller. This file is used for the Add on "Enhanced Exercise" which adds a 2nd exercise cycle. The built in exercise cycle is controlled by the controller, not genmon although you can set it from the Maintenance page. Do you have "Enhanced Exercise Time" setting enabled on the Settings page?

You can set the built in exercise time in the Maintenance page of the web interface. It is not clear if you are doing that as your response does not mention anything about how you forced a change (web interface, from the controller, trying to edit the genexercise.conf file, etc). 

If you have an active alarm (you mentioned "Today failed code 1505", which I assume is an error code. If you are trying to change the exercise time via the controller or the Maintenance page then the controller may require that you clear the alarm condition before the engine can be started. 

You logs were not submitted.

If you still have questions you can send more data and a clear description of what you have done and where you are getting your data on the exercise time.

wrvolz, 2024-08-21:
Sorry. I don't know if I have Evolution 1 or 2 - how can I tell? I'll upload the logs. I went to Maintenance section, scrolled down to Set Exercise time and used that. Entering any time, got confirmation popup, answered OK.  Went to Status page and back to Mainrenance page but exercise time not changed. I manually started the generator and that seemed to clear the alarm code. - The panel said "Ready to run" when I put it auto.  On the Settings page, Enhanced exercise time was off. I turned it on and tried again but still didnt work.

jgyates, 2024-08-21:
Setting the built in exercise parameters involves reading and writing to the modbus registers of the controller. Genmon only writes to the registers and reads them back to display. There are two methods for writing the exercise information to the controller: Legacy mode (only weekly method is supported) or Enhanced mode (supports weekly, biweekly and monthly). Generac added the Enhanced mode at time time with a firmware upgrade to the Evolution 1 controller.

Here is what I am seeing from your registers:

The set of registers to read and write the Legacy registers shows your current exercise time.
The set of registers for the Enhanced mode are all zeros.

Based on this data with the fact that you are running Firmware version 1.07 (which is an old version of the controller firmware), I am guessing that you are running and early version of the firmware that does not support enhanced exercise (I think it started with V1.10 on Evo1).

I added a check in the code for early firmware version on Evo1 that should revert to the older style writes. All you should have to do is update from the About page like you appeared to do today since you are running version 1.19.04. the version number is still 1.19.04. I also recommend setting the  "Enhanced Exercise Time" option on the setting page back to disabled, then save your settings.

If this solves you issue then you should be able to set the exercise time and day for a weekly exercise (but not biweekly or monthly). 

Let me know how it goes. I am going to close this thread but feel free to post here with any status updates or additional questions.

closed 2024-08-21 by jgyates

wrvolz, 2024-08-21:
I first turned off the enhanced exercise mode. Then updated the software and tried again but it didn't work Then updated the software again and it worked this time. It's exercising now. Not sure why it didn't work the first time. Perhaps it was between your updates. Thanks,

jgyates, 2024-08-21:
sometimes there is a lag in the github servers before the update is available to every one. I am guessing that is the issue. I am glad it is working for you now.

wrvolz, 2024-08-22:
There might be an "off by one" issue. When I went back to reset the exercise time to 1300. (Maintenance - > set Exercise Time.) It actually set it to 12:59. I tried several times. I then tried 13:01 and it set it to 13;00.

jgyates, 2024-08-22:
Unfortunately this is a know issue. The older firmware uses a value of how many 128 second intervals from the generators current time to set the exercise time, however the firmware does not support exporting seconds with the current time. There is probably some other way to do that with the older firmware to make it more precise, but since this was 100% reverse engineered and generac changed it with newer firmware to support a more straight forward method of setting the exercise time I opted to not spend time more time figuring out how to sync with the generator time to the second. This is only an issue with the older method of setting the exercise time (i.e. Evo1 V1.09 and earlier and Nexus models). I think the current Evo1 firmware is at least 1.22.

jgyates, 2024-08-22:
Also, you can always set it from the controller to make it exact, or upgrade your firmware.

wrvolz, 2024-08-22:
That's fine. I rarely change that setting and being one minute off isn't going to affect anything.

Updating the firmware would require a Generac service person to do it and that'll cost several $100 and not worth it just for that. Next time I have one out, they'd do it then.

I hate the way Generac keeps their system closed. Thanks for this genmon tool. Very useful, especially for a nerd like me.

Bill Volz
832-724-5183

[d921777a-12e9-46ce-96aa-260af088907e]