jgyates/genmon #1171: Mqtt Reporting Air Cooled But Liquid Cooled

heffneil, 2024-08-27:
### Steps to Reproduce

I have a liquid cooled engine.  In mqtt I see it saying air cooled.  Also my generator serial number says "None - Controller has been replaced"

### Expected Result

Controller Detected does show "Evolution 4.5L, Liquid Cooled".  I really don't know if this is a MQTT add-on issue or something within Genmon or how it determines what is on this unit.  Where in the genmon interface would it reflect this information?

### Actual Result

See Above

### Version Info

Genmon
Version V1.19.04

heffneil, 2024-08-27:
Ok I found it in genmon it does reflect the wrong cooling:

Maintenance :
    Model : Generic Air Cooled

heffneil, 2024-08-27:
Run Hours in last 30 days : 0.00 h
    Run Hours in the last year : 0.00 h

Is this a number that can be gotten from the Evolution controller?

jgyates, 2024-08-27:
I would need to see your logs to determine if something unexpected is happening. What you are describing can happen if you have edited the /etc/genmon/genmon.conf file with the incorrect data.

The info on the maintenance page includes Model, Serial Number and controller detected.

The serial number is read from the controller, the controller detected is read from the controller and the Model is an internet lookup based on the serial number. If your controller does not have a serial number programmed into the controller then this usually indicates that it has been replaced. You can add a serial number on the advanced settings page if yours is missing from the controller. The model lookup will only occur when the version number changes (to minimize internet lookup querys) so once you add a serial number then you can edit the file /etc/genmon/genmon.conf and remove the lines 

model = Generic Air Cooled

and 

version = V1.19.04

Save the file, then if you restart genmon it will use the serial number you provide to perform a look up on your model number.

I am not sure why it is stating Generic Air Cooled, it should say Generic Liquid Cooled if it can not determine the model due to a lack of a serial number. I you send your logs I can look into that. Is your controller detected properly?

If you outbound email is setup and working you can submit your log files from the About page.

Also, the Run hours in the last 30 days and year are calculated based on the genmon power log, so this data is only valid for the time periods that genmon has been installed.

heffneil, 2024-08-27:
I submitted over the weekend but I just submitted the registers and all again.  Maybe that will help. No idea.

I didn't edit: /etc/genmon/genmon.conf 

I don't believe I changed anything in the advanced settings (but the baud rate we discovered over the weekend)

jgyates, 2024-08-27:
I got your registers and logs, thanks. This helps as I only have one other set of registers for a 4.5L unit. I got the one on the weekend, it did not have any real data except your communication stats as you were not communicating with the controller at the point. You logs look good. I am guessing what happened is this:

Genmon will only check for model info when the version info changes. When you started genmon initially, due to communication issues it was unable to identify the controller or model so it populated it with the default values (air cooled). Until the version number changes (usually via an upgrade) the model check will happen again (until a valid model is detected) at every version change detected. 

You can delete the line "version = V1.19.04" in the /etc/genmon/genmon.conf and restart and it will likely put "Generic Liquid Cooled" for the model, or you can wait until a new version is released and it will updated automatically, or you can follow the instructions above and add your own serial number.

Let me know if that does not make sense. I am going to close this thread but feel free to post here with any updates or additional questions on this topic.

closed 2024-08-27 by jgyates

heffneil, 2024-08-27:
I removed the version line and it does indeed show the proper engine.  I have to find my serial number to manually enter it I guess - not that I think its terribly important.

Maintenance :
    Model : Generic Liquid Cooled
    Generator Serial Number : None - Controller has been replaced
    Controller Detected : Evolution 4.5L, Liquid Cooled

jgyates, 2024-08-27:
yeah, it does not get you any new features in the software, mostly for maintenance reasons. You can find it on the name plate, usually inside the generator encolosure. It usually looks like this:

[image]

heffneil, 2024-08-27:
This is an unrelated question but who maintains the MQTT add-on? I ask because its my understanding the retains are an issue now that I have it working.  Thanks!

jgyates, 2024-08-27:
I maintain most things but if anyone wants to contribute a PR is always welcome.

heffneil, 2024-08-27:
I wish I could help.  I can use AI and test :)

heffneil, 2024-08-27:
On Line 622 of genmqtt.py we change the code from:

self.MQTTclient.publish(FullPath, value)

to

# Set the retain flag to True for retained messages
self.MQTTclient.publish(FullPath, value, retain=True)

I don't know how to do a pull and all that jazz but that's what I am testing with now.  I'll update you as I see how it works.

jgyates, 2024-08-27:
I can make this update and add it as an option to the add in setting on the Add On page of the web interface.

jgyates, 2024-08-27:
It will probably be tomorrow before I can add this

jgyates, 2024-08-28:
I added this feature. Update from the About page, then you can enable it from the Add On page. The version number has not changed. Let me know how it goes.

heffneil, 2025-02-25:
Hello sorry to reopen this but my messages in my mqtt server aren't being retained.  When I look at them in mqtt explorer it should show retain but it doesn't.  I just upgraded to the latest version.  My settings show retain and I have the following at the bottom of my genmqtt.conf

```
# unique identifier for each instance of genmon running. This can be any value
# if you are running one instance of genmon. Otherwise each instance must be
# unique
client_id = genmon
numeric_json_object = false
retain = true
```

jgyates, 2025-02-25:
Just to clarify, a message marked retained will not retain all instances of that message, only the last one. Are you getting any messages? Also, what is the flush interval in the add on settings?

heffneil, 2025-02-25:
I modified my file to add the retain = true I just wanted to make sure that didn't break anything.  I am getting very minimal data only a few items. I am looking at the latest.
Flush Interval is 0

jgyates, 2025-02-25:
With a flush interval of zero the add on will only send data if the value changes. If your MQTT server was restarted or started after genmon then I would expect that you do not have values for all of the settings with a flush interval of zero. If you set flush interval to some number of seconds, for example 3600 would be a flush interval of one hour, then this would make the add on resend the values every hour even if they have not changed. Values that do not change (like the firmware version or controller type) would never be present if the MQTT sever was started after genmon. Genmon sends all the values via MQTT when it starts but if the MQTT server is restarted it will not send them again until they change if flush interval is zero.