jgyates/genmon #1175: MQTT Briggs & Stratton Missing Messages

Mundane0, 2024-09-09:
Just got this all up and running and when checking the MQTT messages with MQTT Explorer there are somethings that seem to be missing from the list.

[image]

there are no MQTT messages for Utility Frequency 1 or 2, Generator Frequency 1 and 2, Generator L1-L2 Voltage, Generator L1 and L2 Amps. All of which function fine when doing a test run from the main status page.

[image]

is this something I can configure on my end or is this part of the Briggs_Stratton_GC-1032.json file?

thanks in advance

jgyates, 2024-09-09:
this issue here is that the MQTT add on will only take the text output from genmon (not the gauge data) and output it as MQTT topics. I can add these values to the text output and that should allows these values to show up via MQTT. I should be able to do this in the next day or 2.

closed 2024-09-09 by jgyates

Mundane0, 2024-09-09:
Update from the about page did not work, not sure if it is because I am running Genmon in docker, was able to manually copy and paste the new json file into my docker image and restart Genmon to get it working. Thanks for the quick reply, now to finish my Home Assistant dashboard.

jgyates, 2024-09-09:
For docker it depends on how it was setup. For the update to work you do have to check out the repository with git as described [here](https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software). The update is just using the git command to update directly from github. Just copying the repository will not allow the administrative git files to be created.

Also, I did not update the version number so the current version is still 1.19.04.