jgyates/genmon #1181: Generator L1/L2 Amps displaying wrong

F4iguy69, 2024-09-21:
### Steps to Reproduce

When generator is under load the graphically display always shows Generator L1 and L2 amps as 60, regardless of what the amps actually are.  When looking under the Engine section, the amps under Generator L1/L2 is correct.   Briggs and Stratton 22KW GC-1032 Controller

### Expected Result

Expect the graphic display to show correct amps. 

### Actual Result

Graphic only shows 60amps. 

### Version Info

V1.19.04

### Logs

Submitted via about section.

jgyates, 2024-09-22:
I got your logs. I took a look at how the registers are currently mapped to the gauges and the text data and everything looks OK. Can you send me your log files while it is running and you are seeing the issue? When you send the logs it also sends a copy of your registers so this will allow me to see what you are seeing.

F4iguy69, 2024-09-22:
New logs sent with home on generator power.  Also attached is a screen shot of what I’m showing.  
[image]

jgyates, 2024-09-22:
Thanks for the data. This helped me isolate the issue. I checked in an update that should fix the problem for you. The original controller definition file had multiple gauges named the same thing (Generator L1 and Generator L1). Each gauge has to be named a unique value as they are referenced by name when updating the value. Once I corrected this it works as expected. Let me know if you have any other issues.

closed 2024-09-22 by jgyates