jgyates/genmon #1184: Preparing metadata fails

chriskinal, 2024-09-25:
### Steps to Reproduce

1. Install latest Debian Bookworm
2. Clone and install Genmon

### Expected Result

Genmon installs successfully.

### Actual Result

Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [44 lines of output]
      + meson setup /tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458 /tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458/.mesonpy-d5alm7o2 -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458/.mesonpy-d5alm7o2/meson-python-native-file.ini
      The Meson build system
      Version: 1.5.2
      Source dir: /tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458
      Build dir: /tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458/.mesonpy-d5alm7o2
      Build type: native build
      Project name: scipy
      Project version: 1.14.1
      C compiler for the host machine: cc (gcc 12.2.0 "cc (Debian 12.2.0-14) 12.2.0")
      C linker for the host machine: cc ld.bfd 2.40
      C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14) 12.2.0")
      C++ linker for the host machine: c++ ld.bfd 2.40
      Cython compiler for the host machine: cython (cython 3.0.11)
      Host machine cpu family: x86
      Host machine cpu: i686
      Program python found: YES (/home/chris/genmon/genenv/bin/python)
      Did not find pkg-config by name 'pkg-config'
      Found pkg-config: NO
      Run-time dependency python found: YES 3.11
      Program cython found: YES (/tmp/pip-build-env-ouv5955s/overlay/bin/cython)
      Compiler for C supports arguments -Wno-unused-but-set-variable: YES
      Compiler for C supports arguments -Wno-unused-function: YES
      Compiler for C supports arguments -Wno-conversion: YES
      Compiler for C supports arguments -Wno-misleading-indentation: YES
      Library m found: YES
      
      ../meson.build:84:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['ifx'], ['g95']]
      The following exception(s) were encountered:
      Running `gfortran --version` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `gfortran -V` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `flang --version` gave "[Errno 2] No such file or directory: 'flang'"
      Running `flang -V` gave "[Errno 2] No such file or directory: 'flang'"
      Running `nvfortran --version` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `nvfortran -V` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `pgfortran --version` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `pgfortran -V` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `ifort --version` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifort -V` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifx --version` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `ifx -V` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `g95 --version` gave "[Errno 2] No such file or directory: 'g95'"
      Running `g95 -V` gave "[Errno 2] No such file or directory: 'g95'"
      
      A full log can be found at /tmp/pip-install-3tobli1y/scipy_70f15e20cca141efaf82671306e2d458/.mesonpy-d5alm7o2/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

### Version Info

Python 3.11.2
Genmon 1.19.04

Internet searches on the issue seem to indicate dropping Python back to 3.10.x solves the issue.

jgyates, 2024-09-25:
Is this an x86 based system you are installing? 

Also, is the 32 bit or 64 bit debian? 

Your info above does not indicate what part of the install is failing.

I know on the ARM variants the sometimes the 32 bit version will have issues but he 64 bit variant will work fine. In the past when an issue with the installation of packages has come up they are usually resolved by the package maintainer. 

Since python is installed by the OS / distribution and not genmon install script you may want to try porting back to pytyhon 3.10 but there is a chance that that would break other things.

chriskinal, 2024-09-26:
Sorry. x86-64. I solved the immediate issue by installing Debian 11 which uses Python 3.9. The VM has been destroyed so I can't get the full log file. I can recreate the VM if needed.

closed 2024-09-26 by jgyates