jgyates/genmon #1186: New Error "Under Voltage : Alarm Code: 0000"

eBoon123, 2024-09-26:
### Steps to Reproduce

Happens as part of weekly scheduled exercise. This just started showing up since August. I have checked the panel on the generator and the Mobile Link app - nothing shows up there.

Is this an actual error?

### Expected Result

No Error or something to see on the panel or on Mobile Link app.

### Actual Result

[image]

### Version Info

Version V1.19.04

### Logs

Sent 2024-09-26 @ 16:04 EST

jgyates, 2024-09-26:
Hi @eBoon123 

The entries on the log page are read directly from the controller. If you look in the menu on the screen on the controller you can validate this. Look at the actual alarm log, rather than looking for an active alarm.  The alarm log should be under History in the controller menu. 

My guess is that these are events rather than alarms (since the alarm code is 0000). They also look like they all fall at the beginning of your scheduled exercise cycle, possibly there is some check in firmware that validates the output voltage and your system is not getting to that point soon enough, just a guess.

I received your logs with the register data and in my simulator I get the same results, which are read from the registers you provide. You can see all the registers, with the log registers with this command (assuming http with an IP of 192.168.1.10 of the pi):

       http://192.168.1.10:8000/cmd/allregs_json

Once you show these registers you should see this:

"Log Registers": {"Logs": {"Alarm Log": ["03e8:32081e0c092a18160000", "03ed:32071e0c0922180f0000", "03f2:32061e0c091b18080000", "03f7:32051e0c080a18190000", "03fc:32041e0c082e18040000", "0401:32031210083118010000", "0406:4802150a091d15160af0", "040b:48010000040114010af0", ........

This represents the first 8 entries of the alarm log (there are a total of 50, the rest of them are blank). These values translate to the entries displayed on the log page. 

In short this means that genmon is displaying what the controller is giving it. The controller may not be classifying it as an alarm, if it did then genmon would notify you an alarm has occurred as the contents of the alarm log and an active alarm are notified differently by the controller, but the controller does appear to be adding these entries in your controller based alarm log.

Let me know if you have any other questions or if you do not see these values in the alarm log on the controller.

closed 2024-09-26 by jgyates

skipfire, 2024-09-26:
@eBoon123 I would suggest starting the generator for a few minutes (no need to transfer) and see what is reported for the output voltage and make sure that it reports around 240v.  If it doesn't present the correct voltage, more diagnostics may be needed.

eBoon123, 2024-09-27:
> @eBoon123 I would suggest starting the generator for a few minutes (no need to transfer) and see what is reported for the output voltage and make sure that it reports around 240v. If it doesn't present the correct voltage, more diagnostics may be needed.

I manually started it - no issues - no alarm code. Output voltage was 240v

lwbeam, 2025-01-07:
@eBoon123 
> OK - took a look at the Alarm Log - and it shows "TRANSFER WIRE WARN" every time that Genmon shows the Under Voltage log message. Looks like I probably need someone to come and take a look.

I know it's been a while, but if you haven't had someone have a look at it, don't waste your money. The problem may simply be that you need a new battery. I'll explain.

Last June I had to replace the battery in my 2023 14kW Generac unit (I'm cheap, and simply used the 6 year old battery from my old machine). When I went to purchased the new battery I wasn't paying attention and pulled the wrong one off the shelf. I ended up with a Group 26R, 510CCA battery, instead of one with the specified capacity of 540CCA. So I saved CAN$18 (US$12.50). Seemed to work fine. That was until December when the temperature dropped to -20°C (-4°F). That's when I also started to get "Under Voltage : Alarm Code: 0000". Once during a test run, and once during one of several back-to-back power outages. As the generator is located at my seasonal residence, which isn't normally occupied during the winter, today was my first opportunity to check the Alarm Log. Sure enough it shows "TRANSFER WIRE WARN" both starts.

When I saw this thread in December, I started looking through the Diagnostic Repair Manual for my machine. **Test 25 - Check Wire 23** (page 97) discusses the procedure to ensure that Wire 23 has been connected correctly. Wire 23 provides the signal to the transfer switch to change state (i.e., open or close), after the ganerator has started. The procedure states: "This test assumes that battery voltage is sufficient (at least 12.2 VDC)." From the e-mails sent by Genmon, the battery voltage had dropped to 12.7 and 12.8 volts during the two starts. From that I concluded that the controller throws the "Under Voltage : Alarm Code: 0000" message when it _thinks_ the battery voltage might drop too low for it to successfully operate the transfer switch. This is entirely undocumented, but the threshold for the alarm appears to be about 0.5 volts above that needed to operate the transfer switch contactor. I would have posted this earlier, but I wanted to check the Alarm Log first.

So if you're getting this alarm, and there's nothing wrong with the connections on Wire 23 (easy enough to check), you probably need a new battery (also DIY).

jgyates, 2025-01-07:
@lwbeam 

thanks for the info!!