jgyates/genmon #1187: power output

danielsnapon, 2024-09-27:
### Steps to Reproduce

[image]
[image]
current displayed is mutch lower than the actual current(watts) 32kw generac.

### Expected Result

can the output current be calabrated?

### Actual Result

[genmon_logs.tar.gz](https://github.com/user-attachments/files/17157789/genmon_logs.tar.gz)

### Version Info

1.19.04

jgyates, 2024-09-27:
You likely will need to tune genmon to compensate for the Hall Effect sensor in the liquid cooled models not reporting the current accurately. There appears to be a lot of variance in firmware / hardware accuracy for Evolution Liquid Cooled models. See [number 5 in the Known Issues wiki page](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues#5-current-and-power-output-values) for more info. Also, [these threads](https://github.com/jgyates/genmon/issues?q=is%3Aissue+label%3APowerDisplay+Hall) have more info on power display on a liquid cooled Evolution.

I have a liquid cooled generator and opt to use external CTs and import the data into genmon with the MQTT input add on as the built in Hall Effect sensor is not as accurate as current transformers. Generac air cooled models use CTs instead of Hall Effect Sensors.

jgyates, 2024-09-27:
I am going to close this issue as I think it is covered by the wiki. Let me know if you have any questions.

closed 2024-09-27 by jgyates

danielsnapon, 2024-09-27:
Is that an option that could be added to the Settings page?

 

Dan Anderson

Cell      (352) 266-4480

Home  (352) 694-4575

* ***@***.*** ***@***.***>

jgyates, 2024-09-27:
the settings for this (Current Divider and Current Offset) can be made on the advanced settings page. Double click the icon in the top right of the web interface to access the advanced settings.

danielsnapon, 2024-09-27:
I will start playing with the settings and see if I can get it close.

 

Thanks

 

Dan Anderson

Cell      (352) 266-4480

Home  (352) 694-4575

* ***@***.*** ***@***.***>