jgyates/genmon #1188: Pre-Nexus Controller. Genmon won't report RPM or frequency to webui it's just stuck at 0 rpm and 0 Hz but everything else works and monitors fine

TheawesomeMCB, 2024-09-28:
### Steps to Reproduce

1. Plug into control panel
2. boot up genmon
3. turn on generator so engine is running
4. No output on page

### Expected Result

It shows rpm and frequency

### Actual Result

It's at 0 for both
[image]

### Version Info

V1.19.03

jgyates, 2024-09-28:
I would need to see your logs/registers, which can be submitted from the about page in the web ui.  Preferably submit the logs when your generator is running. 

You need to have outbound email setup and functional on the settings page to submit your logs/registers.

This is just a guess but it could be that some versions of the Pre Nexus firmware did not support full functionality.

TheawesomeMCB, 2024-09-28:
> I would need to see your logs/registers, which can be submitted from the about page in the web ui. Preferably submit the logs when your generator is running.
>
> You need to have outbound email setup [...]

Okay I just sent them. It's the one that was submitted at 2:33:11PM

jgyates, 2024-09-29:
I got your logs. It looks like the modbus registers for RPM and Frequency are all zeros. I also compared your firmware version (1.10) to the other two sets of registers that people have submitted for Pre-Nexus models. Yours is not the lowest firmware. I do not have another set of Pre-Nexus registers while the generator is running so this could be a limitation of the pre-nexus firmware. 

If anyone else using a Pre-Nexus with genmon can weigh in that would be helpful.

TheawesomeMCB, 2024-09-29:
> I got your logs. It looks like the modbus registers for RPM and Frequency are all zeros. I also compared your firmware version (1.10) to the other two sets of registers that people have submitted [...]

Soooo good news, it actually did start reporting. So compared to the other gauges it takes anywhere from 45-60 seconds before it starts reporting the RPM and frequency. It updates fast enough that the generator will say: “Running” but in terms of the rpm and hertz it takes a bit before it starts reporting that. So there’s just an massive delay before it starts reading the rpm and frequency.

[image]

jgyates, 2024-09-29:
That is great. Thanks for letting me know. I am going to close this for now but feel free to post any follow up to this thread.

closed 2024-09-29 by jgyates