jgyates/genmon #1191: Pre-Nexus controller keeps saying inspect spark plug and inspect air filter with genmon plugged in

TheawesomeMCB, 2024-10-03:
### Steps to Reproduce

1. Plug genmon into Pre-Nexus controller.
2. Will take from 12-24 hours
3. Inpect warning
This happens and tried on 2 different pre-nexus controllers and only happens when they are plugged in

### Expected Result

Not having to remove the warning every single single day and not having the warning pop up at all

### Actual Result

Warning appears every day even after warning cleared

### Version Info

1.19.03

### Logs

[genmon_logs.tar.gz](https://github.com/user-attachments/files/17239794/genmon_logs.tar.gz)

skipfire, 2024-10-03:
I haven't heard of other reports for that.  Were you using the same hardware on both generators, or different hardware?

jgyates, 2024-10-03:
A few questions:

You mention that you are running version 1.19.03. I would update to the latest version first.

It would be helpful to get a copy of your registers which can be submitted from the About page. Outbound email must be enabled to submit registers. Sending your logs this way also includes the register data and a log of any errors in genmon software updates.

Are you seeing the spark plug and air filter warning in the alarm log or is this an active alarm? The controller reports these two types differently.

How are you clearing the warning?

You do have an issue in your /var/log/genmon.log file. This is the error:

2024-10-03 01:30:26,983 : Error in GetServiceDueDate: timestamp out of range for platform time_t : generac_evolution.py:4619

This may be due to the pre nexus not supporting reading the service due counter via modbus. Your registers would validate this. This could also be due to the fact that you do not have the latest version of the software.

jgyates, 2024-10-03:
also, when I test with other sets of submitted registers I do not see any issues like this.

If you are seeing this on two different sytems, then submit the registers and logs on both systems.

TheawesomeMCB, 2024-10-03:
> A few questions:
>
> You mention that you are running version 1.19.03. I would update to the latest version first.
>
> It would be helpful to get a copy of your registers which can be submitted from the [...]

1. Updated
2. Just sent, it should come from an gmail address
3. active alarm, enough that it sends an alarm through the slack addon
4. pressing enter then escape

The previous control panel is in an different generator, but it's the same one from Sat Sep 28th Time: 02:45:55 PM with log being at Day and the Time: 02:33:02 PM, like in the previous issue I was having with the RPM and Hz that we ended up figuring out what was going on.
(I had to change the current ones to an gmail address because for some reason outlook didn't like it, could never figure out why so I just switched the email)

jgyates, 2024-10-03:
Can you send your logs now that you have updated. The send registers button only sends registers. The send logs button sends logs and registers.

Also, can you look at the alarm log (both on genmon and on the controller) and see if these events are getting logged?

TheawesomeMCB, 2024-10-03:
> Can you send your logs now that you have updated. The send registers button only sends registers. The send logs button sends logs and registers.
> 
> Also, can you look at the alarm log (both on genmon and on the controller) and see if these events are getting logged?

So I sent them, and I was looking at the alarm log on the generator and genmon and both are being logged. Also with that error timestamp out of range I went ahead and on the generator reset the maintenance time to see if that helps. So I'll keep an eye on it for the rest of the day to see if that helped. I talked to my dad which is an electrician and said that every time we push enter then esc it should of been resetting the maintenance time but maybe wasn't correctly maybe? I'll keep an eye on it and let you know if it does it again. But I just find it odd that it was doing on both controllers.

jgyates, 2024-10-03:
I got your logs. Looks like the update to 1.19.04 fixed the "GetServiceDueDate" error in your logs.

These warnings are triggered either by a specifc number of hours or a specific date (e.g. 200 hours or 12/12/24). You can also look on the maintenance page near the bottom and it should show the number of hours and time for each maintenance timer (spark plug, air filter, etc).

Here is the snap shot of my simulation with your registers you submitted today.
[image]

It looks like oil and filter were reset yesterday and battery service was reset today (hence the one year from now warning timer).

I am going to close this issue for now but feel free to post any follow up to this thread.

closed 2024-10-03 by jgyates

TheawesomeMCB, 2024-10-04:
It’s now doing the alarm again, now for oil and filter despite me resetting the maitnance yesterday. I’ve sent logs again
[image]

jgyates, 2024-10-04:
Have you validated that you do not get this alarm when genmon is unplugged? You would need to check it manually on the controller.

How are you powering the pi?

jgyates, 2024-10-04:
Also, you could try power cycling your controller. I don't know how to do this with the pre-nexus but this is the procedure for the Nexus and Evolution. 

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

another thing to check is to make sure the alarm entry names on the controller match the log entries names that genmon is reading. Each log entry has a code that represents the alarm in the alarm log. Genmon assumes that all the log codes are the same for Nexus and Pre-Nexus.

TheawesomeMCB, 2024-10-04:
> Have you validated that you do not get this alarm when genmon is unplugged? You would need to check it manually on the controller.
> 
> How are you powering the pi?

8/2/24 and before was when the pi was NOT plugged in
9/30/24 and after WAS when the pi was plugged in.

[image]

The pi is being powered by the battery terminals, when I first tried it the pi wouldn’t turn on showing this controller doesn’t give power through the port so I had to connect the 12v plug through the pi.

I went ahead and power cycled it by removing the fuse and putting it back in. Also when the alarm went off I pressed enter and then enter again to clear the error instead of enter then esc. 
I’ll watch it for the day again, but I have an feeling if pressing enter twice didn’t stop the alarm coming back it’s something else

TheawesomeMCB, 2024-10-04:
And yes Genmon and the controller on the generator alarm matches.

jgyates, 2024-10-04:
Also get a data point for removing genmon and see if you still get the alarm. This will rule out the the behavior started independently of genmon being attached.

TheawesomeMCB, 2024-10-04:
Well I unplugged the Genmon, for now and I’ll keep an eye on it. But all this did start happening at the very beginning when Genmon was plugged it, as shown by the logs. It never did this before

jcleek, 2024-10-04:
@jgyates I just took a look and my generator panel is fine, no errors or warnings on it or the GENMON interface. As you know I have been running it for honestly as long as I can remember without much issue. Photo of panel I just took and screenshot of interface below. Let me know if you want logs or any info.
[image]
[image]

jcleek, 2024-10-04:
And my generator is old as dirt.

jgyates, 2024-10-04:
@jcleek , thanks for that data point

TheawesomeMCB, 2024-10-04:
This is so weird, why is mine the only one that’s doing it. It shouldn’t matter what color end goes in the pi right?
[image]
I previously had it so the white wire on the top left end was plugged into the pi. But maybe I can try switching it to this end so the white wire is on the top right maybe? That will have to wait until tomorrow afternoon to see if the alarm comes up again with the pi unplugged. And if the alarm doesn’t come up again and then I plug it in again and the alarm comes up again before 24 hours, then there is something about my generator alone that it doesn’t like about the pi.

skipfire, 2024-10-04:
The unmodified Molex cables are reversible so it doesn't matter which end you plug into the generator.  Typically if it is a data cable problem you see either no data or intermittent losses, not a real error.

TheawesomeMCB, 2024-10-04:
Oh okay then, anyway maybe we will know for sure if it is the pi is causing the oil, air filter and spark plug alarm by tomorrow if it does throw any alarms till afternoon and then try 24 hours with the pi in to see if it throws an alarm. If what I said makes sense but you get my point. 😅

TheawesomeMCB, 2024-10-05:
It’s been 24 hours and still green, no alarms. Now I’m going to connect the pi to it and see if within 24 hours an alarm goes off or not
[image]

TheawesomeMCB, 2024-10-05:
So shortly connecting the pi, I noticed something… interesting.
Look at these service times

2:43 PM 
[image]
2:44 PM
[image]
2:45 PM
[image]
2:46 PM
[image]
2:48 PM
[image]
2:49 PM
[image]

So it’s like having the pi connected is scrambling the maintenance times. And as I’m writing this the maintenance alarm went off again.
Also on the previous controller that I don’t have in anymore, having the pi plugged in actually caused the screen to go crazy, like on boot the checksum would fail and the only way to get the generator was to unplug the pi from it. And the screen had symbols like arrows and number symbols go crazy all over the screen.
[image]

jgyates, 2024-10-05:
I wonder if this is a power issue, possibly the pi is causing electrical issues. Try stopping genmon, but leave the pi attached. You can log in via ssh the change directories to the genmon folder:

     cd genmon

Then type this to stop genmon:

     ./startgenmon.sh stop

To restart type:

      ./startgenmon.sh restart

If you get the alarms with the pi attached but without genmon running then it is likely electrical vs a software issue.

skipfire, 2024-10-05:
On the chance that it's a hardware issue I can ship out a replacement HAT if you contact me through email replying to the order or shipment email, my [website](https://pintsize.me/), or [facebook messenger](https://www.facebook.com/profile.php?id=61555703325298) so I can find your order and confirm your address for shipment.  I don't think it's a hardware issue since it has a connection and seems to get valid data, but swapping the HAT (or pHAT) is fairly easy to try.

jgyates, 2024-10-05:
The only reason I suggested a hardware issues is that I know pre nexus and nexus appear to be more sensitive to anything that pulls power away from the battery, based on past posts.

skipfire, 2024-10-05:
Shutting down genmon and leaving the hardware connected does rule out software and is a test that can be conducted now, best case on hardware change is Wednesday (ship out Monday and receive Wednesday if in the US).

@TheawesomeMCB how old is the battery (manufacture date not install date) and what Pi model do you use?  Power draw is different on Pi models and if it does turn out to be an issue with the drain on the battery that may be one option.  I have seen 2 instances where having a Pi hooked up to an aged battery showed battery end of life before the generator detected it.  If you have an Ethernet cable run to the generator another would be putting the equipment inside on a UPS with AC power instead which would take the full load off the battery.

TheawesomeMCB, 2024-10-05:
The sticker on the battery shows 5/21 so May of 2021 but while monitoring it the battery never went below 12v. The pi not running genmon but still plugged in and on and connected to the controller, the alarm went off again. I can try running an extension cord to the generator and plug the pi into an outlet. Let me try that.

TheawesomeMCB, 2024-10-05:
>what Pi model do you use?

The 3B+

jgyates, 2024-10-06:
also, a few photos of how things are connected may help.

TheawesomeMCB, 2024-10-06:
[image]
Well it's been 5 hours and the system still hasn't gone into alarm yet and the service times isn't jumping all over the place. So if the system doesn't go into alarm overnight will determine that if having the pi connected to the battery was the problem. And if it doesn't go into alarm overnight then we will know that was the problem and we will get an outlet installed just for the pi, and we will know for some reason my specific pre nexus controllers doesn't like the pi using it's battery.

jgyates, 2024-10-06:
Thanks for the info. Let us know if the problem comes up again.

skipfire, 2024-10-06:
@TheawesomeMCB the if you run a power outlet to the generator keep in mind that with the Pi not being powered by the battery it will shut down with every power blip and with outages and you will have to wait for it to get booted back up before you can access it. 

If you are considering trenching a line, you may want to consider a Cat5/6 cable instead which gives you a couple of options and eliminates the need for figuring out backup power for the Pi at the generator.
1. Running Ethernet from a PoE switch and getting a PoE HAT would provide power for the Pi and give you an Ethernet connection rather than WiFi.
2. You could use my RJ45 kit to run the Molex wires over Cat5/6 as a cable extender into the house and have the Pi inside with your other network equipment on a UPS, this has the advantage of the Pi being easily accessible (especially in bad weather) and it puts the Pi inside your nicely climate controlled house.

skipfire, 2024-10-06:
@TheawesomeMCB can you contact me through email replying to the order or shipment emails, my [website](https://pintsize.me/), or [facebook messenger](https://www.facebook.com/profile.php?id=61555703325298) so we can discuss a couple of hardware options?  I want to make sure you end up in a reasonable end state from a hardware standpoint.

TheawesomeMCB, 2024-10-06:
@skipfire
I appreciate it but as long as the pi boots and sends an message that the generator is on after the power goes out once the pi gets going it's all good.
Like so this is the current setup, the 4 port is in its own room, the 8 port is in another room, and the 2+4 switch is in the same room:
[image]
And if we were to use injectors and an UPS to get this this going this is what we would have to do:
[image]
So we would need to buy 2 injectors and 2 splitters and another UPS on top of it if we wanted to keep it on at all times to keep the network going for it. Even with the Pi needing to boot again and communicate with the generator, as long as soon as it boots and instantly says the generator is on when the other 3 switches boots then that's fine. It's pretty much much no different right now when the pi was connected to the generator battery because it still needed to wait for the other 3 switches to start up anyway. (Oh if you are wondering why I have an 4 swtich plugged into an 2 switch, that's because that 2 switch is an deco mesh that only has 2 ports so we needed to get more ports, so I just plugged another router/switch combo unit and turned off the wifi on it and put it in AP mode so it's just basically an 4 port switch.)

skipfire, 2024-10-06:
Ok, please let me know if you change your mind.

TheawesomeMCB, 2024-10-06:
@skipfire 
Appreciate it!
Oh and also @jgyates @skipfire 
[image]
It’s been around 24 hours and the system still hasn’t gone into alarm.
So it’s something about this controller that doesn’t like the pi using its generator battery that was causing the problem.