jgyates/genmon #1192: Parity Error

F4iguy69, 2024-10-03:
### Steps to Reproduce

Change the parity under advance setup to any value other than NONE. 

### Expected Result

Expected everything to continue to work. 

### Actual Result

Once this change is made, Genmon site will no longer load. 

I’m trying to experiment with a HAT and connect the Pi directly to the generator to see if it helps with some of the false errors I received via wifi and serial over IP.  

I’m on a Briggs and Stratton and need the parity set to EVEN.  

I’ve tried it many times and as soon as I make the change from NONE, then Genmon page will no longer load.  I run the script under troubleshooting section that reloads the config files and then everything will load again.  

### Version Info

1.19.04

jgyates, 2024-10-03:
you are likely experiencing a long delay in loading due to communication timeout errors. Since you are trying to connect to a briggs unit, I would set it to even, restart and wait 5 - 10 min and see if the web interface comes up. The web interface will not load until genmon has finished trying at least once to identify the controller. If your serial comms are not working then each modbus transaction will time out (3-5 seconds for each transaction with about 80 registers). once all the registers have tried to be read then the web interface will load with default values serial comms are not active.

You can look at the files /var/log/myserial.log and /var/log/mymodbus.log for info regarding why your serial comms are not working. 

Make sure your serial port is correct, your data rate and parity match your controller settings outlined here:

https://github.com/jgyates/genmon/wiki/Appendix-P-Briggs-and-Stratton-Controller-Information

Let me know if you have any other questions or if after waiting your ui still does not load.

jgyates, 2024-10-04:
I am going to close this issue but feel free to post any follow up status or questions to this thread.

closed 2024-10-04 by jgyates