jgyates/genmon #1195: Setup fails if Python isn't installed first

skipfire, 2024-10-09:
### Steps to Reproduce

1. Use an OS that doesn't have python installed by default such as dietpi
2. Run the install, any arguments.

### Expected Result

Genmon installs successfully

### Actual Result

genenv fails to be created because python isn't installed, python appears to be created in subsequent steps.  Likely the call for apt to install python and whatever else is needed for the virtual environment creation needs to be called prior to trying to create the virtual environment while the remainder of the dependencies can be called later.

### Version Info

V1.19.04 source pulled October 8, 2024

jgyates, 2024-10-09:
I will probably not add a the explicit installation of python to the script but rather a note in the install instructions. I can add a check and instruct them to install python. Python is included in all major linux distributions with the exception of ones that are intentionally trying to be minimalistic. Installing python likely has larger SD card requirements and has some options on what to install so I will leave that up to the end user. If the installer has the wherewithal to use a non Raspberry pi distro then they should be able to handle installing python

skipfire, 2024-10-09:
That would be fine.  I didn't even realize it wasn't there, installing it myself isn't a problem I just didn't realize it and then had a heck of a time getting the venv built out correctly since it was half-built.

jgyates, 2024-10-09:
I checked in an update that incudes this check to genmonmaint.sh. I also updated the wiki with system requirements:  https://github.com/jgyates/genmon/wiki#system-requirements

Thanks for finding this.

closed 2024-10-09 by jgyates