jgyates/genmon #1197: Losing run hours and power_log_json on reboot

skipfire, 2024-10-10:
### Steps to Reproduce

I installed on a NanoPi Neo Plus2 with DietPi and everything seems to be working, except run hours on the maintenance tab and power_log_json get reset when I restart genmon.  kwlog.txt has the data and gets updated while running, and the power graph shows it, but it seems like the power graph is the only thing that knows about it.

I did copy config and logs from my prior install.

### Expected Result

Power data stays available across reboots.

### Actual Result

power_log_json after restarting genmon, it had non-zero data for a transfer I did today to verify this wasn't a one-time issue prior to restarting:

[["10/10/24 15:27:49", "0.0"], ["10/10/24 15:24:48", "0.0"], ["10/09/24 20:07:51", "0.0"], ["10/09/24 20:03:40", "0.0"], ["10/07/24 15:51:27", "0.0"], ["10/07/24 15:30:46", "0.0"], ["10/07/24 15:02:16", "0.0"], ["10/07/24 15:01:55", "0.0"], ["10/07/24 15:01:16", "0.0"], ["10/07/24 14:55:45", "0.0"], ["10/07/24 10:05:25", "0.0"], ["10/07/24 10:04:44", "0.0"], ["10/07/24 10:04:05", "0.0"], ["10/07/24 10:01:54", "0.0"], ["10/07/24 10:01:15", "0.0"], ["10/07/24 10:00:44", "0.0"], ["10/01/24 09:30:00", "0.0"], ["10/01/24 08:30:39", "0.0"], ["09/30/24 10:04:15", "0.0"], ["09/30/24 10:00:44", "0.0"], ["09/23/24 10:06:11", "0.0"], ["09/23/24 10:02:00", "0.0"], ["09/16/24 10:06:00", "0.0"], ["09/16/24 10:01:29", "0.0"], ["09/11/24 19:11:09", "0.0"], ["09/11/24 18:11:47", "0.0"], ["09/09/24 10:04:37", "0.0"], ["09/09/24 10:01:16", "0.0"], ["09/08/24 18:20:13", "0.0"], ["09/08/24 18:19:02", "0.0"], ["09/08/24 10:56:22", "0.0"], ["09/08/24 10:55:31", "0.0"], ["09/02/24 10:05:29", "0.0"], ["09/02/24 10:05:18", "0.0"], ["09/02/24 10:04:59", "0.0"], ["09/02/24 10:01:48", "0.0"], ["09/02/24 10:01:39", "0.0"], ["09/02/24 10:01:18", "0.0"], ["09/01/24 15:32:35", "0.0"], ["09/01/24 15:30:14", "0.0"], ["09/01/24 10:29:54", "0.0"], ["09/01/24 09:31:23", "0.0"], ["09/01/24 09:30:44", "0.0"], ["09/01/24 09:30:23", "0.0"], ["08/31/24 13:24:59", "0.0"], ["08/31/24 13:24:38", "0.0"], ["08/26/24 10:05:14", "0.0"], ["08/26/24 10:01:23", "0.0"], ["08/26/24 10:00:54", "0.0"], ["08/26/24 10:00:33", "0.0"], ["08/25/24 12:03:59", "0.0"], ["08/25/24 12:02:08", "0.0"], ["08/21/24 11:35:10", "0.0"], ["08/21/24 11:34:49", "0.0"], ["08/19/24 10:05:30", "0.0"], ["08/19/24 10:01:09", "0.0"], ["08/12/24 10:05:20", "0.0"], ["08/12/24 10:00:59", "0.0"], ["08/12/24 10:00:50", "0.0"], ["08/12/24 10:00:29", "0.0"], ["08/05/24 10:06:16", "0.0"], ["08/05/24 10:01:35", "0.0"], ["08/01/24 10:30:01", "0.0"], ["08/01/24 09:31:19", "0.0"], ["08/01/24 09:30:50", "0.0"], ["08/01/24 09:30:39", "0.0"], ["07/29/24 10:06:05", "0.0"], ["07/29/24 10:01:44", "0.0"], ["07/29/24 10:01:25", "0.0"], ["07/29/24 10:01:14", "0.0"], ["07/22/24 10:05:59", "0.0"], ["07/22/24 10:01:58", "0.0"], ["07/15/24 10:05:44", "0.0"], ["07/15/24 10:05:23", "0.0"], ["07/15/24 10:05:14", "0.0"], ["07/15/24 10:01:23", "0.0"], ["07/08/24 10:05:39", "0.0"], ["07/08/24 10:01:28", "0.0"], ["07/08/24 10:01:09", "0.0"], ["07/08/24 10:00:58", "0.0"], ["07/01/24 10:29:44", "0.0"], ["07/01/24 09:31:02", "0.0"]]

[image]

### Version Info

V1.19.03

### Logs

Submitted

jgyates, 2024-10-11:
Hi @skipfire 

I would try with version 1.19.04.

Also, do you have log2ram enabled on the /etc/genmon folder. There may be an issue with the RAM disk getting flushed with dietpi? 

Genmom should only attempt to delete the kwlog if it has reached the maximum size (the default maximum size is 15MB) and emails are sent out warning at 85% and 95% of the max.

jgyates, 2024-10-11:
Also, I would disable log2ram and see if you can duplicate the issue.

jgyates, 2024-10-11:
your logs look good

skipfire, 2024-10-11:
Upgrade to 1.19.04 worked, thanks!  I've got the install on the NanoPi on eMMC so I skipped log2ram  Good to know about the file size limit.