jgyates/genmon #1203: Error in MopekaBT:Start: [Errno 22] Invalid argument : mymopeka.py:402

msteil4vse, 2024-10-18:
### Steps to Reproduce

Having unrelated WiFi connectivity issues, and upgraded Pi Packages, and Genmon to see if simply updating corrected issues.   post upgrade Gemon is not reading sensor data.  I have submitted registers and logs, from web interface.  Error in the mopeka log is "Error in MopekaBT:Start: [Errno 22] Invalid argument : mymopeka.py:402"

### Expected Result

see submitted logs

### Actual Result

see submitted logs

### Version Info

current version.

### Logs

submitted

jgyates, 2024-10-18:
I have not received your logs. Do you have outbound email setup and functional? Outbound email is needed to send your logs via the About page.

It is not clearn that this was working before.
How did you upgrade the pi packages? apt-get or pip or did you update genmon?
Did you reboot after updating? 
Have you validated that bluetooth is enabled with the raspi-config program?
Are you using the 32 bit or 64 bit version of the OS
What pi variant are you using?

msteil4vse, 2024-10-18:
Hmm, don’t know, the logs are attached to this email.  I to have email configured and it sends alerts to me.  See other answers bellow:

 

Yes it was working prior to upgrading outdated Pi Packages and upgrading to current version and Genmon.  This particular unit has experienced some WiFi connectivity issues in recent days and the reason for upgrading was to get to latest and greatest platform wide in attempt to chase that issue.

Used apt-get and Yes upgraded genmon.  Commands issued:

 

apt -y update

apt -y upgrade

apt autoremove

apt autoclean

cd /home/genmonpi/genmon/

/home/genmonpi/genmon/genmonmaint.sh -u

/home/genmonpi/genmon/startgenmon.sh restart

shutdown -r now

Yes

Didn’t find the option to control Bluetooth in raspi-config.

32 Bit

cat /proc/cpuinfo

processor       : 0

model name      : ARMv7 Processor rev 4 (v7l)

BogoMIPS        : 38.40

Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 

CPU implementer : 0x41

CPU architecture: 7

CPU variant     : 0x0

CPU part        : 0xd03

CPU revision    : 4

 

processor       : 1

model name      : ARMv7 Processor rev 4 (v7l)

BogoMIPS        : 38.40

Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 

CPU implementer : 0x41

CPU architecture: 7

CPU variant     : 0x0

CPU part        : 0xd03

CPU revision    : 4

 

processor       : 2

model name      : ARMv7 Processor rev 4 (v7l)

BogoMIPS        : 38.40

Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 

CPU implementer : 0x41

CPU architecture: 7

CPU variant     : 0x0

CPU part        : 0xd03

CPU revision    : 4

 

processor       : 3

model name      : ARMv7 Processor rev 4 (v7l)

BogoMIPS        : 38.40

Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 

CPU implementer : 0x41

CPU architecture: 7

CPU variant     : 0x0

CPU part        : 0xd03

CPU revision    : 4

 

Hardware        : BCM2835

Revision        : 9020e0

Serial          : 000000007d44fb1c

Model           : Raspberry Pi 3 Model A Plus Rev 1.0

jgyates, 2024-10-18:
github will strip email attachments. If you want to send the logs you have to attached (drop them in the text window) on the github web site.

What happens if you type this

     bluetoothctl

Then type "exit" to quit. This should show if your controller is enabled. 

I just validated that the mopeka add on works for me with the latest OS and genmon updates on a 32 bit OS with a pi 3 B. 

It is possible you have some package conflicts with the "apt upgrade". I usually use "apt dist-upgrade" as it attempts to do some dependancy mapping instead of just blindly updating everything. From the man page:

     upgrade
        upgrade is used to install the newest versions of all packages
        currently installed on the system from the sources enumerated in
        /etc/apt/sources.list. Packages currently installed with new
        versions available are retrieved and upgraded; under no
        circumstances are currently installed packages removed, or packages
        not already installed retrieved and installed. New versions of
        currently installed packages that cannot be upgraded without
        changing the install status of another package will be left at
        their current version. An update must be performed first so that
        apt-get knows that new versions of packages are available.

    dist-upgrade
        dist-upgrade in addition to performing the function of upgrade,
        also intelligently handles changing dependencies with new versions
        of packages; apt-get has a "smart" conflict resolution system, and
        it will attempt to upgrade the most important packages at the
        expense of less important ones if necessary. So, dist-upgrade
        command may remove some packages. The /etc/apt/sources.list file
        contains a list of locations from which to retrieve desired package
        files. See also apt_preferences(5) for a mechanism for overriding
        the general settings for individual packages.

msteil4vse, 2024-10-18:
[genmon_logs.tar.gz](https://github.com/user-attachments/files/17441144/genmon_logs.tar.gz)

here are logs?

msteil4vse, 2024-10-18:
bluetoothctl

Agent registered

[bluetooth]# show 

Controller B8:27:EB:EE:51:B6 (public)

        Name: HeadendGenMon.vsenterprisesltd.com

        Alias: HeadendGenMon.vsenterprisesltd.com

        Class: 0x002c0000

        Powered: yes

        Discoverable: no

        DiscoverableTimeout: 0x000000b4

        Pairable: yes

        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)

        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)

        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)

        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)

        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)

        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)

        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)

        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)

        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)

        Modalias: usb:v1D6Bp0246d0537

        Discovering: no

        Roles: central

        Roles: peripheral

Advertising Features:

        ActiveInstances: 0x00 (0)

        SupportedInstances: 0x05 (5)

        SupportedIncludes: tx-power

        SupportedIncludes: appearance

        SupportedIncludes: local-name

[bluetooth]#

 

apt dist-upgrade

Reading package lists... Done

Building dependency tree... Done

Reading state information... Done

Calculating upgrade... Done

0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

jgyates, 2024-10-18:
The damage may be done on the apt upgrade vs dist-upgrade. Running dist-upgrade after upgrade will not resovle existing dependency issues. dist-upgrade will attempt to resolve dependancies "before" the upgrade.

I got your logs. The error is coming from the call to start scanning here:

https://github.com/jgyates/genmon/blob/3ffb03c9643f9d83694b5e90b62c9421ec33e563/genmonlib/mymopeka.py#L402

This is a function inside the [bleson python module](https://github.com/TheCellule/python-bleson/). Genmon does assume you only have one bluetooth controller and passes this value as zero.

As a test you can run this from the genmon directory:

[image]

If you get an error on the line with get_provider then try changing the line to this 👍 

     controller = get_provider().get_adapter(1)

When I try a value of 1 I expect and error because there is not a 2nd controller on my pi. When I do try a value of 1 instead of zero I get "OSError: [Errno 19] No such device" not "[Error 22]"

If you get an error 22 with this test then I would suggest reinstalling a fresh copy of the OS. You can backup you genmon settings and restore them as described in the [wiki](https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon).

Let me know how it goes.

I am going to close this issue as I don't think this is something I can fix within genmon but feel free to post follow up status or questions here as others may benefit from the resolution.

jgyates, 2024-10-18:
I forgot to add the start_scanning line in the above test. Run this instead:

[image]

jgyates, 2024-10-18:
here is the plain text if you wanted to copy and past each line:

    pi@test16:~/genmon $ source ./genenv/bin/activate
    (genenv) pi@test16:~/genmon $ sudo ./genenv/bin/python3
    Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
    Type "help", "copyright", "credits" or "license" for more information.
     from bleson import get_provider
     controller = get_provider().get_adapter(0)
     controller.start_scanning()
     controller.stop_scanning()
     exit()
    (genenv) pi@test16:~/genmon $ deactivate
    pi@test16:~/genmon $

msteil4vse, 2024-10-21:
Good AM,

 

Had busy weekend and now just getting back to this.  I must be fundamentally missing some aspect, and I cannot find a “genenv” directory in the filesystem on the Pi, to do as requested

 

Mark

 

(genenv) ***@***.***:~/genmon $ sudo ./genenv/bin/python3
(genenv) ***@***.***:~/genmon $ deactivate

jgyates, 2024-10-21:
The presence of a genenv directory indicates a managed python environment is used by the OS. A managed python environment is active with raspberry pi OS bookworm (Debian 12). If you are running debian 11 (bullseye) or earlier then you will not have a genenv directory. If you don't have a genenv folder then the commands you would run wold be:

     sudo python3
      from bleson import get_provider
      controller = get_provider().get_adapter(0)
      controller.start_scanning()
      controller.stop_scanning()
      exit()

msteil4vse, 2024-10-21:
That explains allot then.  Here is the output

 

***@***.***:~$ sudo python3

Python 3.9.2 (default, Mar 12 2021, 04:06:34) 

[GCC 10.2.1 20210110] on linux

Type "help", "copyright", "credits" or "license" for more information.

>>> from bleson import get_provider

>>> controller = get_provider().get_adapter(0)

>>> controller.start_scanning()

Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

  File "/usr/local/lib/python3.9/dist-packages/bleson/providers/linux/linux_adapter.py", line 280, in start_scanning

    self.set_scan_filter()

  File "/usr/local/lib/python3.9/dist-packages/bleson/providers/linux/linux_adapter.py", line 115, in set_scan_filter

    self._set_filter(filter)

  File "/usr/local/lib/python3.9/dist-packages/bleson/providers/linux/linux_adapter.py", line 54, in _set_filter

    self._socket.setsockopt(socket.SOL_HCI, socket.HCI_FILTER, data)

OSError: [Errno 22] Invalid argument

>>> controller.stop_scanning()

>>> exit()

***@***.***:~$

jgyates, 2024-10-21:
what is the output of this command: 

     uname -r

this shows your kernel version.

jgyates, 2024-10-21:
also show the output of this command:

    sudo pip3 freeze

this shows the versions of the installed python packages

I am trying to isolate the differences between my 32 bit bullseye test system and yours.

msteil4vse, 2024-10-21:
***@***.***:~$ uname -r

6.6.57-v7+

msteil4vse, 2024-10-21:
***@***.***:~$ sudo pip3 freeze

aiohttp==3.9.1

aiohttp-retry==2.8.3

aiosignal==1.3.1

arandr==0.1.10

astroid==2.5.1

asttokens==2.0.4

async-timeout==4.0.3

attrs==23.1.0

automationhat==0.2.0

beautifulsoup4==4.9.3

bleson==0.1.8

blinker==1.7.0

blinkt==0.1.2

buttonshim==0.0.2

Cap1xxx==0.1.3

certifi==2020.6.20

cffi==1.16.0

chardet==4.0.0

chump==1.6.0

click==8.1.7

colorama==0.4.4

colorzero==1.1

configparser==6.0.0

crcmod==1.7

cryptography==41.0.7

cupshelpers==1.0

dbus-python==1.2.16

distro==1.5.0

docutils==0.16

drumhat==0.1.0

envirophat==1.0.0

ExplorerHAT==0.4.2

flask==3.0.0

fluids==1.0.25

fourletterphat==0.1.0

frozenlist==1.4.0

geojson==2.5.0

gpiozero==1.6.2

html5lib==1.1

idna==2.10

importlib-metadata==7.0.0

isort==5.6.4

itsdangerous==2.1.2

jedi==0.18.0

Jinja2==3.1.2

lazy-object-proxy==0.0.0

ldap3==2.9.1

logilab-common==1.8.1

lxml==4.6.3

MarkupSafe==2.1.3

mccabe==0.6.1

microdotphat==0.2.1

mopeka-pro-check==0.1.0

mote==0.0.4

motephat==0.0.3

multidict==6.0.4

mypy==0.812

mypy-extensions==0.4.3

numpy==1.19.5

oauthlib==3.1.0

paho-mqtt==1.6.1

pantilthat==0.0.7

parso==0.8.1

pexpect==4.8.0

pgzero==1.2

phatbeat==0.1.1

pianohat==0.1.0

picamera==1.13

picamera2==0.3.12

pidng==4.0.9

piexif==1.1.3

piglow==1.2.5

pigpio==1.78

Pillow==8.1.2

ply==3.11

psutil==5.9.6

pyasn1==0.4.8

pycairo==1.16.2

pycparser==2.21

pycryptodomex==3.17

pycups==2.0.1

pygame==1.9.6

Pygments==2.7.1

PyGObject==3.38.0

pyinotify==0.9.6

PyJWT==2.6.0

pylint==2.7.2

PyOpenGL==3.1.5

pyOpenSSL==23.3.0

pyotp==2.9.0

pyowm==2.10.0

PyQt5==5.15.2

PyQt5-sip==12.8.1

pyserial==3.5

pysmbc==1.0.23

pysmi==0.3.4

pysnmp==4.4.12

python-apt==2.2.1

python-prctl==1.7

pytz==2023.3.post1

rainbowhat==0.1.0

reportlab==3.5.59

requests==2.25.1

requests-oauthlib==1.0.0

responses==0.12.1

roman==2.0.0

RPi.GPIO==0.7.1

RTIMULib==7.2.1

SciPy==1.8.1

scrollphat==0.0.7

scrollphathd==1.2.1

Send2Trash==1.6.0b1

sense-hat==2.6.0

simplejpeg==1.6.4

simplejson==3.17.2

six==1.16.0

skywriter==0.0.7

smbus==1.1.post2

sn3218==1.2.7

soupsieve==2.2.1

spidev==3.6

ssh-import-id==5.10

thonny==4.0.1

toml==0.10.1

touchphat==0.0.1

twilio==8.10.3

twython==3.8.2

typed-ast==1.4.2

typing-extensions==3.7.4.3

unicornhathd==0.0.4

urllib3==1.26.5

v4l2-python3==0.3.2

voipms==0.2.5

webencodings==0.5.1

werkzeug==3.0.1

wrapt==1.12.1

yarl==1.9.4

zipp==3.17.0

jgyates, 2024-10-21:
what about this command:

     cat /etc/os-release 

I am guessing you are running bookworm as your python version is 3.9.2

Are you running the Raspberry pi OS? I see that your kernel version is 6.6.57-v7+.

On my system with bullseye (fully updated pi3b with 32 bit OS) the kernel version is 6.1.21-v8+

On my system with bookworm (fully updated pi 4 with 64 bit OS) the kernel version is 6.6.51+rpt-rpi-v8

Your kernel is newer than both of the test systems I am using.

I am thinking that running such a forward leaning version of the kernel with a relatively back dated OS may be the source of the incompatibility you are experiencing.  we are running the same version of python, same bluetooth python module version (bleson) but the kernel version is very different.

I did find a similar error in [this post](https://github.com/Adam-Langley/pybleno/issues/63). Note that in this post the issue has more to do with the kernel than the python library.

Since downgrading the kernel and keeping envything running can be a tricky thing to do, in my view your best option is to backup the genmon settings and reinstall. Personally I would only update the system software with these commands:

     sudo apt-get update
     sudo apt-get dist-upgrade

I also only allow the above command to change the kernel. I never update the kernel otherwise as it introduces a lot of untested scenarios. 

Backup and restore genmon info: https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

Let me know if you have any questions of if there is any updated status on this issue.

I am going to close this thread but feel free to post here if you have any other questions or status updates.

closed 2024-10-21 by jgyates

msteil4vse, 2024-10-21:
Thank you for the assistance.

 

I have a spare unit that has been setting in the drawer for over a year, -- both units were from Printsize.me same architecture.  I did upgrade the unit as it did not have the current version of genmon, I did as outlined below, and I’m about to go install the spare unit but it appears the dist-upgrade upgraded this unit to the same code as the one that is installed is running.

 

 

apt-get update

apt-get dist-upgrade

shutdown -r now

/home/genmonpi/genmon/genmonmaint.sh -u

/home/genmonpi/genmon/startgenmon.sh restart

shutdown -r now

ftp -i 192.168.30.11

tar -vxzf genmon_backup.tar.gz

cp -v ./genmon_backup/*.* /etc/genmon

/home/genmonpi/genmon/startgenmon.sh restart

cat /root/.bash_history 

 

cat /etc/debian_version

11.11

 

pip3 freeze

arandr==0.1.10

astroid==2.5.1

asttokens==2.0.4

automationhat==0.2.0

beautifulsoup4==4.9.3

bleson==0.1.8

blinker==1.4

blinkt==0.1.2

buttonshim==0.0.2

Cap1xxx==0.1.3

certifi==2020.6.20

chardet==4.0.0

chump==1.6.0

click==7.1.2

colorama==0.4.4

colorzero==1.1

configparser==5.3.0

crcmod==1.7

cryptography==3.3.2

cupshelpers==1.0

dbus-python==1.2.16

distro==1.5.0

docutils==0.16

drumhat==0.1.0

envirophat==1.0.0

ExplorerHAT==0.4.2

Flask==1.1.2

fluids==1.0.22

fourletterphat==0.1.0

geojson==2.5.0

gpiozero==1.6.2

html5lib==1.1

idna==2.10

isort==5.6.4

itsdangerous==1.1.0

jedi==0.18.0

Jinja2==2.11.3

lazy-object-proxy==0.0.0

ldap3==2.9.1

logilab-common==1.8.1

lxml==4.6.3

MarkupSafe==1.1.1

mccabe==0.6.1

microdotphat==0.2.1

mopeka-pro-check==0.1.0

mote==0.0.4

motephat==0.0.3

mypy==0.812

mypy-extensions==0.4.3

numpy==1.19.5

oauthlib==3.1.0

paho-mqtt==1.6.1

pantilthat==0.0.7

parso==0.8.1

pexpect==4.8.0

pgzero==1.2

phatbeat==0.1.1

pianohat==0.1.0

picamera==1.13

picamera2==0.3.12

pidng==4.0.9

piexif==1.1.3

piglow==1.2.5

pigpio==1.78

Pillow==8.1.2

ply==3.11

psutil==5.8.0

pyasn1==0.4.8

pycairo==1.16.2

pycryptodomex==3.17

pycups==2.0.1

pygame==1.9.6

Pygments==2.7.1

PyGObject==3.38.0

pyinotify==0.9.6

PyJWT==2.6.0

pylint==2.7.2

PyOpenGL==3.1.5

pyOpenSSL==20.0.1

pyotp==2.8.0

pyowm==2.10.0

PyQt5==5.15.2

PyQt5-sip==12.8.1

pyserial==3.5b0

pysmbc==1.0.23

pysmi==0.3.4

pysnmp==4.4.12

python-apt==2.2.1

python-prctl==1.7

pytz==2022.7.1

rainbowhat==0.1.0

reportlab==3.5.59

requests==2.25.1

requests-oauthlib==1.0.0

responses==0.12.1

roman==2.0.0

RPi.GPIO==0.7.0

RTIMULib==7.2.1

SciPy==1.8.1

scrollphat==0.0.7

scrollphathd==1.2.1

Send2Trash==1.6.0b1

sense-hat==2.6.0

simplejpeg==1.6.4

simplejson==3.17.2

six==1.16.0

skywriter==0.0.7

smbus==1.1.post2

sn3218==1.2.7

soupsieve==2.2.1

spidev==3.5

ssh-import-id==5.10

thonny==4.0.1

toml==0.10.1

touchphat==0.0.1

twilio==7.16.4

twython==3.8.2

typed-ast==1.4.2

typing-extensions==3.7.4.3

unicornhathd==0.0.4

urllib3==1.26.5

v4l2-python3==0.3.2

voipms==0.2.5

webencodings==0.5.1

Werkzeug==1.0.1

wrapt==1.12.1

 

uname -r

6.6.57-v7+

msteil4vse, 2024-10-21:
Any chance you can point me to the Mopeka Bluetooth setup documentation?  I don’t seem to be asking the right questions of the search tool.

 

Mark

jgyates, 2024-10-21:
Here is the info on setting up mopeka with genmon.  https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmopekapy-optional

In short you just need the bluetooth MAC address of the sensor to give to the genmon add on. You can get this form the mopeka mobile app or with the genmon utility described in the link above.

For the raspberry pi, on early versions of the pi (like the 3A that you are using) there were some issues with using an external serial port and bluetooth at the same time that you had to work around. The [project serial port setup tool](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialconfigpy-optional) should handle these for the pi OS, but I am not sure how compatible this is on other distributions as I only include Raspbian as a test point.

What linux distribution are you running? you showed /etc/debian_version but not /etc/os-release. The kernel version implies something different from the Raspberry pi OS (which is also debian based)

I just ran distro-update on my 32 bit debian 11.11 system and my kernel is 6.1.21-v8+

My Raspberry pi OS system using debain 12.7 is on kernel 6.6.51+rpt-rpi-v8 

If you are on 6.6.57-v7+ with debian 11 it sounds like you are not using a distro from the raspberry pi foundation?

msteil4vse, 2024-10-21:
Sorry, I thought I had given it to you.  See below.  Both units were using Raspbian bullseye.  The spare unit is exhibiting the same issues, so there is probably a code related issue that is rearing it’s ugly head.  Am I to infer that are a V4 Pi might work better?  

 

In all our genmon implementations there is a V3B or a V3A that was purchased from pintsize.me with the associated HAT and cabling.  If upgrading the Pi Hardware would fix the issues with the sensors I’d happily order some V4 Pi’s 

 

 

cat /etc/os-release 

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"

NAME="Raspbian GNU/Linux"

VERSION_ID="11"

VERSION="11 (bullseye)"

VERSION_CODENAME=bullseye

ID=raspbian

ID_LIKE=debian

HOME_URL=http://www.raspbian.org/

SUPPORT_URL=http://www.raspbian.org/RaspbianForums

BUG_REPORT_URL=http://www.raspbian.org/RaspbianBugs

***@***.***:/home/genmonpi#

jgyates, 2024-10-21:
Unfortunatly I can only guess that a pi upgrade may work. The pi 3A may work differently than pi 3B or 4, but I don't have any data to say that it will or will not regarding bluetooth. My test system with 32 bit debian 11.11 system and kernel 6.1.21-v8+ is a pi 4B, it also tested it with a pi 5 and it works there too.

jgyates, 2024-10-21:
I installed a fresh copy of 32 bit raspbian 12.7 on a pi 3B when we started this thread, did a distro-upgrade and my kernel is 6.6.51+rpt-rpi-v7

ju2wheels, 2024-10-22:
> I did find a similar error in [this post](https://github.com/Adam-Langley/pybleno/issues/63). Note that in this post the issue has more to do with the kernel than the python library.

I dont know anything about this tool or any components/plugins but if the issue is similar to what I hit then its related to the CVE fixintroduced into 5.16, 6.8.7, and 6.9 or newer kernel lines which started to more strictly enforce sizing constraints around structs passed into kernel userland C functions by other libraries (such as from Python). In my cases it was related to Bluetooth, but I dont recall if it was BT only and it could have been patches across various parts of the general networking stack too.

So if you are looking for a search, you may find it by looking for any Python components leveraging C bindings to userland kernel functions and ensure any structs being passed into calls are properly sized. Thats even more important for ARM architecture which may not use the same C type sizings as AMD64. If it is this issue, its not going to be a 32bit vs 64bit issue since both kernels would have the same patches. Its just a matter of are you using a kernel version before these patches or after.

msteil4vse, 2024-10-22:
Thank you for the help.  I have the unit operational now.  I suspect others may have the same issue at some point, so salt this away.   The Version 3, both A and B Pi sourced from Pintsize.me the unit had been running flawlessly with Mopeka sensors.  In my case the Version A unit started experiencing WiFi connectivity issues, and while attempting to chase down that problem I updated the packages on the Pi, and that is the point I broke the sensor readings, and did not seem to make any difference with my WiFi issues.  I had a spare Pi this one a 3B but sourced same way with same software load.  I repeated the package upgrade process and again reading the Mopeka sensors got broken in the upgrade. So I then did a clean install of the 32 bit Raspbain 12.7, updated all the packages, did a clean install of Genmon, and restored config from backup.  Presto all is working again. 

 

Not entirely certain what to do about the 3 other deployments of Genmon.  They are all working with un-updated original software load, so I am presuming going forward I will need to repeat the process at some point.

 

Again thanks for the effort you have put forth assisting me.

 

Mark

jgyates, 2024-10-22:
@ju2wheels 

Thanks for the info this helps a lot.

@msteil4vse 

Were the other two pi units loaded with Raspbian from the Raspberry pi foundation? If they are not, then I would do a fresh install. I have 5 different pis all with Raspbian  (a mix of bullseye, bookworm and one buster) all of them have been updated to the latest from their respective repository and I have never seen this issue which leads me to believe that the Raspberry pi foundation either has not merged this change, rejected this change or has some type of patch to accommodate the change. I am not sure of the bleson maintainer would make the needed change, but worst case, I could submit a PR to bleson (if they accept it) or make a small version of bleson for this project. I did a similar thing with the [mopeka_pro_check python library](https://github.com/spbrogan/mopeka_pro_check), which as abandoned by the developer.

msteil4vse, 2024-10-23:
The units with the problem all came from Pintsize.me around March of 2023, and it is my understanding that they were pre-loaded with Raspberry Pi OS Bullseye, current for the time.  The units were deployed and worked flawlessly reading the Mopeka tank sensors until we recently updated the packages for an unrelated issue.  The update only affected the reading of the Mopeka sensors so there probably isn’t a huge number of affected users.