jgyates/genmon #1204: External Temperature Sensor

philruel, 2024-10-20:
Hi,
What values should I put to have a dial from -40 to 50 C.  Right now, I have 170 Nominal and 250 Maximum and the dial is shown bellow.

[image]

[image]

Thank you!

jgyates, 2024-10-20:
I will have to make a mod to gentemp.py to accommodate a gauge that starts at a value below zero. I will update this thread when I have made the change.

jgyates, 2024-10-20:
I updated the software to support non zero minimum values from this add on. 

first, update the software from the about page, then add the minimum value to the add on settings. Let me know if you have any issues or questions.

zachliu, 2024-10-21:
So the maximum value is the max on the dial. Then what exactly is the "nominal" value? The almighty internet explains it as "the typical value of the sensor's output". But I still don't understand why we need them here. Shouldn't we only need min and max for a dial?

jgyates, 2024-10-21:
Nominal values help determine where the gauge green area ends and the yellow begins. The external temp method does not allow the input of the exact tuning of the gauge coloring (to keep things simple) so it just added a fixed value to the nominal to determine where the yellow starts and another fixed value to where the red starts.

zachliu, 2024-10-21:
ohhh, so the fixed values are 20F and 40F

[image]

jgyates, 2024-10-21:
:)

closed 2024-10-21 by jgyates

zachliu, 2024-10-22:
i naively thought i could submit a PR to add this info to the tool tip :joy: 
 
[image]

jgyates, 2024-10-22:
fyi, the latest update corrects the typo and added a sentence to give more details on the nominal, max and minimum values

zachliu, 2024-10-22:
yep, i noticed. but the tool tip doesn't get texts from `conf/gentemp.conf` right?

jgyates, 2024-10-22:
correct. The [add on tool tips are in genserv.py](https://github.com/jgyates/genmon/blob/b5a81ca12d1b1d7fe00e9b6bcf9ce350b0ab521e/genserv.py#L2087)

zachliu, 2024-10-22:
damn, forgot to add `""` when i did the search, that's why i couldn't find where they are...