jgyates/genmon #1206: Unable to set mopeka sensor conf file to debug mode.

notms, 2024-11-01:
### Steps to Reproduce

find conf file in conf folder for the mopeka sensor. Set debug=True, restart the pi, go back and nothing. I also tried modifying the conf file in the conf2 folder as well as bringing the conf file for the mopeka sensor to the genmon folder as other comments were made on the forum and nothing that activated the debug mode. I verified that it's able to write to the log file by setting the tank to be custom and it gave me an error.

### Expected Result

to write a debug data

### Actual Result

nothing

### Version Info

latest V1.19.04

jgyates, 2024-11-01:
The files in the genmon/conf folder are the original, unmodified files. The files that actually hold the settings are located in the /etc/genmon folder. You should be editing the file /etc/genmon/genmopeka.conf

jgyates, 2024-11-01:
I am closing this issue but feel free to post any follow up questions or status to this thread.

closed 2024-11-01 by jgyates

notms, 2024-11-01:
I didn't have the genmopeka.conf file in the /etc/genmon/ folder I put it there manually. Where is it writing the config file then?

jgyates, 2024-11-01:
The conf files are copied from the ~/genmon/conf folder to /etc/genmon by the install script. Have you followed the install instructions in the wiki?

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

notms, 2024-11-01:
Yes I have it fully running, should I re-run the install?

notms, 2024-11-01:
I am running a re-install, what I am ultimately trying to do is log the fuel gauge from topeka to log to a file where I will take the data and put it in grafana since MQTT doesn't look like captures the data.

jgyates, 2024-11-01:
MQTT should capture the fuel level. If it shows up in the text in the web interface then it will be exported to MQTT.

Yes, you should follow the directions in the wiki. If you do not checkout the files via git and run the install script then some things will not work (like software updates and some add on programs).

notms, 2024-11-01:
Thank you for your quick response

notms, 2024-11-01:
@jgyates I just did a fresh install and my directory structure is the following:

[image]

jgyates, 2024-11-01:
That looks correct.

The above only shows the ~/genmon folder. This is relative to the pi users home folder (/home/pi/genmon). The ~ denotes it is relateive to the home folder of the current logged in user.

To see the /etc/genmon folder type 

      ls -la /etc/genmon

and you should see the conf files. A software update will update all of the files in ~/genmon which will overwrite the ~/genmon/conf files. The files that keep persistent settings are located in /etc/genmon since they will not be overwritten with a software update.