jgyates/genmon #1212: CallMeBot Issue after switching to PiZero2W

Ic3Nine, 2024-11-26:
### Steps to Reproduce

Hey folks, 
I just switched from pi 3b to a  Pi zero 2w.  I don't have any external power so i was using power from the battery and seems like it wasn't enough and kept getting cpu throttling due to under-voltage.  Every now and then the pi would stop working and needed to get power cycled.

Enter PiZero2W.  Lower power consumption and under-voltage issue gone but now callmebot stopped working.  I restored the config files from the backup.  Everything else seems to be working but i've turned off the add-on, reconfigured it. No messages to signal.  Last one was about 5pm from the 3b complaining about undervoltage again(prior to the switch). gencallmebot.log is empty so not sure what to look at.

### Expected Result

Receive Generator Ready/PiHealth OK alerts in Signal from CallMeBot

### Actual Result

No messages

### Version Info

Genmon
Version V1.19.04

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

### Logs

[genmon_logs(1).tar.gz](https://github.com/user-attachments/files/17910683/genmon_logs.1.tar.gz)

jgyates, 2024-11-26:
You log file for the callmebot addon is zero so there is not useful information in this file. 

How did you install genmon on the  Pi zero 2w? Did you run the install script?
Did you backup and restore your config files from the pi 3b to the pi zero 2w?

Ic3Nine, 2024-11-26:
Yes, fresh install of os. Fresh install of genmon and then restored config files following the wiki article.

jgyates, 2024-11-26:
ok. Let's do a few things to validate the setup.

1) type this command (you don't have to show the output here as it may contain information specific to your account):

       cat /etc/genmon/gencallmebot.conf

validate that the settings are correct for your installation. This should also allow you to validate the settings / config restore process worked as expected.

2) Edit the file /etc/genmon/gencallmebot.conf and add the following line:

      debug=True

Then restart genmon with this comand in the ~/genmon folder (the genmon folder in your home directory that you installed genon:

      ./startgenmon.sh restart

Post the output of this command to this thread.

Once genmon has restarted then show the contents of the file /var/log/gencallmebot.log with this command:

     cat /var/log/gencallmebot.log

Post the output of this command to this thread.

Ic3Nine, 2024-11-26:
I did check the callmebot config file and it seemed fine. 

I'll add debug and post in a few

jgyates, 2024-11-26:
Also, do you recall if your pi 3b was also running debian 12 bookworm?

Ic3Nine, 2024-11-26:
> Also, do you recall if your pi 3b was also running debian 12 bookworm?

it is running 12 - 
`PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"`

Ic3Nine, 2024-11-26:
errr - okay so I added debug = true and restarted.  And it started working.  This is weird but i'll take it.

`barry@genmon:~/genmon $ cat /var/log/gencallmebot.log
2024-11-25 21:02:50,218 : SIGNAL Selected
2024-11-25 21:02:50,219 : Type:signal
2024-11-25 21:02:50,225 : Min Wait Between Messages: 5
2024-11-25 21:02:51,081 : Message Sent
2024-11-25 21:02:56,681 : Message Sent
2024-11-25 21:03:02,358 : Message Sent
2024-11-25 21:03:08,066 : Message Sent
2024-11-25 21:03:13,712 : Message Sent`

jgyates, 2024-11-26:
you can try to remove the debug and see if it still works. Maybe it just needed are restart?

Ic3Nine, 2024-11-26:
> you can try to remove the debug and see if it still works. Maybe it just needed are restart?

Did that, still worked.  Also did sudo reboot, worked as well.  I'll take it.  Not sure why it didn't work before, i rebooted a few times.

Are all the timeouts in mymodbus expected?

also myserial.log - these timeouts okay/expected?
`2024-11-25 21:28:25,515 : Error: timeout receiving slave packet for register 04dd Buffer: 8, sequence 258
2024-11-25 21:28:27,115 : Buffer = [0x9d,0x03,0x0a,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xc1,0xd8]`

jgyates, 2024-11-26:
No. The major cause of time out eras is usually caused by a weak wireless signal when using serial over TCP. If you are using a wired serial connection then timeouts are typically caused by a issue with your cabling or grounding. If you're using a non-wireless medium, you really should have no timeouts.

jgyates, 2024-12-11:
I am going to close this issue but if you have any follow up questions or status feel free to post to this thread.

closed 2024-12-11 by jgyates