jgyates/genmon #1216: can you add support for mebay generator controller

maindrive, 2024-12-08:
I'm going to raise the white flag... I need help!
I also have an MeBay gen controller that I would like your help with adding support.
I can communicate with the controller in the Python virtual environment  with modbusdump and other scripts, but unable to get anything to load with the genmon web interface. I have included the modbusdump with IP, Serial and the controller json file that I have started, any help would be greatly appreciated.

[image]
[image]
[DC42D-ABF MK3_controller_json.txt](https://github.com/user-attachments/files/18054410/DC42D-ABF.MK3_controller_json.txt)
[DC42D-ABF_MK3_modbusdump.txt](https://github.com/user-attachments/files/18054409/DC42D-ABF_MK3_modbusdump.txt)

Thanks for your support!

### Discussed in https://github.com/jgyates/genmon/discussions/1159

<div type='discussions-op-text'>

<sup>Originally posted by **yellowdre** July 29, 2024</sup>
### Checklist

- [X] I made sure that there are *no existing issues or discussions* - [open issues](https://github.com/jgyates/genmon/issues),[closed issues](https://github.com/jgyates/genmon/issues?q=is%3Aissue+is%3Aclosed),[open discussion](https://github.com/jgyates/genmon/discussions), [closed discussion](https://github.com/jgyates/genmon/discussions/categories/questions?discussions_q=is%3Aclosed) - which may answer my question or that I could contribute my information to.
- [X] I have read the [Known Issues Wiki Page](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues) and my question / issue is not listed.
- [X] I have provided the version of the software and the operating system in my post.
- [X] If I have an issue with genmon not communicating with the controller, I have read and followed the [serial troubleshooting guide](https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting).
- [X] I have included the generator make and model in my submission.

### Have you submitted your log files from the About page? Note that outbound email must be working for logs to be submitted.

I have submitted logs

### What is/are your question(s)?

i have a mebay generator controller and i have the modbus  registers but i dont know how to get it working with genmon
can someone please help me with this, i have include modbus manual
[DC6xD MODBUS_Communication protocol_V1.2-20231124.pdf](https://github.com/user-attachments/files/16413771/DC6xD.MODBUS_Communication.protocol_V1.2-20231124.pdf)

### Additional information

_No response_</div>

jgyates, 2024-12-09:
Hi @maindrive 

Thanks for the data. Give me a few days and I can get a working controller file.

maindrive, 2024-12-09:
Woo Hoo!! Big thanks for your support and fast response.

jgyates, 2024-12-09:
Hi @maindrive 

I have a lot of it working, just working on some rough edges...

I noticed there are some modbus registers defined in your JSON file that are not in the data sheet, for example 

"1048": {"length": 2, "text": "Model_number/hw_version"},
and Box temperature is in the data sheet but not your JSON file. 

Also, the doc you sent is for DC6xD controller but you labeled the controller DC4xDMK3 in the JSON file. I am trying to make sure this works on the right models. On one hand I can lock this down to a specific controller with the model number modbus registers, but on the other hand it would be nice for this to work on a wider set of controllers.

do you have any other documents? If you do you can send them to generatormonitor.software@gmail.com

maindrive, 2024-12-10:
Hello jgyates,

The original post was not mine.

I found this on the web in someone else code.

     """The Mebay controller identifies itself by a combination
    of manufacturer code and model number."""

    def __init__(self):
        super().__init__(0x1048, 2)

    def decode(self, values):
        model_number = values[0]
        hw_version = values[1]
        ident_str = f"{ model_number }-{ hw_version }"
        return self.update(ident_str)

The DC8xD, DC6xD controllers all have the same base modbus registers as the DC4xD mine being the lower end controller.
I do not think DC4xD has register 1057H Box temperature value, I'm seeing all 0000.
I have sent the PDF of modbus registers for the DC4xD controllers to the email you provided. 

Big thanks for your support!

jgyates, 2024-12-11:
I am going to close this thread an move it to the discussion forum. We can continue on this thread:

https://github.com/jgyates/genmon/discussions/1159

closed 2024-12-11 by jgyates