jgyates/genmon #1218: External Tank Gauge shows zero

lampe208, 2024-12-17:
### Steps to Reproduce

[image]

### Expected Result

Tank utiliity app shows correct reading but gen mon is zero.  The batteries were low and were replaced prior to replacing all of a sudden the gauge was zero. 

### Actual Result

[genmon_logs.tar.gz](https://github.com/user-attachments/files/18170450/genmon_logs.tar.gz)

### Version Info

v1.19.04

lampe208, 2024-12-17:
actual reading should be 30%

lampe208, 2024-12-17:
Im also getting notices that the tank is below zero?   I confirmed tank reading at tank visual at 30%

jgyates, 2024-12-17:
I don't really have enough info to make anything but a guess.

You mentioned that you are getting notices that the tank is below zero. What is the message? Is this sent from genmon or and add on or the tank utility app.

You logs show that the request is being rejected by the tank utility server. your gentankutility.log file is filled with this error:

2024-12-17 13:27:08,798 : tankutility: Error logging in, error code: 400
2024-12-17 13:27:08,800 : Error logging in in TankCheckThread, retrying

This error should have nothing to do with the tank sensor batteries as the add on does not read from the sensor directly but from the online readings stored by the tank utility. Has your password changed recently and if so did you update the add on setting?

lampe208, 2024-12-17:
Time: 02:04:19 PM
Date: Tue 17-Dec-2024
Warning: The estimated fuel in the tank is at or below 10%. Fuel Level is
0.0%

Above is the email .  What’s weird is it’s sending the notice but it’s in
accurate. I’ll check the password.

Sent from Gmail Mobile

lampe208, 2024-12-17:
thank you very much. I fixed it. I changed the password. It's not accurate.

lampe208, 2024-12-17:
Now accurate

closed 2024-12-17 by lampe208

jgyates, 2025-07-23:
I have not received any logs today. Is your outbound email setup and functional? This is required to submit logs from the About page.

What version of the software are you using?

lampe208, 2025-07-23:
You are not running the latest version.
    Current Version: V1.19.05
    New Version: V1.19.06
[image: image.png]
Im getting error on the email as well when I test it.

jgyates, 2025-07-23:
you did not mention or show the scree capture of the error. 

If you are responding to this thread in email, github will strip any attached photos. To submit a screencap or photo you have to attach it from the github web interface.

Have you ever updated the software successfully before? If not, did you follow the directions for installation here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

if you did not use the "git clone" command as mentioned in the link then your software updates will not work.

lampe208, 2025-07-23:
[](

[image]

this is the error when i do a test email settings

lampe208, 2025-07-23:
ok will do

lampe208, 2025-07-25:
> Both the email error message and a the he failure to update could be
> caused by your pi not having access to the Internet from you local network.
> I would try rebooting the pi.

This morning I started getting emails that the tank is low so I know it’s
communicating. I did a test email that works. I did the update that works.
However now I have no dashboard no visuals like a hand with all the gauges.

Sent from Gmail Mobile

lampe208, 2025-07-25:
I have not refreshed the browser. I do have gauges visible, but no readings
on any gauges. I do have CPU tip, which is my raspberry pie temperature
that’s it

Sent from Gmail Mobile

jgyates, 2025-07-25:
I received you logs. You have several issues.

You first issue is that your serial comms are not working. From the logs you send here is the summary of communication stats:

Comm Stats :
   Packet Count : M: 77, S: 5
   CRC Errors : 0
   CRC Percent Errors : 0.00%
   Timeout Errors : 70
   Timeout Percent Errors : 90.91%
   Modbus Exceptions : 3
   Validation Errors : 0
   Sync Errors : 0
   Invalid Data : 0
   Discarded Bytes : 1
   Comm Restarts : 0
   Packets Per Second : 0.38
   Average Transaction Time : 0.0412 sec
   Modbus Transport : Serial
   Serial Data Rate : 9600

You should be able to see these your self on the Monitor page. You logs indicate that genmon is only running for 3 minutes so I would not expect the web interface to come up yet with not serial data. The software would need to attempt every modbus read at least once so if you waited about 10 - 15 min you will see a default web page. If serial comms are not working the software can not detect any thing about the controller so all of your data displayed will be zero or default values.

The next issue is that you email still has a configuration error. You have this error in /var/log/mymail.log:

2025-07-25 06:48:06,405 : Error selecting mail folder! (select)

This indicates that you have an inbound email folder that does not exist. If you are not using inbound email command then do not enable the option "Optional - Inbound Email Commands Processing"

I also noticed that there are several add on programs enabled that do not appear to be configured correctly. I would recommend disabling any add on programs that you are not using as this just takes system resources. For example, you have the "Notifications via SMS - LTE Hat" add on enabled, but you do not appear to have the additional hardware requried to run this add on.

You also have the SMS Twillio add on enabled, not sure if you are using this but there are errors in the log. This could be caused by your previous network issuse, but if you are not using and active Twillio account you should disable this add on.

You have the tank utility enabled but the error in /var/log/gentankutil.log is this:

2025-07-25 06:44:38,311 : tankutility: Error logging in, error code: 400
2025-07-25 06:44:38,312 : Error logging in in TankCheckThread, retrying

Has this add on worked for you before?

The add on for Slack, Alexa and email to SMS are also enabled, all of these have errors related to network isseus which may be related to your previous netowrk problem that was solved with a reboot, but I would recommend disabling any add that you are not using.

Your main issue is that your  serial comms are not working. I would inspect your cabling and then see the serial troublshooting guide: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Regarding your tank reading zero, if you still see this after you serial comms are restored, the next question is this: is your tank utility a LTE or Wifi version. If it is the LTE version, you may have the issue described here: https://github.com/jgyates/genmon/issues/1284

lampe208, 2025-07-25:
> I have always been able to update.

I have resolved the issues .  Simply rebooting the pi has pretty much
solved the problem.

lampe208, 2025-07-25:
Thank you sir.  I did send a separate email.  I managed to ssh into the pi
remotely and it all seems to be working.  I will disable those add ons.  I
do use slack, but twilo is not working so I will disable that as well.

Thanks again.

lampe208, 2025-11-25:
> great, I am glad it is working for you.

I have also lost my tank monitor window

lampe208, 2025-11-25:
> I got your logs. You have an intermittent serial connection. that data
> from your logs indicate that genmon was running for about 20 min but here
> are your communication stats:
>
> Comm Stats :
> Packet Count : M: 408, S: 8
> CRC Errors : 0
> CRC Percent Errors : 0.00%
> Timeout Errors : 399
> Timeout Percent Errors : 97.79%
> Modbus Exceptions : 3
> Validation Errors : 0
> Sync Errors : 0
> Invalid Data : 0
> Discarded Bytes : 0
> Comm Restarts : 0
> Packets Per Second : 0.34
> Average Transaction Time : 0.0426 sec
> Modbus Transport : Serial
> Serial Data Rate : 9600
>
> The monitor status was: "Monitor Health": "System Initializing. Not
> receiving data. "
>
> It could be a lose wire of connector. You should be able to isolate the
> issue once you are in proximity to the controller with the serial
> troubleshooting guide
>
> Let me know if you have any questions.

Thanks a bunch. I will check it out.

lampe208, 2025-12-14:
> Has anything changed on this system, like installed software or software
> updates. I can't thing of ever hearing of a serial port just changing
> unless some other action has been taken.
>
> Normally you would set the genmon settings on the web interface or via the
> install script to select which serial port you are using, but since your
> serial port has changed in some way you have a couple of options on how to
> proceed.
>
> you can try setting up the software again
> <https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software>, but
> you really need to get a handle on what is going on with your serial port
> first.
>
> Type this:
>
>  ls -la /dev/ser*
>
> and this:
>
> ls -la /dev/ttyA*
>
> note these have the lowercase L not a one.
>
> This will display the serial port
>
> You can also run raspi-config, the program that comes with the pi OS to
> enable / disable hardware.
>
> The genmon install script uses the program serialconfig.py. I would run
> this first. Log into the pi via ssh and then type this:
>
> cd genmon
> cd OtherApps
> sudo python3 serialconfig.py -c
>
> Show the output of this and the output of the above "ls" commands.

***@***.****:*~ $*  ls -la /dev/ser*

lrwxrwxrwx 1 root root 5 Dec 13 14:24 */dev/serial1* -> *ttyS0*

***@***.****:*~ $* ls -la /dev/ttyA*

ls: cannot access '/dev/ttyA*': No such file or directory

***@***.****:*~ $* cd genmon

***@***.****:*~/genmon $* cd OtherApps

***@***.****:*~/genmon/OtherApps $* sudo python3
serialconfig.py -c

Using files: /boot/config.txt and /boot/cmdline.txt

Model: Raspberry Pi 3 Model B Rev 1.2

Checking : Is enable UART in boot config: Fail

Checking : Is BT overlay disabled: Fail

Checking : Serial console command line removed: OK

Checking : Serial console service disabled: OK

Checking : BT service disabled: OK

Serial port may not work as expected. Not all required settings changes
were detected.
Sent from Gmail Mobile

lampe208, 2025-12-14:
> I did not get your attachment. If you reply in email you can not send
> attachments as github will strip them out. If you want to post an image you
> will have to use the github web interface. There is a link in the bottom of
> the email notification for this thread to take you to the github genmon
> forum thread.
>
> If you can log into ssh, then type this command once logged in to you pi
>
>  cd genmon
>  python3 ClientInterface.py
>
> If genmon is running, then you will get a prompt. From the prompt type
> this:
>
> generator: monitor
>
> If genmon is running this will output a lot of text, you can save your
> console log or scroll back to see all of the text. It should look something
> like this:
>
> Generator Monitor Stats :
>     Monitor Health : OK
>     Controller : Evolution, Liquid Cooled
>     Run time : Generator Monitor running for 4 days, 22:05:10.
>     Power log file size : 1.16 MB of 15.00 MB
>     Generator Monitor Version : V1.19.07
>     Update Available : No
>
>
> Communication Stats :
>     Packet Count : M: 14047775, S: 14047774
>     CRC Errors : 0
>     CRC Percent Errors : 0.00%
>     Timeout Errors : 0
>     Timeout Percent Errors : 0.00%
>     Modbus Exceptions : 0
>     Validation Errors : 0
>     Sync Errors : 0
>     Invalid Data : 0
>     Discarded Bytes : 0
>     Comm Restarts : 0
>     Packets Per Second : 66.09
>     Average Transaction Time : 0.0302 sec
>     Modbus Transport : Serial
>
> Send me the output.
>
> to exit the prompt type
>
> exit
>
> Let me know how it goes.

Still cannot access the 192.168.4.181 via chrome or safari or firefox.

jgyates, 2025-12-14:
So what changed? Your serial ports were not present before and now they are. Did you do anything to make that happen or did they just show up after not being there? This information is required to determine if you still have a problem that can be solved without replacing hardware.

You mentioned "After redoing the software here is the output". Can you give more details on what you did to redo the software?

You also posted contradictory data with the output of the ClientInterface.py command. 

You appear to have an underlying issue of your serial port went away and then came back. That behavior will cause all of the behavior you are seeing (no serial ports, genmon not loading successfully, genmon reporting communication failures if the port goes away after genmon is loaded).

It is possible you have a failing SD card and this is causing the serial port to not be recognized by the OS, which would cause genmon to fail. 

The data you are posting needs context, for example if you show the serial ports with the "ls" command, is genmon working at that time? If genmon stops working are the serial ports active?

lampe208, 2025-12-14:
i just reloaded the software then ran thru the checks .

Sent from Gmail Mobile

lampe208, 2025-12-14:
so esentially i reloaded the genmon software as if I just started to do this project.   I then ran the check of the serial ports after you sent me how to see what kind I had.  So the last post is the checking of what serial port i have.   What now should I do now that we know what type of serial port I have according to my last post.?

lampe208, 2025-12-14:
pi@raspberrypigenerator:~/genmon/OtherApps $ sudo python3 serialconfig.py -c

Using files: /boot/config.txt and /boot/cmdline.txt

Model: Raspberry Pi 3 Model B Rev 1.2

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: Fail
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: Fail

Serial port may not work as expected. Not all required settings changes were detected.

This is the latest as of now .

lampe208, 2025-12-15:
> ARNING: System Initializing.  Not receiving data. : ,

>generator: monitor

Monitor :

    Generator Monitor Stats :

        Monitor Health : OK

        Controller : Evolution 2.0, Air Cooled

        Run time : Generator Monitor running for 5:16:01.

        Power log file size : 0.22 MB of 15.00 MB

        Generator Monitor Version : V1.19.07

        Update Available : No

    Communication Stats :

        Packet Count : M: 611332, S: 611331

        CRC Errors : 0

        CRC Percent Errors : 0.00%

        Timeout Errors : 0

        Timeout Percent Errors : 0.00%

        Modbus Exceptions : 40

        Validation Errors : 0

        Sync Errors : 0

        Invalid Data : 0

        Discarded Bytes : 0

        Comm Restarts : 0

        Packets Per Second : 64.48

        Average Transaction Time : 0.0306 sec

        Modbus Transport : Serial

        Serial Data Rate : 9600

    Platform Stats :

        CPU Temperature : 60.80 F

        Pi Model : Raspberry Pi 3 Model B Rev 1.2

        Pi CPU Frequency Throttling : OK

        Pi ARM Frequency Cap : OK

        Pi Undervoltage : OK

        CPU Utilization : 1.4%

        OS Name : Raspbian GNU/Linux

        OS Version : 10 (buster)

        System Uptime : 10:32:44

        Network Interface Used : wlan0

        WLAN Signal Level : -65 dBm

        WLAN Signal Quality : 45/70

        WLAN Signal Noise : -256 dBm

        WLAN ESSID : FBI Access point

        System Time : Sunday December 14, 2025 18:41:05

lampe208, 2025-12-15:
So what changed? Your serial ports were not present before and now they
are. Did you do anything to make that happen or did they just show up after
not being there? *I simply renistalled the software as if I was starting
from scratch*. This information is required to determine if you still have
a problem that can be solved without replacing hardware.

You mentioned "After redoing the software here is the output". Can you give
more details on what you did to redo the software?  *I installed the genmon
software from https://github.com/jgyates/genmon
<http://github.com/jgyates/genmon>*You also posted contradictory data with
the output of the ClientInterface.py command.* Not sure what you mean by
contradictory data?*

You appear to have an underlying issue of your serial port went away and
then came back. That behavior will cause all of the behavior you are seeing
(no serial ports, genmon not loading successfully, genmon reporting
communication failures if the port goes away after genmon is loaded).

It is possible you have a failing SD card and this is causing the serial
port to not be recognized by the OS, which would cause genmon to fail.

The data you are posting needs context, for example if you show the serial
ports with the "ls" command, is genmon working at that time? If genmon
stops working are the serial ports active?  *Please tell me what I need to
do to get you this information.  I sent some information what exactly as
far as commands do i need to do to see where we are at now?*

lampe208, 2025-12-15:
> Is your web interface working?
>
> In not, if you use ClientInterface.py can you still get status from the
> "generator: monitor" command?
> If you are not seeing the web interface then show the last 10 lines in
> both /var/log/genmon.log and /var/log/genserv.log
>
>  cat /var/log/genmon.log
>  cat /var/log/genserv.log

Is your web interface working?  *NO*

In not, if you use ClientInterface.py can you still get status from the
"generator: monitor" command?  *Yes, I can get the status from this command*
If you are not seeing the web interface then show the last 10 lines in both
/var/log/genmon.log and /var/log/genserv.log:

See Below the last lines

2025-12-14 08:01:38,723 : Possible cause is invalid serial port specified.

2025-12-14 08:09:02,914 : Selected Controller: generac_evo_nexus

2025-12-14 08:09:03,254 : GenMon Loaded for site: SiteName using python
3.7: VEnv: False

2025-12-14 08:21:21,047 : Generator Monitor Shutdown

2025-12-14 08:21:23,795 : Selected Controller: generac_evo_nexus

2025-12-14 08:21:24,134 : GenMon Loaded for site: SiteName using python
3.7: VEnv: False

2025-12-14 08:41:49,320 : Selected Controller: generac_evo_nexus

2025-12-14 08:41:49,653 : GenMon Loaded for site: SiteName using python
3.7: VEnv: False

2025-12-14 13:24:30,975 : Generator Monitor Shutdown

2025-12-14 13:25:04,586 : Selected Controller: generac_evo_nexus

2025-12-14 13:25:04,922 : GenMon Loaded for site: SiteName using python
3.7: VEnv: False

2025-12-14 18:54:24,837 : Selected Controller: generac_evo_nexus

2025-12-14 18:54:25,170 : GenMon Loaded for site: SiteName using python
3.7: VEnv: False

2025-12-14 19:10:54,620 : Selected Controller: generac_evo_nexus

2025-12-14 19:10:54,647 : Error opening modbus device: Error opening modbus
device. : generac_evolution.py:331

2025-12-14 19:10:54,647 : Possible cause is invalid serial port specified.cat
/var/log/genmon.log

 cat /var/log/genserv.log

—

 cat /var/log/genmon.log
 cat /var/log/genserv.log

lampe208, 2025-12-15:
Help me understand why i can use the generator monitor command and see data, but my browser is still not allowing me to log into genmon .

lampe208, 2025-12-15:
> No problem. The program genmon.py outputs data in one format. That format
> is not exactly human readable but close. The format is JSON. The program
> ClientInterface.py takes the JSON data from genman.py and converts it to
> text and prints it to the screen. The program genserv.py functions as a web
> server and takes data in the JSON format from genmon.py and converts it to
> data that is presentable as a web page, HTML and JavaScript. The problem
> you are having with the web server in genserv.py is not loading, I'm
> guessing due to a corruption it with your python libraries.

*Does this mean its running?*

***@***.****:*~ $* ps -ef | grep genserv.py

root       922     1  0 10:45 ?        00:00:06 /usr/bin/python3
/home/pi/genmon/*genserv.py* -c /etc/genmon/

pi        2094  1672  0 13:32 pts/1    00:00:00 grep --color=auto
*genserv.py*

***@***.****:*~ $*

lampe208, 2025-12-15:
[image]

lampe208, 2025-12-15:
making some progress.  Just no data now 

[image]

jgyates, 2025-12-15:
I have not received any logs. You have to have outound email setup and working to be able to submit logs. Also, you should get email when genmon starts if you have your email setup correctly.

By default genmon uses port 8000. If you enable secure HTTPS option in the web interface then it will use the default port (i.e. not use 8000 at the end) with a user name and password. The default settings (which I assume you got when you reinstalled if you told the installer to copy the config files) do not use a user name and password.
 
do you have any output when you type the following

   cat /var/log/genserv.log 

Before you said there was nothing, but if the web server is loading you should be getting something. 

What you are describing is normal considering you reinstalled. What is not normal is you say you don't have anything in the genserv.log file. If that is true something is very wrong with your file system and I would not trust the system to run genmon. If you do have data in /var/log/genserv.log then you *might* can run this reliably.

You can see the serial data stats on the monitor page.

lampe208, 2025-12-15:
So how do i get back to my normal screen I had a week ago when i log in and
can see my voltage , tank monitor, email etc... I dont have any of that
just wifi signal and CPU temp.,,, its not reading my generator

lampe208, 2025-12-16:
/dev/serial0

lrwxrwxrwx 1 root root 7 Dec 15 10:41 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Dec 15 10:41 /dev/serial1 -> ttyS0

pi@raspberrypigenerator:~ $   ls -la /dev/ttyA*
crw--w---- 1 root tty 204, 64 Dec 15 18:29 /dev/ttyAMA0

lampe208, 2025-12-16:
> That looks correct. At this point I would look at your connections from
> the pi to the converter and from the converter to the controller. Make sure
> all connections are good, especially the ground connection. You can have
> everything connected correctly and loopback test pass but if the ground
> connection between the controller and the RS-232 interface is not making a
> solid connection, then serail communications will not occur. If you want to
> post pictures of your wiring it may help.

how do i get the voltage gauges to work?  i had that even though i had not
receiving data  i posted a pic of what i had before

Sent from Gmail Mobile